jmamma/MCL #4: tempo drift

yatli, 2019-03-25:
there's some kind of feedback loop in the clock. the reading on MC jitters between two integral values (e.g. 127 & 128), while MD says 126.5; in some conditions, hitting play will bump the tempo on MD to 127 -- which in turn bumps MC clock to jitter between 128 & 129.

go downstairs for a smoke, come back and get some DnB :p

jmamma, 2019-03-25:
It's not jitter, it's just rounding of the tempo calculation for visual purposes.

313 void GridPage::display_grid_info() {
314 #ifdef OLED_DISPLAY
315   uint8_t x_offset = 43;
316   uint8_t y_offset = 8;
317 
318   oled_display.setFont(&Elektrothic);
319   oled_display.setCursor(0, 10);
320   oled_display.print(round(MidiClock.tempo));

This is expected behaviour and mimics the elektron devices.
For example, if  you slave your A4 to your MD, you will see the same rounding.

MCL does not generate its own clock, it uses either clock from MIDI1 or MIDI2. 

Tempo calculation is in MidiClock.cpp and is re-calculated every 16th note. 
MIDI clock messages are sent at a rate of 24 ppqn. That means there will be 24 MIDI messages per beat. 24 / 4 = 6 messages per 16th note.

284 void MidiClockClass::incrementCounters() {
285   mod6_free_counter++;
286   if (mod6_free_counter == 6) {
287     tempo = ((float)150000 / ((float)midi_clock_diff(last_clock16, clock)));
288     div192th_time = midi_clock_diff(last_clock16, clock) / 12;
289     last_clock16 = clock;
290     mod6_free_counter = 0;
291   }

---

All internal sequencer events are triggered within ISR, as a response to a MidiClock message.
This means, they do not rely on tempo calculation and are locked in to the external clock.

jmamma, 2019-03-26:
"in some conditions, hitting play will bump the tempo on MD to 127 ".

I'm aware of this one, my suspicion is that the MidiStart event coincides closely with a MidiClock message. There is a short interrupt that runs on MidiClockStartImmediate, that will probably block the read of the clock message. So the first tempo calculation after MidiStart will be in some circumstances unreliable.

closed 2019-04-16 by jmamma

jmamma, 2019-06-03:
bc0a7ae6502a0912e8e00fd6138f400e5300328b

^

Fixed.