jmamma/MCL #11: Program Size. Investigate ways of reducing MIDICtrl size.

jmamma, 2019-04-05:
Currently MCL is using:

**Sketch uses 203550 bytes (80%) of program storage space. Maximum is 253952 bytes.**

Might want to start thinking of ways to cut this down.

A lot of the low level code is inlined. All the SYSEX Encoder/Decoder is inlined. Not sure what the space saving will be vs performance hit.

A general rule I've been following is to inline in ISR.

jmamma, 2019-04-05:
Unused glyphs from the Adafruit-GFX-Library should be culled.

yatli, 2020-03-14:
Current situation:

[image]

yatli, 2020-03-14:
```
g=global
l=local
F=function
O=progmem
```

yatli, 2020-03-14:
So, functions are absolute heavy hitters.
After the previous rounds of optimization, now the PROGMEM is negligible.

yatli, 2020-03-14:
Tried a few tricks.

The ones that I thought would be very effective, but actually made things worse:
- NO_INLINE some big routines
- Fold some repeated logic by hand

The ones that I thought wouldn't matter but very effective:
- int -> uint8_t

yatli, 2020-03-14:
Setting up a new branch for size optimization.

yatli, 2020-03-14:
#108

yatli, 2021-04-22:
To wrap up this thread:

- Two attempts are made in 2020, bringing down sketch size to 78% (#108, #125)
- As of 2021, with features are being added, it creeps back to 90% and keeps growing
- #161 introduces compression to squeeze the last drops out of the mega2560

closed 2023-08-01 by jmamma