jmamma/MCL #13: MIDI TX Buffer sizes.

jmamma, 2019-04-06:
@yatli (might want to think about this)

A word on the TX Buffers:

The TX ring buffers are currently sized at 3KB. This seems sufficient for the largest dumps I have been transmitting in MCL (MDPattern). TX buffer transmission happens on ISR, when the TX queue for the UART is clean another byte from the buffer is read, and transmitted. The rate at which the TX buffer gets consumed is limited by the port speed.

These might need to be bumped up for handing A4Pattern MNMPattern sizes to prevent buffer overflow. 

Alternatively, the data encoder code, could be modified to "wait for the UART queue to be empty' before sending more encoded bytes.

yatli, 2019-04-08:
why does TX even need a buffer?
will the user buffer be modified during asynchronous send?

jmamma, 2019-04-08:
MidiUart class has two commands for sending data. m_putc() and m_putc_immediate().

m_putc_immediate() will skip the TX buffer and wait for the UART queue to be clean before transmitting. 

Without the buffer, program execution would be slow and would have to wait for each byte to be sent, before proceeding with next line of code.

yatli, 2019-04-08:
thanks for the explanation. I think I misunderstood the intent.

Anyway, I'm providing a little bit background info here:

In operating systems we have zero-copy async I/O (example: Windows I/O completion port, abbr. IOCP), where the OS accepts a user buffer and use it directly without a kernel buffer. The requirement is that the user buffer should never be touched by the user code until completion.

The assumption is quite different, that a user posts "buffers" to a network driver, not byte-by-byte.
In the byte-by-byte scenario, we do need a buffer.

jmamma, 2019-04-08:
Understood. Welcome to the world of MIDI :smiley:

jmamma, 2019-05-27:
tx buffers now stored in high RAM bank.

closed 2019-05-27 by jmamma