jmamma/MCL #14: Chain mode: not all tracks progress together

yatli, 2019-04-08:
It's a little bit confusing, not all tracks progress together in chain mode. I'm applying loop=4,row=(cur+1),apply=(1-20) to a row, and after 4 bars only some tracks progress to the next row:

[image]

yatli, 2019-04-08:
the result is fascinating though.. the pattern never repeats

jmamma, 2019-04-08:
Can you confirm chain mode == AUT (automatic)

yatli, 2019-04-08:
Like this:

[image]

jmamma, 2019-04-08:
Hmm. 

That looks correct.

I can't replicate your behaviour. Which branch are running off?

The transition step for each track is calculated using the internal sequencer track length, and the number of loops you have set in the slot settings.

So if all tracks are to transition at the same time, they must have the same track length + number of loops.

```
665 void MCLActions::calc_next_slot_transition(uint8_t n) {
666 
667   DEBUG_PRINT_FN();
668   uint16_t len;
669   DEBUG_PRINTLN(n);
670   //  DEBUG_PRINTLN(next_transitions[n]);
671   next_transitions_old[n] = next_transitions[n];
672 
673   if (n < NUM_MD_TRACKS) {
674     len = chains[n].loops * mcl_seq.md_tracks[n].length;
675   } else {
676     len = chains[n].loops * mcl_seq.ext_tracks[n - NUM_MD_TRACKS].length;
677   }
678   if (len < 4) {
679     len = 4;
680   }
681   next_transitions[n] += len;
682 
683   // check for overflow and make sure next nearest step is greater than
684   // midiclock counter
685   while ((next_transitions[n] >= next_transitions_old[n]) &&
686          (next_transitions[n] < MidiClock.div16th_counter)) {
687     next_transitions[n] += len;
688   }
689   DEBUG_PRINTLN(next_transitions[n]);
690 }
```

jmamma, 2019-04-08:
If track length is consistent across slots, can you upload your project file so I can take a look.

yatli, 2019-04-09:
Uploaded the projects here:
https://www.yadli.net/index.php/s/89tGf5fWRQ6Kmjj

jmamma, 2019-04-09:
yat_md808 project.

Looks like your track lengths are different for each slot in row 0.

1) Use chain mode to write 16 tracks from row 0. This will load the slot(s) sequencer data in to the MCL internal sequencer.

2) Once the tracks are loaded, open the Step Sequence Page for each track and you will see that some track lengths are 16, and others are 64.

3) In Step Edit Page, you can set all 16 tracks to be the same track length by holding shift 2 and rotating encoder 3.
 
4) Then you must save these slots again from the Save Page.

5) Saving slots overwrites the chain settings for each slot, so you must configure chain settings again from the slot menu.

You will see some tracks have track length 16, others have track length 64.

yatli, 2019-04-09:
I double checked my procedure:

1. I uploaded the md808/md909 kits and patterns with C6; Patterns in bank F,G,H; Kits from 001 to 013.
2. Switch to pattern F01 on MD (md909 pattern 1); Kit "MD909HOUSE2" is automatically loaded. Scale=64, plays fine on MD.
3. Now, on MC, create a new project. Push Save button, then Shift2 to pull the entire pattern down.
4. On MD, switch to an empty pattern (say B01), clear the pattern to double check.
5. Play. The pattern is streamed back from the internal sequencer. All good.
6. On MC, check the slots individually. Track1 scale=64, track2 scale=16, ...

I checked F01 and found that, despite Scale=64, some tracks are empty (no active triggers in 64/64) so, during [Save] it is converted to Scale=16?

jmamma, 2019-04-09:
Okay, your procedure looks right. And I think i know the problem.

With Seq Merge == AUTO, the MC should merge all MD sequencer data to the internal sequencer on Save. The track length for each internal sequencer track should be set to the MDPattern length.

However, i added a fix recently to prevent the internal sequencer length changing, if the MD track is empty. 

I think i need to change this, to check to see if the internal sequencer  is also empty, if it's empty then still use MD->track length.

```
390 void MDSeqTrack::merge_from_md(MDTrack *md_track) {
391   DEBUG_PRINT_FN();
392 
393   if (md_track->trigPattern == 0) {
394   //If the MD sequencer data is empty, abort merge.
395   //This will prevent unnecessary length change of internal seq pattern
396   return;
397   }
398   set_length(md_track->length);
399 
```

can you change line 393 to:

 if ((md_track->trigPattern == 0) && (pattern_mask != 0)) {

yatli, 2019-04-09:
Yep that works. 💃

yatli, 2019-04-09:
as this is a minor touch, I'm pushing it up to the sysex branch.

jmamma, 2019-04-09:
Cool. Small fixes like this can go on dev, or in this case I will cherry-pick on to master.

yatli, 2019-04-09:
how do I mute the internal sequencer on MC?
I applied "clear" on a row, and cleared the MC sequence, and then chain-write it back so there's nothing on MD. But it kept streaming row0.

jmamma, 2019-04-09:
You can quickly clear all MD internal Sequencer tracks by entering a sequencer page.
and [ holding down shift 2 ] + [ press write ].

But let me check to see if i can replicate your behaviour

yatli, 2019-04-09:
could you please try this:

1. clear a row in MC
2. observe that it kept streaming row0
3. manually edit the sequence to clear it
4. observe that nothing remains while playing, looking good
5. chain-write the empty row
6. row0 is back

jmamma, 2019-04-09:
yeah i just noticed.

jmamma, 2019-04-09:
ha! i think it's the track-cache buffer. let me try and work out why this is happening

jmamma, 2019-04-09:
it's not the buffer. i can have two empty rows still load up sequencer data.

jmamma, 2019-04-09:
okay. i'm not checking to seee if the slot is empty.

This was sort of done intentionally, as I want the chain data for empty rows to still be loadable.

yatli, 2019-04-09:
I guess it's yet another check during write.
"if the slot is empty then ... "

yatli, 2019-04-09:
Switching off chain mode does not affect this behavior.

jmamma, 2019-04-09:
this issue has a few dependencies. i'm going to need to think it through.

yatli, 2019-04-09:
emm, I save the empty row and then it can be written properly.

yatli, 2019-04-09:
So this is actually a behavior not covered in the manual:

```
Internal sequencer tracks are linked to the slot positions in the Grid.

When storing a track within the Save menu, the internal sequencer data for that track is
stored along with the track data in the specified slot.
Internal Sequencer tracks are loaded when tracks are sent to the MD or A4 in the Write
page.
```

But what I'm actually doing is to chain-write without first saving the row into the grid.
I do have auto save=ON, not sure if it's for this purpose.

jmamma, 2019-04-09:
> So this is actually a behavior not covered in the manual:

Which behaviour?

--- 

I'm looking in to this issue. The fact that it's also appearing in Write mode, with chain mode disabled is perplexing. 

For instance...

``` 26 bool MCLActions::place_track_inpattern(int curtrack, int column, int row,
 27                                        A4Sound *analogfour_sound,
 28                                        EmptyTrack *empty_track) {
 29 
 30   //       if (Grids[encodervaluer] != NULL) {
 31   MDTrack *md_track = (MDTrack *)empty_track;
 32   A4Track *a4_track = (A4Track *)empty_track;
 33   ExtTrack *ext_track = (ExtTrack *)empty_track;
 34   
 35   DEBUG_PRINT_FN();
 36   if (column < NUM_MD_TRACKS) {
 37   
 38     if (md_track->load_track_from_grid(column, row)) {
 39       memcpy(&(chains[column]), &(md_track->chain), sizeof(GridChain));
 40       
 41       grid_page.active_slots[column] = row;
 42       if (md_track->active != EMPTY_TRACK_TYPE) {
 43         md_track->place_track_in_sysex(curtrack, column);
 44         return true;
 45       }
 46     }
```

Line 43 should place the slot/track in to the MD pattern+kit sysex and ALSO load the sequencer data.
But it shouldn't do that if the guard statement on line 42 fails.

jmamma, 2019-04-09:
Found one of the problems...

```
587 void GridPage::apply_slot_changes() {
...
```
625   for (uint8_t track = 0; track < count && track + getCol() < 20; track++) {
626     if (slot_clear == 1) {
627       grid.clear_slot(track + getCol(), getRow());
628       reload_slot_models = true;
629     } else if (slot_copy == 1) {
630       SET_BIT32(slot_buffer_mask, track + getCol());
631     }
632     slot.active = row_headers[cur_row].track_type[track + getCol()];
633     //  if (slot.active != EMPTY_TRACK_TYPE) {
634     slot.store_track_in_grid(track + getCol(), getRow());
```

Clears the slot, but then saves it again. d'oh

jmamma, 2019-04-09:
```
632     else {
633     slot.active = row_headers[cur_row].track_type[track + getCol()];
634     //  if (slot.active != EMPTY_TRACK_TYPE) { 
635     slot.store_track_in_grid(track + getCol(), getRow());
636     }
```
fix ^

yatli, 2019-04-09:
that's some intense logic.. that fixes it up :)

jmamma, 2019-04-09:
Yeah, that looks like the culprit. The other behaviour should be explained from this.

yatli, 2019-04-09:
looks like now it doesn't write back to MD?

jmamma, 2019-04-09:
what do you mean?

jmamma, 2019-04-09:
there's another scenario currently if you're in write mode (or chain mode and sequencer stopped) 
the empty row won't be loaded. ill probably fix this

yatli, 2019-04-09:
> what do you mean?

Both [Write] and [Chain] now clears MD sequence data.

yatli, 2019-04-09:
oooh I know now, it's because the stored track data is stale.

jmamma, 2019-04-09:
not sure that I follow. can you explain in more detail.

yatli, 2019-04-09:
it's more of a UI issue.

I had an empty row pulled down from MD, and the machine names were displayed in the grid slots (IM, B2, SD etc)
Then, I started programming on the MC internal sequencer, thinking that a [Write] would send it back to the MD sequencer.
What actually happened is that the old empty tracks are sent back to MD, right?

yatli, 2019-04-09:
So in other words, MC internal sequencer data -> MD sequencer is impossible?

jmamma, 2019-04-09:
Yeah that's not supported.

yatli, 2019-04-09:
Got it, thanks!
I'll think about how to make improvements to the UI.
For example, the grid slots should ideally provide info about both MD sequencer and internal sequencer. "Unsaved changes" would be cool too.
Also, There's space left on the save/write/chain page so I'll probably draw some pixel icons to fill it :p

jmamma, 2019-04-09:
"So in other words, MC internal sequencer data -> MD sequencer is impossible?"

I guess to better appreciate why, you have to understand how the firmware evolved.

Originally Write mode was the only mode available. You would save and write tracks too and from the MD.

When the internal sequencer was added we started saving internal sequencer data along side the MD pattern + kit information.

Finally, I worked out that chain mode was possible, but in order to do it, we couldn't transmit pattern data back to the MD. So SEQ MERGE was invented to import your MD pattern data in to the internal sequencer. 

Write mode still has its uses. For instance, when the sequencer is stopped and chain mode is active, write mode is used behind the scenes to send the full kit information including Kit Name and Master FXs.

--

I think in the future write mode might become inaccssible to users. But for the time being it's useful to have both.

yatli, 2019-04-09:
Yeah I understand this (did my homework and read the manual!). To chain it smoothly we even need to swing across multiple patterns (P+/P-) to get the kit information right.

I use write mode as a poor man's PlusDrive :p

yatli, 2019-04-09:
I'm imagining the following workflow:

1. Create some sequence data on MD. Currently it's still more accessible because the correspondence between the screen and the parameters, easier to jump between tracks etc.
2. Save and merge it with MC like committing from working set into a version control system.
3. After that the MD pattern can be cleared & to be sketched for something else.

jmamma, 2019-04-09:
Write mode == poor man's plus drive. ✅ 

P+/P- are WriteMode specific behaviours and are poor man's chain mode.
(I wouldn't bother using these anymore unless you insist on using write mode)

Chain mode does seamless transitions by interleaving Machine/Sound data in between 16th notes. There's no pattern switching on the MD, just reassigning Machines to Tracks.

jmamma, 2019-04-09:
That workflow is valid, and similar to how I use it. 

Sometimes I will work with the internal sequencer. But most times I'll use the MD sequencer for quick work, and then save + merge when I'm satisfied.

yatli, 2019-04-09:
a quick question: with MC connected, can I tweak the parameters on MD while playing?
I'm getting some kind of cross-talking with parameters, between tracks.

jmamma, 2019-04-09:
Did you accidentally select voices in polyphonic mode? Parameter changes on one voice will automatically update the others.

jmamma, 2019-04-09:
You can tweak parameters, provided the trigger interface is not active.
The current page will close automatically if you try and tweak a parameter with the trig interface active. This is because the trigger exploit fails when the MD display is updated.

yatli, 2019-04-09:
I'm leaving the default settings for poly mode (first 4 tracks) but I'm not actually using this feature.
Should I disable the poly voices?

jmamma, 2019-04-09:
eek. are those default values ?

yatli, 2019-04-09:
yeah. :D

jmamma, 2019-04-09:
82   poly_mask = 0;

default should be no tracks selected. unless there is a bug somewhere?

jmamma, 2019-04-09:
i wonder if there is overflow
```
 14 class MCLSysConfigData {
....
 28   uint8_t cur_row;
 29   uint8_t cur_col;
 30   uint16_t poly_mask;
```

yatli, 2019-04-09:
hmm, it should not overflow in that way.. right?

jmamma, 2019-04-09:
I might have something weird like mcl_config.cur_col = (uint16_t) val;
somewhere

yatli, 2019-04-09:
perhaps I did change the settings but I'm not sure how I picked up the number 4.
I was like "ooh brilliant 4-voice poly mode!" :D

jmamma, 2019-04-09:
i'm suspicious because i think i may have seen this before, but ignored it for same reason.
will  need to investigate.

yatli, 2019-04-09:
I'll disable the poly voices and test for a while.
Let's see if the problem can be contained in the config initialization stage.

commit c24d2f39 (2019-04-09): fix #14
commit 19c8ff8d (2019-04-09): fix #14 again
closed 2019-04-16 by jmamma