jmamma/MCL #15: Seeing SYSEX ERROR CODE 1 in write mode.

jmamma, 2019-04-09:
This is on DEV branch.

jmamma, 2019-04-09:
```
For the Machinedrum there are 4 possible error codes:

-1 length error
-2 checksum error
-3 memory full/position out of range
-4 version error
```

jmamma, 2019-04-09:
Seems to happen when writing Row without internal sequencer data, after writing a row with internal data

jmamma, 2019-04-09:
In the past this was caused by internal sequencer data being mixed in with sysex messages.
However we now have  
```
65 void MidiUartClass::m_putc(uint8_t c) {
 66   //#ifdef BLAH
 67   if (c == 0xF0) {
 68     uart_block = 1;
 69   }
 70   if (c == 0xF7) {
 71     uart_block = 0;
 72   }
```

Sequencer should be automatically disabled during SYSEX.

jmamma, 2019-04-13:
I have reason to suspect that the MD is the problem here.

The SYSEX error code 1 happens when a pattern is transmitted and the MD sequencer is triggering sounds.

If I clear the current pattern, and then send a new pattern the problem does not arise.
If we use mute quantisation (muting all voices during pattern send) the problem does not arise. 

--

My best guess is that when improving the MIDI TX performance, we are now overloading the MD on pattern send.

yatli, 2019-04-13:
Try lower turbo midi speed?

jmamma, 2019-04-13:
Makes no difference above 1x.

I'm looking at implementing throttling.

jmamma, 2019-04-13:
Hmm throttling not as easy to implement due to asynchronous MIDI TX.

jmamma, 2019-04-13:
we will have to use m_putc_immediate in throttle

jmamma, 2019-04-13:
i've got a throttle implemented. doesnt seem to be helping.
very, very strange.

yatli, 2019-04-13:
if it works in 1x, I suggest we check out the turbo midi implementation.

I remember seeing something like “wait for the slave for a breathe” in the turbo midi spec.

jmamma, 2019-04-13:
"wait for breathe" -> that's only the handshake protocol. Once the turbo speed is established there is no further Turbo Midi protocol exchange. An active sense message or other MIDI message must traverse the link to keep the turbo speed active, otherwise it switches down to 1x.

I'm not using handshaking to set the turbo speed. Elektron was kind enough to tell me of a force speed  message which we use to connect at 8x. The turbo protocol on the MD, wont natively connect at 8x, due to some constrains on the handshake speed check.

yatli, 2019-04-13:
does it repro on different patterns?
maybe try some:
```
........xxxxxxxx
```

jmamma, 2019-04-13:
Okay throttle is working. I think my testing was not quite right.

I have a guard statement that only turns throttle on if sequencer is running. However when the megacommand reboots it cant tell if the MD is already running. So the guard was failing.

yatli, 2019-04-13:
Which branch? I'm on sysex_refactor and cannot repro:

1. Put full pattern on a MD sequencer track.
2. Save it to a row.
3. Play, and use the write mode on MC to send all tracks (a lot).
4. Kit switch ok. Pattern write ok.

yatli, 2019-04-13:
ooh I was in turbo 2x!
switching to 8x and I get 50% repro (timing chances perhaps)

jmamma, 2019-04-13:
Cool.

It seems that I need to throttle on kit send as well for it to be reliable.

jmamma, 2019-04-13:
80us throttle between midi bytes.

yatli, 2019-04-13:
great!

jmamma, 2019-04-13:
https://github.com/jmamma/MIDICtrl20_MegaCommand/pull/16

Still seeing occasional error even at 100us. might bump it up futher.

yatli, 2019-04-13:
Do you observe errors on 2x or 4x? With or without throttle.

yatli, 2019-04-13:
I suspect signal integrity if we are seeing different results.

jmamma, 2019-04-13:
yeah good point. let me try a different MIDI cable. and disable the throttle.

I will check 1x, 2x, 4x, 8x.

jmamma, 2019-04-13:
I can't reproduce on 1x, 2x.

4x and above and I see the problem.

I will test from TurboMIDI adapter and see if MD is doing the same thing,

jmamma, 2019-04-13:
I can't reproduce with TurboMIDI Adaptor + C6

jmamma, 2019-04-13:
maybe the throttle is not the answer here.

yatli, 2019-04-13:
Try capture traffic and replay?

jmamma, 2019-04-13:
Edit: MC is a very nice reveng tool!

I think traffic capture is a good idea. To see if the sysex from the MC is actually being truncated, or if the MD is too slow to read it all.

The fact that the problem only occurs when MD sequencer is running is highly suspicious

jmamma, 2019-04-13:
Going to eat some dinner and think about this more.

jmamma, 2019-04-13:
Found the problem.. i think.

jmamma, 2019-04-13:
md_setsysex_recpos(8, MD.pattern.origPosition); <- this guy

jmamma, 2019-04-13:
```
SYSEX set receive position for dump:
MIDI Byte | Purpose ------------+----------------------------
$6b %000aaaab
%0ccccccc %0ddddddd $f7
| | | | | |
Receive position ID
receive pos on type 0001=global 0010=kit 0100=pattern 1000=song
to b=0 => pos ccccccc, b=1 original position for ddddddd sysexes of this type
SYSEX end
```

jmamma, 2019-04-13:
[image]

jmamma, 2019-04-13:
hmm i think it's okay. from memory the MD needs one of these commands proceeding dump, otherwise it ignores it.

jmamma, 2019-04-13:
Above statement is not true.

MD will receive to correct location provided pattern/kit has origPosition set.

jmamma, 2019-04-13:
I'm seeing this issue as far back as : fadcd7862a9629efcd499edbbd2bceab8c638ed5

Version 2.16 does not have this problem. The Ring buffer speed fix was introduced shortly after, and the speed boost may have revealed this problem.

jmamma, 2019-04-13:
Listening on the UART_WRITE macro.

I see send length is always the same for Kit and Pattern sysex, regardless of pass or fail.
Packed length in DataEncoders is also consistent.

jmamma, 2019-04-13:
I can replicate same problem with C6 and MD

jmamma, 2019-04-13:
If sequencer is running and sending pattern + kit dump to MD, same SYSEX ERROR 1

jmamma, 2019-04-13:
The fix is throttle. But it might not be full proof.

This provides more motivation to switch to  Chain mode. Could disable write mode when sequencer is running indefinitely, and force users to use chain mode.

yatli, 2019-04-13:
Nicely done, sir!

jmamma, 2019-04-13:
I've discovered that if you disable MIDI note note transmit from the MD. By setting global channel to: --- . The SYSEX errors stop.

Also only if a certain number of tracks are triggering, does this problem occur.

Thus it seems like the MIDI output load is affecting the SYSEX receive behaviour.

jmamma, 2019-04-14:
The above does not hold true for higher sequencer load.

Seems like the actual DSP/cpu load is also a contributing factor.

jmamma, 2019-04-14:
The throttle solution seems to be the only way here.

I'm really not happy about implementing it. 

I've increased the throttle to 160 microseconds which should limit the transfer rate to 2x MIDI.
1/160us = 6250 Bps

1x MIDI BAUD Bps = 31250/10 = 3215Bps. 
2 x MIDI Midi Bps = 3215x2 = 6250 BPS

yatli, 2019-04-14:
Does the length of the triggered sound contribute?

This should isolate DSP related problems.

My alternative theory is, if DSP is ruled out, and the problem is more of MIDI workload problem, then do some cycle stealing between triggers should be better than a throttle.

yatli, 2019-04-14:
e.g. only transfer in a “safe house”

jmamma, 2019-04-14:
You'd need to anticipate the behaviour of the sequencer completely. Including swing.. 

Too much effort. 

I'm ready to deprecate user-facing write mode.

yatli, 2019-04-14:
Is throttle + higher turbo speed beneficial?
How about just limit turbo to 2x.

I mean, 8x is definitely useful, for transferring samples etc. but the problem here is stability during live performance.

yatli, 2019-04-14:
The root cause might be as simple as the MD 68k cpu dropping RX bytes in a high load situation.

jmamma, 2019-04-14:
Shifting turbo speeds causes clock sync drop out.

At 8x turbo, MD MIDI out is not throttled, so you can get speed boost in one direction at least.

yatli, 2019-04-14:
I agree. Plus, protocol reinit may cause its own problems. Probably not worth it.

About the “safe house” theory:

I haven’t looked at your implementation but I guess there is a throttling flag.

How about dynamically flipping on/off the throttle in our timing ISR? We have higher resolution, so we can bring the throttled period down to what MD can digest (e.g. 50% green zone between triggers)

jmamma, 2019-04-14:
The throttle is currently implemented in the ElektronDataEncoder.

You could include some conditional statements that check to see if MidClock.mod6_counter != 0 
This would mean, you're not on a 16th note.

Problem is, this doesnt account for tracks that have swing. In other words MD can trigger notes at any time.

Also, we're assuming the problem is  only occurring on Trigger. which might not be the case.

yatli, 2019-04-14:
ha. A better idea:

Whenever MD triggers a midi out byte, we employ throttle.

The ultimate problem should be the load, not simultaneous tx+rx

yatli, 2019-04-14:
Like a crude version of:

MC: ringring. Got a min? Got a min? Got a min?
MD: .........busy....busy..busy............

jmamma, 2019-04-14:
Yeah that's possible actually.

Somewhat tricky to implement. 

You'd need be constantly reading the MidiUart RX buffer. That currently only happens in main program loop. 

Midi.handleByte(MidiUart.m_getc()); 

Also, throttle may need to be applied before MD triggers a sound, not after.

jmamma, 2019-04-14:
Doesn't appear to be DSP related. If I substitute machines for GND, we have same issue.

jmamma, 2019-04-14:
``` 55 void ElektronDataToSysexEncoder::uart_send(uint8_t c) {
 56     if (throttle) {
 57       bool do_throttle = false;
 58       uint8_t v = 0;
 59       while ( v = uart->m_getc() > 0) { do_throttle = true; }
 60       if (do_throttle) {
 61               DEBUG_PRINTLN("throttlin...");
 62       delayMicroseconds(THROTTLE_US);
 63       }   
 64       uart->m_putc_immediate(c);
 65     }
 66     else {
 67 uart->m_putc(c);
 68     }
 69 }
```
Just tried this. 

It throttles, but we still get sysex 1 errors

jmamma, 2019-04-14:
throttling on MidiClock.mod6_counter = 0; works.

Breaks when swing is added.

Now just need to throttle on swing clock time as well.

jmamma, 2019-04-14:
I've got throttle working on swing now.

I've tested with slides and it's okay. 
My test patterns include notes every 16th step across 8 or more tracks.
If the patterns are busier (which is unrealistic) the delay time needs to be increased.

This fix, breaks with scale mode = 2x or 3/2

jmamma, 2019-04-14:
I have a fix for specific tempo modes now. But it appears the MD and MC step counters go out of sync if you switch scales on the fly, including changing scales between patterns.

jmamma, 2019-04-14:
this rabbit hole never ends... 🐇 🕳

[3 more replies]

closed 2019-04-16 by jmamma