jmamma/MCL #19: warning: integer overflow in expression

jmamma, 2019-05-06:
@yatli just noticed these warnings. can you look in to this?

```/Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/MCLMemory.h:19:55: warning: integer overflow in expression [-Woverflow]
 #define BANK1_MD_TRACKS_START BANK1_SYSEX2_DATA_START + SYSEX2_DATA_LEN
                                                       ^
/Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/MDTrack.h:51:58: note: in expansion of macro 'BANK1_MD_TRACKS_START'
                      public Bank1Object<MDTrackLight, 0, BANK1_MD_TRACKS_START> {
                                                          ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/A4Track.h:8:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/Grid.h:7,
                 from /Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/MCL.h:25,
                 from /Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/GridWritePage.cpp:2:
/Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/MDTrack.h:51:79: warning: overflow in constant expression [-fpermissive]
                      public Bank1Object<MDTrackLight, 0, BANK1_MD_TRACKS_START> {
                                                                               ^
/Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/MDTrack.h:51:79: note: in template argument for type 'long unsigned int' ```

yatli, 2019-05-06:
Running a home-made probe script I get:
```
============> Build complete.

Name                           Value
----                           -----
$NUM_A4_TRACKS                 4
$NUM_EXT_TRACKS                4
$NUM_LFO_TRACKS                4
$NUM_MD_TRACKS                 16
$NUM_TRACKS                    20
$RX_BUF_SIZE                   128
$UART2_RX_BUFFER_LEN           128
$MD_TRACK_LEN                  501
$A4_TRACK_LEN                  1742
$UART2_TX_BUFFER_LEN           3072
$UART1_TX_BUFFER_LEN           3072
$TX_BUF_SIZE                   3072
$SYSEX1_DATA_LEN               6192
$BANK1_UART2_RX_BUFFER_START   11904
$BANK1_UART2_TX_BUFFER_START   12032
$BANK1_SYSEX1_DATA_START       15104
$SYSEX2_DATA_LEN               16384
$BANK1_SYSEX2_DATA_START       21296
$BANK1_MD_TRACKS_START         37680
$BANK1_FILE_ENTRIES_START      37680
$BANK1_A4_TRACKS_START         37680
$BANK1_FILE_ENTRIES_END        41776
```

Doesn't look like it's overflowing, if it's uint32_t.

yatli, 2019-05-07:
Maybe plain integer literals are signed int16_t

yatli, 2019-05-07:
Note, there's a problem that MD_TRACK_LEN and A4_TRACK_LEN are not taken into account when calculating the higher addresses. Nevertheless,
```
41776+(501*16) + (1742*4)=56760
```
Still looking good. That's 86.6% utilization.

yatli, 2019-05-07:
The warnings are eliminated by adding 'UL' suffix to the literals.

commit 7d16ee4f (2019-05-07): fix #19
closed 2019-05-08 by yatli