jmamma/MCL #22: graphics assets

yatli, 2019-05-10:
@jmamma do you think it's a good idea to add graphics to the UI? I'm thinking about making a few icons, and improve the UI controls (possibly extract some common bits and build our own UI toolkit).

jmamma, 2019-05-10:
Sure. I've thought about this, at least for menus.

The Menu system already has screen space reserved for icons. Adding them in would be trivial.

What did you have in mind for improving the UI ?

yatli, 2019-05-10:
1. Would love to see an improved sequencer interface
2. Value visualization -- those `FLT` abbreviations alone are a little bit heavy for me when I'm not in programmer mode. :)  Elektron does this by drawing the knobs, filter slops etc. as live icons.
3. Utilize knob 3 & 4 in grid page, probably one for range selection without holding the menu button
4. Meta comments about the pages -- the UI might be more intuitive if the spatial arrangement correlates with the physical layout, for example:

[image]

jmamma, 2019-05-10:
cool. I think we're on the same track here.

The ideas you've mentioned have all been in my mind at some point ;-)

yatli, 2019-05-10:
oh and the page switch menu could use some icons

yatli, 2019-05-10:
3 tracks on screen might not be a good idea... too few pixels and it's visually distracting.
But I do think 2 tracks will be helpful, with one button "Track A/B Switch", and a knob to indicate track nr.

jmamma, 2019-05-10:
- Having labels for the button function would be useful. Not sure if we can cram everything on the screen without it looking cluttered.

- I like the idea of showing circular encoders on screen instead of numerical values (where applicable)

There's loads of ideas here that we can explore.

yatli, 2019-05-10:
> There's loads of ideas here that we can explore.

YES.

jmamma, 2019-05-10:
> Edit: track selection is actually a pain point for me in the current version. I have to exit editing mode, dial the right track on MD and then come back.

Yeah it's not perfect. But the trigger interface/exploit needs to be switched off if the user is detected changing tracks. Exploit breaks a few seconds after the display is updated. It made sense to train the user to switch tracks outside of the sequencer page to prevent the exploit from breaking.

yatli, 2019-05-10:
my idea is that we could change track on MC not MD so we can programmatically exit/re-enter the hacked TI.

jmamma, 2019-05-10:
oh yeah, that would work. Hold down shift2 and then select trigger?

I also wanted a way of adjust length using the trigger buttons

yatli, 2019-05-10:
actually... entering sequencer from grid: I always thought I'd end up editing the highlighted cell instead of the active MD track! :D

yatli, 2019-05-10:
> oh yeah, that would work. Hold down shift2 and then select trigger?

That'll do for me.

My other idea is, keep it consistent with the grid page that we have X/Y navigation even in sequencer page. We can even show a minimap of the grid in the sequencer mode

jmamma, 2019-05-10:
> My other idea is, keep it consistent with the grid page that we have X/Y navigation even in sequencer page. We can even show a minimap of the grid in the sequencer mode
^^^

Yeah this could be a better implementation and allow intuitive access to the ext sequencer tracks

jmamma, 2019-05-10:
> This is actually useful for that you can navigate both "horizontally in the grid" (track switching) and "vertically in the grid"(pattern switching)

Regarding y axis, are you suggesting that the sequencer pages don't edit the currently loaded tracks (workspace), but instead edit the selected tracks based on x,y location?

yatli, 2019-05-10:
yes, kind of line online editing vs. offline.

yatli, 2019-05-10:
> need a way to differentiate workspace from grid space

how about, say if our coordinates x,y happen to land on the current workspace, the playback cursors show up?

jmamma, 2019-05-10:
If you can come up with a workflow that makes sense. ability to edit grid positions, but also intuitively edit the live sequence, then i'm up for it. This would presumably changes the way saving tracks occurs too.

yatli, 2019-05-10:
Yeah.
I'll think about this now as I walk home from work. :)

jmamma, 2019-05-10:
> how about, say if our coordinates x,y happen to land on the current workspace, the playback cursors show up?

I think in a live performance you dont want to be looking for the current track to edit the step sequence.

Loading tracks to edit them kind of makes sense at the moment. Thinking about this some more,  I'm not sure how useful offline editing is.

jmamma, 2019-05-10:
I'm out for the night. 

Have a good 🚶

jmamma, 2019-05-24:
5b2281092d0b13c2b773d8a9c6d8ce59cd16cb10 in dev addresses track selection and pattern length

yatli, 2019-05-24:
wow, amazing!
I'm gonna try this out!

closed 2019-11-04 by yatli