jmamma/MCL #24: Let's optimize IS_BIT_SET

jmamma, 2019-05-28:
In CommonTools/helpers.h

We have the following macros.

/** 64-bit macros. **/
#define SET_BIT64(port, bit)   ((port) |= (uint64_t)(1ULL << (bit)))
#define CLEAR_BIT64(port, bit) ((port) &= (uint64_t)~(1ULL << (bit)))
#define TOGGLE_BIT64(port, bit) ((port) ^= (uint64_t)(1ULL << (bit)))
#define IS_BIT_SET64(port, bit) (((port) & (uint64_t)(1ULL << (bit))) ? 1 : 0)
#define IS_BIT_CLEAR64(port, bit) (((port) & (uint64_t)(1ULL << (bit))) == 0 ? 1 : 0)

IS_BIT_SET64 is used heavily in the sequencer ISR routines.

According to avr forum, the 32bit variant is extremely inefficient.

https://www.avrfreaks.net/forum/fast-nth-bit-set-long-code

@yatli how are your bit manipulation skills?

We can use https://godbolt.org/ to compare instruction length of various implementations

yatli, 2019-05-28:
ha!
Challenge accepted.

yatli, 2019-05-28:
```C
#define SET_BIT64(port, bit)   ((port) |= (uint64_t)(1ULL << (bit)))
```
Why is uint64_t needed on a port? Aren't they 8bit?

jmamma, 2019-05-28:
ignore the variable names (copy pasta), it's only used for the 64Bit masks that are used to represent pattern/track steps.

jmamma, 2019-05-28:
I had a play around with https://godbolt.org/ earlier. It seems whenever we do a shift, an assembly instruction to a subroutine is called, so it's not possible to count the amount of instructions (afaik). 

Probably need to find another way to bench these.

yatli, 2019-05-28:
shifting a long would require the `__ashldiX` family, but 8bit shifting can be done with `LSL` and `LSR` instructions.

```c
void FN_SET_BIT64(uint64_t& target, uint8_t bit)
{
    uint8_t* pbit = reinterpret_cast<uint8_t*>(&target)+(bit >>3);
    uint8_t mask = bit & 0x7;
    *pbit |= mask;
}
```
```AVR
FN_SET_BIT64(unsigned long long&, unsigned char):
        mov r30,r22
        ldi r31,lo8(0)
        asr r31
        ror r30
        asr r31
        ror r30
        asr r31
        ror r30
        add r30,r24
        adc r31,r25
        andi r22,lo8(7)
        ld r24,Z
        or r24,r22
        st Z,r24
        ret
```

yatli, 2019-05-28:
at this point, we should decide whether to make a lookup table to translate the rotations.

yatli, 2019-05-28:
alright. a better version:

```C++
void FN_SET_BIT64(uint64_t& target, uint8_t bit)
{
    uint8_t* pbit = reinterpret_cast<uint8_t*>(&target)+(bit / 8); // <----------- that.
    uint8_t mask = bit & 0x7;
    *pbit |= mask;
}
```

```
FN_SET_BIT64(unsigned long long&, unsigned char):
        mov r18,r22
        lsr r18
        lsr r18
        lsr r18
        mov r30,r24
        mov r31,r25
        add r30,r18
        adc r31,__zero_reg__
        andi r22,lo8(7)
        ld r24,Z
        or r24,r22
        st Z,r24
        ret
```

yatli, 2019-05-28:
cleaning up.

```C++
#include <stdint.h>
#include <avr/pgmspace.h>

const uint8_t bitmasks[] PROGMEM = {0x1, 0x2, 0x4, 0x8, 0x10, 0x20, 0x40, 0x80};

#define SET_BIT64(target, bit) do{((uint8_t*)(&target))[bit / 8U] |= bitmasks[bit & '\07'];}while(false)

void FN_SET_BIT64(uint64_t& target, uint8_t bit)
{
    SET_BIT64(target,bit);
}
```

yatli, 2019-05-28:
The conversion to uint8_t is critical:

```C++
#define SET_BIT64(target, bit)            do{((uint8_t*)(&target))[(uint8_t)bit / 8U] |=  _bvmasks[(uint8_t)bit & '\07'];}while(false)
#define CLEAR_BIT64(target, bit)          do{((uint8_t*)(&target))[(uint8_t)bit / 8U] &= _ibvmasks[(uint8_t)bit & '\07'];}while(false)
#define TOGGLE_BIT64(target, bit)         do{((uint8_t*)(&target))[(uint8_t)bit / 8U] ^=  _bvmasks[(uint8_t)bit & '\07'];}while(false)
#define IS_BIT_SET64(target, bit)     (bool)(((uint8_t*)(&target))[(uint8_t)bit / 8U] &   _bvmasks[(uint8_t)bit & '\07'])
#define IS_BIT_CLEAR64(target, bit) (!(bool)(((uint8_t*)(&target))[(uint8_t)bit / 8U] &   _bvmasks[(uint8_t)bit & '\07']))
```

yatli, 2019-05-28:
current stats:

```
Sketch uses 186440 bytes (73%) of program storage space. Maximum is 253952 bytes.
Global variables use 29729 bytes (52%) of dynamic memory, leaving 27102 bytes for local variables. Maximum is 56831
 bytes.
============> Build complete.

Name                           Value
----                           -----
$NUM_LFO_TRACKS                4
$NUM_EXT_TRACKS                4
$NUM_A4_TRACKS                 4
$NUM_MD_TRACKS                 16
$NUM_TRACKS                    20
$RX_BUF_SIZE                   128
$UART2_RX_BUFFER_LEN           128
$UART1_RX_BUFFER_LEN           128
$MD_TRACK_LEN                  501
$A4_TRACK_LEN                  1742
$UART2_TX_BUFFER_LEN           3072
$UART1_TX_BUFFER_LEN           3072
$TX_BUF_SIZE                   3072
$SYSEX1_DATA_LEN               6192
$BANK1_UART1_RX_BUFFER_START   8704
$BANK1_UART1_TX_BUFFER_START   8832
$BANK1_UART2_RX_BUFFER_START   11904
$BANK1_UART2_TX_BUFFER_START   12032
$BANK1_SYSEX1_DATA_START       15104
$SYSEX2_DATA_LEN               16384
$BANK1_SYSEX2_DATA_START       21296
$BANK1_MD_TRACKS_START         37680
$BANK1_A4_TRACKS_START         45696
$BANK1_FILE_ENTRIES_START      52664
$BANK1_FILE_ENTRIES_END        56760
```

yatli, 2019-05-28:
PR is up: #25

yatli, 2019-05-28:
`_BV32` also optimized -- however the invocations just involve constants so the optimization does not have any effect on the output.

yatli, 2019-05-28:
sleep mode.
good night :)

jmamma, 2019-05-29:
Excellent. Thanks for this work.

The review looks good. I will test tonight and then merge.

yatli, 2019-05-30:
#25 is now merged.

closed 2019-05-30 by yatli