jmamma/MCL #34: Alternative chassis design RFC

yatli, 2019-08-11:
I'm working on a metal chassis design

[image]

Basically, I'd like to make it low-cost, durable and mountable. The current design consists of two parts, each can be produced with sheetmetal (304, 6061 etc.)

Potential problems:

1. The screw tap hole "ears" are less than optimal. May consider bend the whole edge instead of cutting those 6 ears.
2. Screen height?

yatli, 2019-08-11:
Back:

[image]

yatli, 2019-08-11:
Close-in of one of the ears (which might be a problem, cuts finger, etc.):

[image]

yatli, 2019-08-11:
Some by-product that might be useful:

[image]

jmamma, 2019-08-12:
Woah. Cool. That 3D model is excellent.

The base of the screen pcb should be approximately 6mm above the height of the mainboard pcb (ignoring any components).

yatli, 2019-08-13:
Adjusting the design -- should be 1mm taller 😅
Could I upload the board 3D model to this repo?

jmamma, 2019-08-13:
Forgot to say. https://github.com/jmamma/MegaCommand_Design is the hardware repo.

Any hardware related designs should be uploaded there.

yatli, 2019-08-14:
Wilco!

closed 2019-08-14 by yatli