jmamma/MCL #36: Transmit sysex dumps from/to sdcard.

yatli, 2019-10-13:
- [ ] Pattern dumps
- [ ] Sample dumps (MIDI SDS)

yatli, 2019-10-13:
tracking: #35

jmamma, 2019-10-14:
Regarding Sample Dumps.

The idea was to have a Sample Manager page, with basic WAV editing features. A lot of that code is already implemented in the WavDesigner.

Also, to receive samples from the MD you need to use the undocumented SYSEX (macro) commands to navigate the menus and transmit sample from the SampleManager menu.

jmamma, 2019-10-14:
Rather than storing all snapshot data in a single .snp binary. I'm wondering if it would be better to organise snapshots via directories. Each directory would be a single snapshot and hold 128 patterns 64 kits + 8 globals etc.

Then the Snapshot page can be a generalised MD Sysex dump page.  You could choose to restore an entire directory, or just single files.

yatli, 2019-10-14:
The .snp binary is a direct representation of MC data structures and could be more convenient than serializing/deserializing syx files.

That said, having the flexibility of choosing individual elements is very useful. Luckily because the data structures are fixed-size, we can simply seek and access.

I'm thinking of this as, separating .snp and .syx as two features -- .syx involves additional parsing, checking the header (pattern dumps have different header than SDS dumps), entering the MD sysex menu etc, while .snp page simply stop playback and "hack" the content in/out.

yatli, 2020-01-20:
@jmamma time to do sample manager?
I've got command 0x33 for querying sample names.
We can use the name as a hash index that can be back-referenced to an exact WAV file in SD card.

yatli, 2020-03-14:
Tracking: #104

closed 2020-03-14 by yatli