jmamma/MCL #37: FileBrowser Improvements

jmamma, 2019-10-14:
The original idea was to have a file browser class that could be easily extended for use amongst different pages. Having a common base class would save on program size and memory requirements.

The difficulty here is that the FileBrowser often requires additional GUI functionality depending on its context.
For example the SoundBrowser page and LoadProject page both require reading the SD card file system. But have different functional requirements.

The current FileBrowser class allows additional GUI elements to be activated, through the use of boolean flags. The ability to traverse parent directories, to Save or Create new files are accessible by appending faux menu items before the list of files of the cwd.

There might be a better way to do this. Dynamic menu items? Function callbacks?

jmamma, 2019-10-14:
Rather than appending GUI elements to the file system list. Could use the first half of the screen to display an options menu, when Shift2  is pressed. This could then display page functions like [ "Save", "New", "Delete", "Rename"] etc.

yatli, 2019-10-14:
> when Shift2 is pressed

... or use encoder 3/4?
like, encoder 1/2 --> right pane file navigation
encoder 3/4 --> left pane menu items

jmamma, 2019-10-15:
I

> > when Shift2 is pressed
> 
> ... or use encoder 3/4?
> like, encoder 1/2 --> right pane file navigation
> encoder 3/4 --> left pane menu items

I wanted to get users to predict that shift2 may access a hidden sub menu on most pages.
For example, the slot menu is accessible from the GridPage by pressing shift 2.

yatli, 2019-10-16:
I like the consistency with GridPage.

yatli, 2019-10-30:
Implemented, closing.

closed 2019-10-30 by yatli