jmamma/MCL #39: NewProjectPage should use the TextInputPage for project name manipulation

jmamma, 2019-10-14:


yatli, 2019-10-14:
I can do a SHIFT-encoder text input menu for better input experiences :)

jmamma, 2019-10-14:
You mean similar to Elektron. Char select window?

Could be better overall.

yatli, 2019-10-14:
yep, but without 4-way arrow keys, we'd better make a single-row interface (sliding left/right, which gives preview of neighbor chars)

jmamma, 2019-10-14:
2+ rows would make for faster navigation. 
If you're going to open up a new page for Char Select, might as well use up all of the screen space :-).

Only downside I see is that it wont be compatible with HD44780 display. Though I think it might be time to drop support for that display.

yatli, 2019-10-14:
I'll see if I can do it in an hour or two. ;)

jmamma, 2019-10-15:
Cool. Had a quick read through. 

I will upload tonight and test it out :rocket:

yatli, 2019-10-15:
#35 now addresses this issue.

yatli, 2019-10-15:
Closing this issue now.

closed 2019-10-15 by yatli