jmamma/MCL #40: FX Pages.

jmamma, 2019-10-14:
Pages for controller Delay + Reverb.

Also need an intuitive +musical way of loading delay + reverb + compressor + eq settings from the current row/slot

yatli, 2019-10-15:
Add a column (track) for FX?
Pretty much like what A4 does.

jmamma, 2019-10-15:
Perhaps.

I also contemplated allowing columns > 16 to host special track types.
One of which could be of FX type.
Another type could be MIDI allowing tempo changes and such.

The problem is, how do you load these tracks from the MD trigger interface when you're limited to 16 buttons.

A proposed solution would be to have a layered GRID.
We could put each device on it's own layer.

yatli, 2019-11-25:
Primary feature complete. Moving reminding work to backlog.

closed 2019-11-25 by yatli