jmamma/MCL #42: Improve SeqPages GUI.

jmamma, 2019-10-14:
Visually the step sequencer should be drawn using drawrect commands (similar to mixer and cue pages)

Functionally we may be able to improve on encoder+ button layout.

jmamma, 2019-10-15:
I think I already implemented track switching? 
Shift2 + MD Trig.

---

In any case. In addition to visual improvements. The GUI button mapping needs work. 

There are currently at least 4 configurations of button mappings which can be categorised as followed:

1. - Save/Write Pages 
2. - SequencerPages.
3. - Auxillary/PageSelect Pages
4. - FileBrowser + Menu Pages

An example of inconsistency across the above groups:

Auxillary pages are exited by pressing encoder buttons.
FileBrowser + Menu Pages are exited by pressing function buttons. Encoder buttons are used for forward action.
Sequencer Pages are exited by pressing their corresponding entry button, which is an encoder button.

yatli, 2019-10-16:
> I think I already implemented track switching?

Looks like I'm lagging behind again :)

yatli, 2019-10-16:
> There are currently at least 4 configurations of button mappings

Also `TextInputPage`, Encoders == confirm, Button{1,3,4} == Exit

> FileBrowser + Menu Pages are exited by pressing function buttons. Encoder buttons are used for forward action.

This has been a great UX for me. Also consistent with the text input page. I've somehow trained myself to recognize Button1 as "Back".

> Auxillary pages are exited by pressing encoder buttons.
> Sequencer Pages are exited by pressing their corresponding entry button, which is an encoder button.

A bit confusing for me -- there's no clear connection which sequencer page is associated to which encoder button. Some times I just recognize encoder1 as exit, in case it didn't work (switched from another), push again.

jmamma, 2019-10-16:
> Auxillary pages are exited by pressing encoder buttons.
Sequencer Pages are exited by pressing their corresponding entry button, which is an encoder button.

A bit confusing for me -- there's no clear connection which sequencer page is associated to which encoder button. Some times I just recognize encoder1 as exit, in case it didn't work (switched from another), push again.

The only solution here is to put all the sequencer pages in to the quick select page.

yatli, 2019-10-16:
Visual help example:

[image]

yatli, 2019-10-16:
Alternative:

[image]

yatli, 2019-10-16:
A more lightweight "visual cue" approach:

[image]

yatli, 2019-10-16:
"white" version:

[image]

jmamma, 2019-10-16:
I do like the Alternative design.

The lightweight version is more ergonomic but what information is it communicating to the user?

jmamma, 2019-10-16:
These modes can be enabled/disabled from system menu.

yatli, 2019-10-16:
The lightweight cues only tell user that the system will respond when pressing the hinted buttons.

jmamma, 2019-10-17:
@yatli when you have a chance.

Can you create a confirmation window using your Alert Popup.

🛑 YES or NO

Returns true or false.

yatli, 2019-10-18:
Done. See #35, https://github.com/jmamma/MIDICtrl20_MegaCommand/pull/35/commits/52fef1a0fd1911f8c6f0e63be6217e6f1c15e8e7

Usage:

```c++
  if (mcl_gui.wait_for_confirm("Confirm", "Save snapshot?"))
  {
    gfx.alert("Confirmed", "Save");
  }

```

jmamma, 2019-10-18:
We should change the default MCL font to the MD char_select_font. Also visible in the KIT menu.

Need to start drawing some glyphs.

jmamma, 2019-10-19:
> We should change the default MCL font to the MD char_select_font. Also visible in the KIT menu.
> 
> Need to start drawing some glyphs.

Hmm,  I tried creating two variants of this font. After all that I think I prefer the adafruit font.

yatli, 2019-10-19:
The story behind tomthumb font: https://robey.lag.net/2010/01/23/tiny-monospace-font.html

Good old time.

jmamma, 2019-10-19:
The elektron 'small' font pre-dates tom thumb by 10+ years. I made some minor modification to tomthumb to match.

yatli, 2019-11-25:
Implemented. Closing now.

closed 2019-11-25 by yatli