jmamma/MCL #48: Quickselect behavior

yatli, 2019-10-18:
Currently when pressing SHIFT1, the quick select page pops up, and we can use ENC1 to select pages.
I'm thinking of making SHIFT1 more like alt-tab, that is, when entering the quick select page, auto advance to the next entry.
We can still use the rotary to scroll through the pages.

jmamma, 2019-10-18:
Are you aware that the trigger interface is active in the Quick Select page?

yatli, 2019-10-19:
No (⊙o⊙)

jmamma, 2019-10-19:
Maybe we should activate LED2 when trig interface is enabled.

Anyway, the idea is that you can navigate to the pages very quickly using the MD trigger buttons.

jmamma, 2019-10-28:
@yatli I have an idea.

To exit a page accessed from Quick Select. 

1) Re-Enter quick select menu (hold shift1)
2) Press Save.

This will free up encoder buttons in page select pages for fast rotate.

Also, we should update the PageSelect GUI and have an Exit Icon top left.
Other icons could be used to identify pages, as opposed to just text.

jmamma, 2019-10-28:
In addition I think Sequencer Pages should be accessible by entering Page Select + Encoder button.

This would keep the exiting "feel" of accessing the sequencer pages.

yatli, 2019-10-28:
> Re-Enter quick select menu (hold shift1)
> Press Save.

This is a feasible option. Though I'm a bit hesitating on adding more key combos. We have S1+S2, Save+Write. Adding a vertical Save + S1 could be heavy to learn for the users.
Can we also put the grid and the setting menu into Quick Select?
Then we don't need an exit button at all.

yatli, 2019-10-29:
Combining the ideas:

- All pages are accessible through [S1] + ENC1
  - To ensure consistency, draw the menu entries horizontally (ENC1 is X)
  - Though the menu is 2-layer, rotary scrolls first through the 2nd level entries, when exhausted, jump to the next 1st level entry.
- [SAVE] + [S1] goes to the first page -- the GRID
- [WRITE] + [S1] goes to the last page -- the CONFIG
- First of each major mode page is numbered sequentially, and accessible with [S1] + TRIG
- [S1] + ENC BUTTON = quick minor mode selection

Major/Minor mode maps:

```
T1. GRID
T2. SEQ
  -- NOTES
    [SAVE] MD/EXT
  -- REC
    [SAVE] RTRK/RLCK
  -- LOCKS
    [SAVE] A/B
  -- CHROMA
    [SAVE] PLAY/REC
T3. MIXER
  -- MIX
  -- ROUTE
  -- MASTER VOL
T4. SOUND
  -- SOUND MANAGER
  -- KIT MANAGER
  -- SD DRIVE
  -- SAMPLES
T5. SYNTH
  -- OSC1
  -- OSC2
  -- OSC3
  -- TX
T6. FX
  -- DYNAMICS
    [SAVE] A/B
  -- EQ
    [SAVE] A/B
  -- REVERB
    [SAVE] A/B
  -- DELAY
    [SAVE] A/B
T7. RAM MACHINE
  -- R1
  -- R2
T8. CONFIG
```

yatli, 2019-10-29:
One conflicting mapping is SeqExtStep resolution switch (S1 + S2), when we put it in the quick selection menu, we need to change the key combo.

jmamma, 2019-10-29:
> One conflicting mapping is SeqExtStep resolution switch (S1 + S2), when we put it in the quick selection menu, we need to change the key combo.

Yes, I had a plan that most pages would feature quick select. Sequence pages could use this to set track resolution and other playback options.

yatli, 2019-10-29:
Global sense of the input channels:
```
[SAVE] = ESCAPE
  -- QUIT
  -- TOGGLE SECTION
  -- COMMIT WORKSPACE EDITS TO GRID
[S1] = NON-INVASIVE POPUP MENUS
  -- QUICK SELECT
  -- QUICK EDIT PANE
[WRITE] = OVERWRITE
  -- CHAIN WRITE TO WORKSPACE
  -- CLEAR PATTERN/LOCKS
  -- CLEAR TEXT
[S2] = INVASIVE SPECIAL
  -- GRID MANIPULATIONS
  -- TEXT CASE SWITCH
  -- FILE MANAGER OPERATIONS
  -- SEQ ALL TRACK MODE
[ENC1] = X
[ENC2] = Y
```

jmamma, 2019-10-29:
"Can we also put the grid and the setting menu into Quick Select?"
That would resolve the additional button combination. And probably makes sense.

> Combining the ideas:

I'm still not convinced by the 2-layer quickselect menu. I think it's fine to group pages logically. But I prefer the 1-1 button mapping ( 1 trig per page)

I guess we need to test out both implementations and compare.

```
SAVE] + [S1] goes to the first page -- the GRID
[WRITE] + [S1] goes to the last page -- the CONFIG
```

Probably overkill. I still like having Global Settings on Save + Write. 
The plan was to have system menus accessible in pages. For example on RAMPage, Save + Write would open the System -> Auxillary -> RamPage menu.

yatli, 2019-10-29:
> But I prefer the 1-1 button mapping ( 1 trig per page)

Yeah I was trying to assign numbers like this, but then I run out of trigs.

yatli, 2019-10-29:
[SAVE] + [S1] goes to the GRID serves as the EXIT function.

> The plan was to have system menus accessible in pages. For example on RAMPage, Save + Write would open the System -> Auxillary -> RamPage menu.

I like this idea.

jmamma, 2019-10-29:
> but then I run out of trigs.

So we move any non-realtime pages to System/Global settings menu.

1) KIT MANAGER
2) SD DRIVE
3) SAMPLES
4) Wav Designer.

jmamma, 2019-10-29:
There's also 2 LFO pages coming :-)

yatli, 2019-10-29:
> 5. Loudness

If this won't kill the TI, then yes, we can do it from MD.
I'm updating the mapping comment.

jmamma, 2019-10-29:
Confusion. What is MASTER VOL ?

Currently there is the Loudness Page which is used to normalise track volumes by bringing up the VOL of each track.

> "If this won't kill the TI, then yes, we can do it from MD."

yatli, 2019-10-29:
> What is MASTER VOL

I was referring to LOUDNESS. Are they the same thing?

> If this won't kill the TI, then yes, we can do it from MD

I was thinking, if LOUDNESS == MASTER VOL, then we can drop this and use the MD knob.
I think I misunderstand this one.

yatli, 2019-10-29:
I've updated the mapping, with LFO counted, sound pages removed, wav designer out, we now have 14 trigs assigned. :)

jmamma, 2019-10-29:
Yeah, so loudness page iterates over each track and boosts the VOL by a percentage amount.
It also adjusts sequencer locks to VOL and LFOD to VOL to compensate. 

Why is this useful?

The idea is that the maximum loudness of each track should be when LEV == 127. The relative volume of each track is set using VOL.

jmamma, 2019-10-29:
> I've updated the mapping, with LFO counted, sound pages removed, wav designer out, we now have 14 trigs assigned. :)

If we run out of trigs in the future we could have PageSelect A and PageSelect B. toggle between them.

yatli, 2019-10-29:
Yeah that would do. There're multiple ways to achieve this, like double-tapping a trig, or trig chords.

yatli, 2019-11-09:
3AM, after some soldering.

How about this:

For Seq pages:
- S1 is page select, not seq page index switch
- WRITE is clear track.
- SAVE is A/B switch.
- S2 is multi-function with a menu, like the GRID menu, release to confirm. Entries:
  - Seq page index switch. This means, by default, press and release will advance to next seq page.
  - Track selection. Encoder1/TI == automatically move to this entry.
  - Clear all tracks
  - Ext page resolution change. S1 + S2 == automatically move to this entry.

The menu is only displayed after a timeout, so one can quickly use key combinations without visual distraction.

yatli, 2019-11-25:
Implemented.

closed 2019-11-25 by yatli