jmamma/MCL #54: RAM machine sample chain

yatli, 2019-10-20:
We can leverage RAM machine to implement sample chains, similar to OctaTrack.
The idea is to have MC upload chained WAV to RAM machine, and load the metadata about slice positions & pitches.

Then we will have a few more interesting modes:

- Pre-sliced loops
- One-machine sample packs
- PCM synth!

jmamma, 2019-10-20:
Yes. do this.

yatli, 2019-10-20:
I'm wondering if uploading to RAM machine works around the lock up bug?

jmamma, 2019-10-20:
I dont think you can upload to RAM.
I know sending RAM doesn't work.

jmamma, 2019-10-20:
We need a routine to determine Sample Bank layout.
Use Macros to send each ROM slot . Only send enough sample data so MCL can detect if a sample is present or not, then proceed to next ROM slot. and repeat.

jmamma, 2019-10-20:
Jus had a quick look. 

Sending RAM from the MD works. 

Upload to RAM doesnt work. I've tried via MIDI as well. Specifying the slot number corresponding to RAM the MD does not ack. (You can test with WAV designer)

yatli, 2019-10-20:
Yeah I just tried with the sample manager, selecting RAM machine doesn't turn it into RECV mode

yatli, 2019-10-20:
> Specifying the slot number corresponding to RAM the MD does not ack

What's the slot number to RAM?
Did you try other (nonsense) slot numbers?

jmamma, 2019-10-20:
If you select ORG from SAMPLE Manager. and then send to RAM slot it throws an error "ERROR INVALID POSITION."

yatli, 2019-10-22:
I'm going to give it a try, to see how much of SDS protocol does MD support.

yatli, 2019-10-28:
Picking up this thread after the GUI rush now.. :)

yatli, 2019-10-29:
Observation:

On boot up, without entering the sample manager, the MD refuses to handshake with incoming SDS dumps.
Once a sample is transferred (and the device locked up), the MD continues to accept incoming SDS dumps even not in the sample manager.

So I'm having a theory now.. ;)

yatli, 2019-10-29:
The theory was not particularly useful.
Tried to send some `NAK`, `CANCEL` and `EOF`, MD just freezes for a while, but still locked up.
2AM and burning out on this, time to sleep.

jmamma, 2019-10-30:
> On boot up, without entering the sample manager, the MD refuses to handshake with incoming SDS dumps.

Hmm I wasn't aware of this. 

I remember doing some digging on this "GUI lockup" bug. The dependency on "MD sysex request" is interesting. You can send a device id request SYSEX message (http://www.somascape.org/midi/tech/spec.html#nusx_gi) and the bug doesn't appear, but an elektron status request gets the MD in to a specific state where it will lock up on sds rec.

If I recall correctly, you can observe some interesting "state" behaviour in the SYSEX receive menu. before and after receiving a sample.

jmamma, 2019-10-30:
I had a mitigation in place that would stop MCL from sending elektron style requests on startup. at least allowing you use the Wav Designer without the sample bug. But had to  abandon it, because receiving the current kit was a necessary for majority of the firmware functionality.

yatli, 2020-01-20:
Long detour, but striking back now!

jmamma, 2020-01-20:
Return of the Jedi, Master Yatli.

yatli, 2020-03-14:
Related: #104

closed 2023-08-01 by jmamma