jmamma/MCL #63: Standardise Page GUI.

jmamma, 2019-10-31:
- [ ] FXPages

- [ ] RAMPages

- [ ] SavePage

- [ ] WritePage

- [ ] Route Page

jmamma, 2019-10-31:
SeqPage::display()

Some of the code here such as printing labels should be made in to member function(s) and take string pointer as arg.

yatli, 2019-10-31:
About Save / Write page:

I'd like a graphical representation of the data flow, for example:

```
SAVE
              SEQ.png \
                              >----- GRID.png
              MD.png /
```

yatli, 2019-10-31:
> Some of the code here such as printing labels should be made in to member function(s) and take string pointer as arg.

You mean the info1 and info2 lines?

jmamma, 2019-10-31:
Yeah. Also, I think we might need some type of MCLGUIPage class parent of SeqPage.
So the draw_knob functions and other standard GUI elements can be shared to other page types.

jmamma, 2019-10-31:
> About Save / Write page:
> 
> I'd like a graphical representation of the data flow, for example:
> 
> ```
> SAVE
>               SEQ.png \
>                       >----- GRID.png
>               MD.png /
> ```

Are you talking about graphical representation of Pages + button combinations?

yatli, 2019-10-31:
> Are you talking about graphical representation of Pages + button combinations?

Actually I mean, "data is going from the SEQ to the GRID, and optionally, merging from MD"

yatli, 2019-10-31:
I just uploaded the latest HEAD with LFO pages.
Excellent work!

jmamma, 2019-10-31:
> I just uploaded the latest HEAD with LFO pages.
> Excellent work!

Pretty much got every feature I wanted in to the LFOPage. And they LFOs work better than I expected.

jmamma, 2019-10-31:
Also, have you seen the Trimetric Sprites:

https://github.com/jmamma/MIDICtrl20_MegaCommand/tree/dev/art/sprites/trimetric

I think they will look quite good when rendered on the OLED.

yatli, 2019-10-31:
Yes, and hence the "MD.png" thing :)

yatli, 2019-10-31:
> Some of the code here such as printing labels should be made in to member function(s) and take string pointer as arg.

Done.

jmamma, 2019-11-01:
> > Some of the code here such as printing labels should be made in to member function(s) and take string pointer as arg.
> 
> Done.

Nice work.

closed 2019-11-04 by yatli