jmamma/MCL #65: Clock goes up to 10000??

yatli, 2019-11-01:


jmamma, 2019-11-01:
What is the MD tempo ?

yatli, 2019-11-01:
MD tempo is 140

jmamma, 2019-11-01:
I can't replicate.

checkout Midi/MidiClock.h and main.cpp from 25fed62b87954d4ed0eb1ad341a4886442c2e6c9

This is the change before 110be3bca7b43702b35cac8a01f2aca6691594b6

and see if your problem goes away.

yatli, 2019-11-01:
I disconnected A4 and let it run for 20mins and I see no problems.

yatli, 2019-11-01:
Repro:

connect both MD and A4.
quickly swipe all the MD trigs.

jmamma, 2019-11-01:
which page? grid?

yatli, 2019-11-01:
yes, grid.

jmamma, 2019-11-01:
couldnt reproduce.

DEBUG MODE ?

stale object files?

yatli, 2019-11-01:
No DEBUGMODE, program size normal.
Doing a clean build now.

yatli, 2019-11-01:
repro 2: smash the A4 keyboard fast enough

yatli, 2019-11-01:
Also, SeqPtcPage seems not working, cannot use MD trigs

jmamma, 2019-11-01:
no deal here.

What is the MD CTRL CHAN ?

jmamma, 2019-11-01:
SYSTEM -> MACHINEDRUM -> CTRL CHAN:

yatli, 2019-11-01:
MACHINEDRUM -> CTRL CHAN = 12
TURBO1 = 8x
TURBO2 = 8x
CLK REC = MIDI 1
CLK SEND = MIDI 2
MIDI FWD = OFF

jmamma, 2019-11-01:
CTRL CHAN : INT (use MD trigger interface for chromatic page)
CTRL CHAN: (all other values):  Use MIDI2 for chromatic page.

yatli, 2019-11-01:
Can I use both?

jmamma, 2019-11-01:
https://www.overleaf.com/read/gsdhhxzpbgdx Do your A4 settings match those in the manual

jmamma, 2019-11-01:
> Can I use both?

I disabled this for the reason that when trig interface is off you can modulate the chromatic  track PARAMs freely. Trig interface will shutdown if it detects CC change.

jmamma, 2019-11-01:
it's late here. almost seems like a midi loop you have going on.

yatli, 2019-11-04:
if BPM is 100000, then diff_clock8 is 1.

yatli, 2019-11-04:
@jmamma I've completely disconnected MC, and connect MD midi out directly to A4 midi in.
I then observe the same behavior, midi clock messages flooded the channel.

So again, this is a problem of my MD, not MC. 😢

yatli, 2019-11-19:
I have a better understanding of the problem today.
Downloaded a serial monitor to MC, which forwards everything to USB.

Turn on internal clock on MD. PC starts to receive `F8` midi clock ticks:

```
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
......
```

Push and release the triggers rapidly, so that the trig midi events interleave with clock ticks:
```
24 00 F8 29 5F F8 F8 2B 5F 28 00 2D 00 F8 F8 F8
F8 24 5F 26 5F 2B 00 F8 2F 5F 29 00 F8 F8 F8 28
5F F8 F8 F8 2F 00 24 00 2B 5F F8 26 00 2D 5F 29
5F F8 F8 2D 00 F8 28 00 F8 F8 24 5F F8 2F 5F 2B
00 F8 26 5F 29 00 F8 F8 28 5F 2D 5F F8 F8 F8 2F
00 24 00 F8 26 00 29 5F 2B 5F F8 2D 00 F8 28 00
F8 F8 F8 99 24 5F F8 2F 5F F8 26 5F 29 00 2B 00
F8 F8 28 5F F8 2D 5F F8 24 00 2F 00 F8 26 00 2B
5F F8 29 5F 2D 00 F8 28 00 F8 F8 F8 24 5F 2F 5F
F8 2B 00 F8 29 00 26 5F F8 F8 28 5F 2D 5F F8 24
00 F8 2F 00 F8 2B 5F F8 26 00 29 5F 2D 00 F8 28
00 F8 F8 F8 2F 5F 24 5F F8 2B 00 F8 26 5F 29 00
F8 F8 F8 28 5F 2D 5F 2F 00 F8 F8 24 00 2B 5F 26
00 F8 29 5F F8 28 00 F8 2D 00 F8 2F 5F F8 24 5F
F8 2B 00 26 5F F8 29 00 F8 28 5F F8 F8 F8 24 00
F8 F8 29 5F 26 00 2D 5F F8 F8 28 00 2B 5F 2F 00
F8 F8 F8 24 5F F8 2D 00 F8 26 5F 29 00 F8 2F 5F
F8 28 5F 2B 00 F8 F8 F8 F8 24 00 26 00 29 5F 2D
5F F8 F8 2F 00 28 00 2B 5F F8 F8 24 5F F8 F8 29
00 26 5F 2D 00 F8 2F 5F F8 F8 28 5F F8 24 00 F8
2D 5F F8 29 5F F8 26 00 2F 00 F8 F8 28 00 F8 2D
00 F8 29 00 2B 00 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
```

So far so good.
However, IF the triggers include TRIG9(CH), then the pattern changes. F8 and 00 are flooding:
```
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8 F8
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
Clock ticks flow very fast, order of magnitude faster than original clock. Interleaved with 00s.

Very much like a midi loopback problem, but why channel 9? Is this related to MC global settings?

yatli, 2019-11-19:
Switch to GLOBAL 1, turn off all mappings in map editor, problem disappears.

...? 🤔

yatli, 2019-11-19:
The problem is correlated with CTRL base channel setting.
Turning CTRL base channel off, and problem disappears. But the trigs don't send out midi events.

yatli, 2019-11-19:
Also, as long as I turn off internal CTRL on MD, problem disappears -- the trigs do not send internal midi events to the tracks though.

Well isn't this a good thing. No need to hack to get access to the trigger interface. Just turn off internal CTRL.

yatli, 2019-11-20:
@jmamma comments? :)

yatli, 2019-11-20:
tried a little bit on https://github.com/jmamma/MIDICtrl20_MegaCommand/tree/mdexploit_no_local

interesting things:
- different tracks get different cc id
- different pages (synth/amp/routing) get different cc id

so in theory, MC can be aware of all the param tweaks.

bad things:
- press FN+LFO and MD dies. :( :( :(

yatli, 2019-11-20:
@jmamma I've narrowed down the problem.

It happens only on particular rows of one file!

https://1drv.ms/u/s!Aql6jSQH9uyZhPpGV1zEUUyGtvQTTw?e=JFghtc

ROW 018
ROW 019

First two tracks are SINE
All other tracks are GND
Kit name is BELL-CB

Chain load row 018, then press trig 9 (CH)

yatli, 2019-11-20:
I am so happy now.
I thought my MD was going to die. Seriously.

jmamma, 2019-11-20:
I will look at the project later tonight.

This kind of scenario usually arises during development. Change project spec or upload bad Kit to the MD, then accidentally save in project. MD state is warped from bad kit upload, and now you have saved this state in the MC, to be recalled.

--

I had a similar recent issue where sending a machine sound, that was set to Trigger itself caused the machinedrum to reliably freeze after a few steps.

jmamma, 2019-11-21:
Row 18 locks up my machine as well.

But only if I send the entire Kit to the MD [ Chain Menu + Shift 2 ]
Sending the individual tracks doesnt break it.

So it's likely something wrong with master fx for the kit. 

Strangely I can't open the LFO menu.

---

Any idea how long ago you saved these? I'm wondering what could have caused the corruption.

yatli, 2019-11-26:
I've just saved these patterns recently.
It's probably related to another issue, where setting the auto chain for a range doesn't always work.

The pattern is created as follow:
1. Chain a blank row to set every track to GND
2. Start creating some sounds on track 1/2
3. Punch in some sequence
4. SAVE, S2, save the working set to the row.
5. Select the next row, SAVE, S2
6. Make some variations
7. Save again.
8. Select the first row. Use the grid menu to set: LOOP=4, ROW=_next_row_, range = all tracks
9. Select the second row. Use the grid menu to set: LOOP=4, ROW=_first_row_, range = all tracks
10. Chain write the first row. Start playback.
11. Observe: the auto chain doesn't work for both tracks. Sequencer data get corrputed.
12. Workaround: Set auto-chain individually for track 1/2, not in a range.

jmamma, 2019-11-26:
Cool. I'll take a look 👁

jmamma, 2019-11-26:
Hmm I can't replicate. 

Do you have the A4 connected + peered? 

Can you send me the project

jmamma, 2019-11-26:
Also, is DEBUG mode off? Chain mode is not working in debug mode currently.

yatli, 2019-11-27:
DEBUG off.

A4 connected & peered. (Doesn't matter though. Had this problem back when I only connect MD)

yatli, 2019-11-27:
The minimal repro evil bits: https://1drv.ms/u/s!Aql6jSQH9uyZhPpHso3R7KQw1kpbsA?e=TQQHDm

yatli, 2019-11-27:
The ranged auto chain just goes wrong. You'll see that some cells do not change over.

jmamma, 2019-11-27:
By Ranged Auto Chain, you mean you've selected the entire ROW and set LOOP and ROW.

yatli, 2019-11-27:
Yes. Confirm.

jmamma, 2019-11-27:
FYI: the next transition step for a slot is

next_transitions[n] += len;

where len = chains[n].loops * mcl_seq.md_tracks[n].length;

For -- len will be 16 steps

jmamma, 2019-11-27:
Wont be able to look at your project until later tonight.

jmamma, 2019-11-27:
```The ranged auto chain just goes wrong. You'll see that some cells do not change over.```

It looks like theres an issue with the GND slots.

The two SN slots are behaving as expected.

Is this what you're seeing?

yatli, 2019-11-27:
In this particular case, yes.

But I saw similar cases running md808 patterns

jmamma, 2019-11-27:
I think they're different problems.

In your example. Connecting the A4 seems to resolve the problem.

jmamma, 2019-11-27:
Is it only happening on EMPTY -- ? Or other slots you're observed?

yatli, 2019-11-27:
On other slots. Creating seemingly random combinations of tracks.

jmamma, 2019-11-27:
okay. and Chain Mode: AUTO . Confirm ?

is this happening on any specific row numbers, or all row numbers?

yatli, 2019-11-27:
Confirm AUTO chain mode.

On md808 project it occurs on the first few rows (because there are 16 patterns in total)

On CHROMA.prj I observe this on row 18 19

yatli, 2019-11-27:
The simplest cause I can think of, is some tracks are treated as random chain.

yatli, 2019-11-27:
Otherwise something is very wrong, and different loop count is read on the same address.

jmamma, 2019-11-27:
There's an array 
mcl_actions.next_transitions[n] that has the next transition step for each slot.

After each transition event, the above array is updated, 

mcl_actions.calc_next_slot_transition(n);

Once all the nearest_steps are determined for each track, the closest event (mcl_actions.next_transition) is calculated by sorting the array.

mcl_actions.calc_next_transition();

Next time grid_task runs, it checks to see if it approaching mcl_actions.next_transition

All tracks that have mcl_actions.next_transitions[n] == mcl_actions.next_transition will get loaded and then the cycle repeats.

jmamma, 2019-11-27:
so either mcl_actions.calc_next_slot_transition(n); is wrong due to a bad load ?

or mcl_actions.calc_next_transition() is missing close events for some reason. could be a timing problem.

yatli, 2019-11-27:
Thanks for the info. I can help debug this when I get home tonight.

jmamma, 2019-11-27:
GridTask and MCLActions are the main pieces of code.

It's complicated, and probably needs a refactor.

yatli, 2019-11-27:
I don't think it's a timing problem, because if I chain the tracks one by one there's no problem.

jmamma, 2019-11-27:
prepare_next_chain(int row)  is called from the Chain Page, and is responsible for loading slots.
In addition, it will initialise the  mcl_actions.next_transitions[n]  array for the slots that are loaded.

On Midi STOP + START mcl_actions.next_transitions[] is also re-initialised. 

I notice on debug output that the initializaiton values are garbage for the A4 tracks (16-19).

```
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
0  **<---- track**
63  **<----- next_transition[n]**
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
1
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
2
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
3
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
4
47
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
5
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
6
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
7
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
8
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
9
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
10
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
11
loops == 0 exiting
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
12
63
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
13
63
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
14
63
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
15
63
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
16
**26**
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
17
**122**
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
18
**398**
func_call: void MCLActions::calc_next_slot_transition(uint8_t)
19
**22**
```

jmamma, 2019-11-27:
Forgot. there are two functions fo loading from the Write Page.

prepare_next_chain(..)

and

send_tracks_to_devices()

..

Send tracks to devices is the legacy format that sends the kit to the MD as well as setting up the chain data.
I suspect the problem lies here.

jmamma, 2019-11-27:
```
186 void MCLActions::write_tracks(int column, int row) {
187   DEBUG_PRINT_FN();
188   if ((mcl_cfg.chain_mode > 0) && (MidiClock.state == 2)) {
189     if (MD.currentKit != MD.kit.origPosition) {
190       MD.getBlockingKit(MD.currentKit);
191     }
192     // MD.saveCurrentKit(MD.currentKit);
193     // MD.getBlockingKit(MD.currentKit);
194     prepare_next_chain(row);
195     //  grid_task.run();
196     return;
197   }
198   MD.saveCurrentKit(MD.currentKit);
199   MD.getBlockingKit(MD.currentKit);
200 
201   send_tracks_to_devices(); <<-- when midi is stopped.
202 
203 }

jmamma, 2019-11-27:
Okay I think I know what's going on.

At least for the -- EMPTY Track types.

When a project is initialised, every slot is cleared using Grid::clear_slot(..)
What this does is write a GridTrack object to each slot position.

The GridTrack contains chain data, which is LOOPS and ROWs but doesn't have the track length (for object compatibility + legacy reasons track length is stored in the child Track Types). Thus, the track length is never initialised for EMPTY slots.

MCLActions::calc_next_slot_transition(..) uses track length in it's formula:
len = chains[n].loops * mcl_seq.md_tracks[n].length;
and
len = chains[n].loops * mcl_seq.ext_tracks[n - NUM_MD_TRACKS].length;

When the sequencer is running two functions load slots in to MCL memory.

MCLActions::prepare_next_chain(int row) and
GridTask ::run() 

Neither of these are initialising the track length for EMPTY slots.

[9 more replies]

closed 2020-06-08 by jmamma