jmamma/MCL #73: Trig interface latency

yatli, 2019-11-12:
Mainly observed in:
1. CHROMA mode
2. In step editing page, sometimes if I quickly press-n-release a trig, it does not register a note.

We may move note interface event detection to a background task.
This will allow more timely detection of the events. For example we can schedule a scan where we scan the buttons, or once every a few timer interrupt.

jmamma, 2019-11-12:
> In step editing page, sometimes if I quickly press-n-release a trig, it does not register a note.
Hmm weren't you seeing something like this a while back?

Note interface populates note_interfaces.notes[i] when a MD trig is pressed and MIDI note is received. It then adds an event to the GUI ring buffer, which will then be processed on next call of main loop(), before the display is updated.

yatli, 2019-11-12:
I observe this quite often. When it happens, I have to press longer for a note to register.

jmamma, 2019-11-12:
SeqStepPage.cpp

Step is immediately set if it is empty.
If there is a already a step, then it is cleared if the press and release are faster than TRIG_HOLD_TIME

```
313       if (!IS_BIT_SET64(active_track.pattern_mask, step)) {
314         uint8_t utiming = (seq_param2.cur + 0);
315         uint8_t condition = seq_param1.cur;
316 
317         active_track.conditional[step] = condition;
318         active_track.timing[step] = utiming; // upper
319         active_track.clear_step_locks(step);
320         SET_BIT64(active_track.pattern_mask, step);
321       } else {
322         DEBUG_PRINTLN("Trying to clear");
323         if (clock_diff(note_interface.note_hold, slowclock) < TRIG_HOLD_TIME) {
324           CLEAR_BIT64(active_track.pattern_mask, step);
325           active_track.conditional[step] = 0;
326           active_track.timing[step] = 12; // upper
327         }
328       }
```

jmamma, 2019-11-12:
You might need to add some toggleLed() to see what is going on.

yatli, 2019-11-12:
Good idea.
When we have an actual display device, why print to serial :)

jmamma, 2019-11-20:
@yatli can you investigate this one, now that the tempo issue has been traced.

yatli, 2019-11-21:
Sure.
I'm taking a mini vacation in the south. Will be back next Monday :)

yatli, 2019-11-25:
I can't repro this anymore -- automagically fixed?

jmamma, 2019-11-25:
I'm wondering if this was related to your bad row/kit situation.

yatli, 2019-11-25:
Could be. Closing for now.

jmamma, 2019-12-01:
Just ran in to this on track 16. GND machine.

jmamma, 2019-12-01:
reproduce. set track to 16 on MD. then enter seq_step_page.

jmamma, 2019-12-01:
Weird bugs, tracks [1-3] work fine. 4-16 appear to be affected.

jmamma, 2019-12-01:
Seems to only happen if I chain/write a blank row to the MD

jmamma, 2019-12-01:
Reproduce:

Chain blank slot on 16.
Set track to 16 MD.
Enter step edit.

jmamma, 2019-12-01:
Can't reproduce anymore after trying to enable debug more.

jmamma, 2019-12-01:
Okay. for the bug to appear you need to have selected Step Edited using the trig interface.

jmamma, 2019-12-01:
@yatli can you take a look at trying to reproduce ^

jmamma, 2019-12-15:
@yatli 

I managed to get this bug showing up in debug mode. 

345       if (!IS_BIT_SET64(active_track.pattern_mask, step)) {

Is returning false when it should be true.

I managed to dump the pattern mask 

0001001110000000000000000000000000000000000000000000000000000000
Trying to clear
10

^ trying to clear position 10. which is 0.

jmamma, 2019-12-15:
hmm maybe not. 
more verbose debug showing 

clear step
last_md_track = 14
step = 2

0010000000000000000000000000000000000000000000000000000000000000
Trying to clear

..

The step is set before the GUI event.

jmamma, 2019-12-15:
I think the bug is in loop();

jmamma, 2019-12-15:
confirmed. bug is in loop

jmamma, 2019-12-15:
206   if (seq_param1.hasChanged() || seq_param2.hasChanged() ||
207       seq_param4.hasChanged()) {

one of these is staying true.

jmamma, 2019-12-15:
fee56e88942d0fa7f2bc40341ff708205b46357e

Fixed ^

closed 2019-12-17 by yatli