jmamma/MCL #79: Feature Freeze and 2.50 Release.

jmamma, 2019-11-24:
Close last PR.
More testing.

Release date Dec 1st.

jmamma, 2019-11-24:
@yatli does this sound good to you?

yatli, 2019-11-24:
Oh yeah.
I'm setting up a tracker for the blocking issues: https://github.com/jmamma/MIDICtrl20_MegaCommand/milestone/1

jmamma, 2019-11-25:
Thanks. I've just updated the Changelog. Take a look to see if I missed anything.

yatli, 2019-11-25:
Changelog not checked in?

jmamma, 2019-11-29:
I've pushed back the release to the 7/12/2019.

yatli, 2019-11-30:
No rush.

jmamma, 2019-11-30:
The HD44780 code is actually very close.

Surprised how well we've managed to preserve functionality across the two displays.

jmamma, 2019-12-09:
@yatli 

I'm more or less satisfied with the state of the MCL 2.50

I've exhausted all the bugs I can find 🤞 

If you would like to do more testing of your own. Before release?

jmamma, 2019-12-09:
Documentation is done.

yatli, 2019-12-09:
Thanks! That's a lot of fixes!

I'm currently wired into the hack project. Please feel free to proceed on the release :)

yatli, 2019-12-11:
I'm wondering peoples' reaction on seeing Him.

:)

jmamma, 2019-12-17:
Just compiled final binaries.

Will test tonight, then tag and release.

jmamma, 2019-12-17:
Need to test these as well

Windows:
https://github.com/xinabox/xLoader
Mac:
http://paulkaplan.me/HexUploader/

yatli, 2019-12-17:
I can test the Win32 part.

jmamma, 2019-12-17:
[binary candidates.zip](https://github.com/jmamma/MIDICtrl20_MegaCommand/files/3971636/binary.candidates.zip)

yatli, 2019-12-17:
I'm trying to figure out how to successfully load the FW with xloader...

jmamma, 2019-12-17:
Maybe it doesnt work with the 2560. I just tried out the mac tool and you have specifically select the board type.

jmamma, 2019-12-17:
[binaries candidate 2.zip](https://github.com/jmamma/MIDICtrl20_MegaCommand/files/3972771/binaries.candidate.2.zip)

yatli, 2019-12-17:
let me hack together a working xloader.

jmamma, 2019-12-17:
can you try https://sourceforge.net/projects/bitburner/

yatli, 2019-12-17:
what's the command you use usually to upload?

jmamma, 2019-12-17:
${AVR_DIR}/bin/avrdude -C${AVR_DIR}/etc/avrdude.conf -v -V -patmega2560 -cwiring -P${DEV} -b115200 -D -Uflash:w:./main.hex

yatli, 2019-12-17:
the stock XLoader is missing the correct arguments for avrdude.
we'll have to edit avrdude.conf and devices.txt

jmamma, 2019-12-17:
ah

jmamma, 2019-12-17:
i'm having a git meltdown. cant auto merge dev in to master. conflicts

jmamma, 2019-12-17:
"git merge -X theirs dev" didn't know this trick.

jmamma, 2019-12-17:
```Justins-MacBook-Pro-3:megacommand jmammarella$ md5 main.hex
MD5 (main.hex) = 248c6b45e21d565c167a326238343299
Justins-MacBook-Pro-3:megacommand jmammarella$ md5 ~/Desktop/mcl_binaries/mcl_2_5_0_oled.hex 
MD5 (/Users/jmammarella/Desktop/mcl_binaries/mcl_2_5_0_oled.hex) = 248c6b45e21d565c167a326238343299
```
merge was good.

yatli, 2019-12-17:
you can `git diff dev` to check the merged result

yatli, 2019-12-17:
working command:
```
.\avrdude.exe -p atmega2560 -c wiring -D -P \\.\COM4 -b 115200 -U flash:w:"C:\Tools\XLoader V1.339\mcl_2_5_0_oled.hex":i
```

trying to hack it into the xloader config file

jmamma, 2019-12-17:
Can you set the avrdude.exe path to be the default location installed by arduino ide.

jmamma, 2019-12-17:
https://github.com/jmamma/MIDICtrl20_MegaCommand/releases/tag/untagged-decad224e317dcb19584

^ release draft

yatli, 2019-12-17:
- update avrdude.exe
- update libusb0.dll
- the format for devices.txt goes like: `name;cpu;programmer;baudrate`
- it's missing a `-D` flag
- simple code injection: `megacommand;atmega2560;wiring -D;115200;`
- bingo!

[image]

jmamma, 2019-12-17:
h4x0r ^

yatli, 2019-12-17:
Assembled uploader:

[XLoader.zip](https://github.com/jmamma/MIDICtrl20_MegaCommand/files/3972900/XLoader.zip)

jmamma, 2019-12-17:
"update avrdude.exe
update libusb0.dll"
^ what does this mean?

jmamma, 2019-12-17:
ah avrdude.exe was part of the archive. naughty.

jmamma, 2019-12-17:
where should we host it?

yatli, 2019-12-17:
well, here?

jmamma, 2019-12-17:
license ?

yatli, 2019-12-17:
because I've modified the devices config section -- the only device option is "megacommand"

yatli, 2019-12-17:
The license of XLoader:

```
Copyright (c) 2012, Geir Lunde
All rights reserved.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
DISCLAIMED. IN NO EVENT SHALL <COPYRIGHT HOLDER> BE LIABLE FOR ANY
DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
(INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND
ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
(INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
```

jmamma, 2019-12-17:
i need a lawyer.

jmamma, 2019-12-17:
i have windows 10 now. i will test.

yatli, 2019-12-17:
In my layman's understanding this is just a no-warranty clause

jmamma, 2019-12-17:
awesome. just tested. works great!

yatli, 2019-12-17:
awesome. big time!

jmamma, 2019-12-17:
We did it.

https://github.com/jmamma/MIDICtrl20_MegaCommand/releases/tag/2.50

jmamma, 2019-12-17:
If you're happy, i'll post it over to elektronauts

yatli, 2019-12-17:
sure!

jmamma, 2019-12-17:
Thanks for accompanying me on this journey Yateo. 

I had dreams of this firmware when I was performing live in 2010. I wrote the first minicommand sketch in 2012. It's taken 7 years to get this far. If only I could send a megacommand back in time.

😄

yatli, 2019-12-18:
I feel warm. :)

And time flies. 15 years ago when I started programming, I work on a miniature 6502 PDA with 160x80 LCD.

I did not know that after all these years, I'll be still enjoying 8 bit programming, with even smaller screens!

jmamma, 2019-12-19:
It's a shame JMamma & Yatli Enterprises doesn't have any money. 
Otherwise we would should have thrown a product launch party!

Just an fyi that I'll be taking a bit of a break from coding these holidays.

Will monitor the forums for bugs etc, but no active development for a little while.
If anything I will shift focus to music production and sound design for the new firmware.

yatli, 2019-12-19:
Now the engineering parts are ready, good luck and have fun with some music!

yatli, 2020-01-20:
Done, closing.

closed 2020-01-20 by yatli