jmamma/MCL #80: Update documentation for v2.5

yatli, 2019-11-24:


jmamma, 2019-11-25:
I have to admit, I have not been looking forward to this.

Setup an overleaf account and I will provide you with editor access.

https://www.overleaf.com/

jmamma, 2019-11-26:
I modified ./mc_display_mirror.py to take a screenshot when pressing '9'

I've gone through and uploaded a bunch of new screenshots 
https://github.com/jmamma/MCL-Documentation/tree/master/MCL%20Documentation/images/images_pending

yatli, 2019-11-26:
So cool.

Oh wait we got visual utiming now??

jmamma, 2019-11-26:
Of course. 💰Premium feature 💰

yatli, 2019-11-26:
MegaCommand:

> INSERT COIN <

yatli, 2019-11-26:
All right first things first. The doc repo is not compiling on my side.
I got TeXLive 2019 and I use latexmk -xelatex.
Can compile earlier revisions with no problem.

jmamma, 2019-11-26:
I just fixed two compile errors relating to special characters.

It's easier if you just work from overleaf.

https://www.overleaf.com/read/gsdhhxzpbgdx

We should be able to work collaboratively .

Just need your account email so I can give you editor access.

yatli, 2019-11-26:
Ok cool. Just realized that I do have a overleaf account: `yatao.li@live.com`

yatli, 2019-11-26:
I'm in 🙌

jmamma, 2019-11-26:
Thanks. It's good to have a different pair of 👀 on the documentation.

I'm going to take a look at the HD44780 display code.

yatli, 2019-11-26:
May be useful:
https://www.tablesgenerator.com/

yatli, 2019-11-26:
Half way there.
Renamed all pending images, and moving them out of pending on-demand.

yatli, 2019-11-27:
@jmamma some screenshots are still missing.

yatli, 2019-11-27:
Seq group ok.

jmamma, 2019-11-27:
@yatli which pages are missing?

The serial transfer code is currently located in GUI.display().

Any draw routines that bypass that code can't be captured currently.

yatli, 2019-11-27:
PTC-A4
PARAM-Edit
RLCK
Ext-Seq

jmamma, 2019-11-27:
I thought PARAM-Edit was there?
[image]

The others are missing.

yatli, 2019-11-27:
hah, I put it in the overview "Parameter Locks" -- no locks and no trigs registered, so doesn't match the text in the detailed section

yatli, 2019-11-27:
@jmamma can you do the ram page doc. You will describe it better I'm sure :)

jmamma, 2019-11-27:
sure

yatli, 2019-11-27:
I'll add a section about the workflow, and describe the difference between the GRID, the INTERNAL SEQUENCER and the MD SEQUENCER.

yatli, 2019-11-27:
Done, see "key concepts"

yatli, 2019-11-27:
Requesting to put my name in there :p

jmamma, 2019-11-27:
yes of course 🥇

jmamma, 2019-11-27:
@yatli 

Maybe we should have a Buttons Assignemnt chart under the Encoders Assignment for each page.

jmamma, 2019-11-27:
Also, can you sync your changes to github, incase you havent already.
Top left Menu -> Sync

Thanks.

yatli, 2019-11-27:
Makes sense.

Combine the two to save some space?

yatli, 2019-11-27:
Synchronized.

yatli, 2019-11-28:
Synchronized, now with button assignment table.

yatli, 2019-11-28:
Two left, the LFO page and the RAM page.

yatli, 2019-11-28:
Done LFO page skeleton.

closed 2019-12-10 by jmamma