jmamma/MCL #87: Black screen after creating a new project

yatli, 2019-11-27:


jmamma, 2019-11-27:
Probably just need to remove the else if:

 51   } else if (proj.project_loaded) {
 52     GUI.setPage(&grid_page);
 53     return true;
 54   }

jmamma, 2019-11-28:
```
func_call: bool Project::new_project(const char*)
Creating new project
Attempting to extend project file
extension succeeded, trying to close
Initializing project.. please wait
func_call: bool Project::write_header()
Writing project header
Write header success
func_call: bool MCLSysConfig::write_cfg()
Writing cfg
Write cfg okay
project created
func_call: bool Project::load_project(const char*)
Loading project
/qroject005.mcl
func_call: bool Project::check_project_version()
Check project version
func_call: bool MCLSysConfig::write_cfg()
Writing cfg
Write cfg okay
project loaded, setting page to grid
calling cleanup
Pushing page
```

```
 22 bool NewProjectPage::handleEvent(gui_event_t *event) {
 23   // don't handle any events
 24   again:
 25   if (mcl_gui.wait_for_input(newprj, "New Project:", sizeof(newprj))) {
 26       
 27     char full_path[sizeof(newprj) + 5] = {'\0'};
 28     strcat(full_path, "/");
 29     strcat(full_path, newprj);
 30     strcat(full_path, ".mcl");
 31   
 32     gfx.alert("PLEASE WAIT", "CREATING PROJECT");
 33 
 34     DEBUG_PRINTLN(full_path); 
 35     if (SD.exists(full_path)) {
 36       gfx.alert("ERROR", "PROJECT EXISTS");
 37       goto again;
 38     }
 39 
 40     bool ret = proj.new_project(full_path);
 41     if (ret) {
 42       if (proj.load_project(full_path)) {
 43         grid_page.reload_slot_models = false;
 44         DEBUG_PRINTLN("project loaded, setting page to grid");
 45         GUI.setPage(&grid_page);
 46         return true;
 47       } else {
 48         gfx.alert("ERROR", "SD ERROR");
 49         goto again;
 50       }
 51       return false;
 52     }
 53   } else if (proj.project_loaded) {
 54     GUI.setPage(&grid_page);
 55     return true;
 56   }
 57 }
```

Any ideas? Should be setting page to &grid_page

jmamma, 2019-11-28:
Ah, think i know what's going on.

yatli, 2019-11-28:
I'm still doing the toggleLed2() thing 😅

jmamma, 2019-11-28:
I had 

 17   gui_event_t event;
 18   handleEvent(&event);

in init()

```
455   void pushPage(Page *page) {
456     if (currentPage() == page) {
457       DEBUG_PRINTLN("can't push twice");
458       // can't push the same page twice in a row
459       return;
460     }
461     DEBUG_PRINTLN("Pushing page");
462     page->parent = this;
463     if (!page->isSetup) {
464       page->setup();
465       page->isSetup = true;
466     }
467     page->init();
468     page->redisplayPage();
469     page->show();
470     pageStack.push(page);
471   }
```

yatli, 2019-11-28:
oh ok I got it.

I just observed it as well. The second time a project is created, it returns to the Grid.

jmamma, 2019-11-28:
moving 
 17   gui_event_t event;
 18   handleEvent(&event);
to loop()

yatli, 2019-11-28:
Or just remove it?

jmamma, 2019-11-28:
needs to be there, because new project page doesnt work on card init otherwise.

jmamma, 2019-11-28:
i think it was only working in your patch because the button release was triggering the handleEvent

yatli, 2019-11-28:
aha. event injection. :)

jmamma, 2019-11-28:
hmm putting in loop doesnt work either.

yatli, 2019-11-28:
Don't do it in handleEvent() then.

jmamma, 2019-11-28:
yeah moving the whole thing  to loop now.

jmamma, 2019-11-28:
have you hacked the MD firmware yet, to run MCL natively?

yatli, 2019-11-28:
I felt dizzy today when I thought about this!

jmamma, 2019-11-28:
i think NewProjectPage should be removed, and just be a callback function in Project.cpp

jmamma, 2019-11-28:
textinputpage is doing all the GUI work here.

yatli, 2019-11-28:
[image]

HOW DOES THAT LOOK

jmamma, 2019-11-28:
what am I looking at 👀

yatli, 2019-11-28:
An emulator running the machinedrum firmware!

jmamma, 2019-11-28:
define "running" 😁

jmamma, 2019-11-28:
find me elvis.

yatli, 2019-11-28:
no I/O board, no DSP, just memory and registers

jmamma, 2019-11-28:
that's pretty cool. how far does the execution get? The MAME notes say that it relies on responses from the sub-chips to detect that it's running on the correct hardware.

yatli, 2019-11-28:
oh yeah. Some register tampering is required.

jmamma, 2019-11-28:
when you get some time, are you able to briefly document how you got this up and running,

yatli, 2019-11-28:
anytime. I can do it now actually

jmamma, 2019-11-28:
I wonder how easy it would be to get the MD LCD working.

yatli, 2019-11-28:
https://github.com/jmamma/MIDICtrl20_MegaCommand/issues/88

jmamma, 2019-11-28:
5bf1b232c9eca357f6fe94b5501b50f7f4f27399

^ This issue should now be fixed.

yatli, 2019-11-28:
> I wonder how easy it would be to get the MD LCD working.

The I/O board communicates with the CPU board with UART.

jmamma, 2019-11-28:
> #88

So we can install a megacommand between the CPU and IO board and hijack the display.

yatli, 2019-11-28:
maybe the I/O board just accepts high level drawing commands like, put a encoder here, print a string there, clear the screen etc.

well that's good enough already :)

jmamma, 2019-11-28:
Is there a microcontroller on the IO board?

yatli, 2019-11-28:
I'm not sure. Didn't really carefully analyze the internals :p

yatli, 2019-11-28:
@jmamma loaded the latest firmware, crash on create project

jmamma, 2019-11-28:
@yatli 

I can't replicate. NewProject creation works fine here. Both on out-of-box startup and from system menu.

Can you check your environment is clean.

yatli, 2019-11-29:
Sure. Trying a clean build now;

yatli, 2019-11-29:
Still crash. Checking SD card status

jmamma, 2019-11-29:
What's the symptom of the crash? firmware reboot?

Can you paste debug output.

yatli, 2019-11-29:
SD card ok. Hardware ok. I swapped in another MC and SD card and still crash.

yatli, 2019-11-29:
Trying a debug build

yatli, 2019-11-29:
```
func_call: void MidiSetup::cfg_ports()
Track sizes:
1742
1382
410
501
4
7
Pushing page
7
4
7
func_call: bool MenuPageBase::enter()
calling callback func
Pushing page
7
7
PLEASE WAIT
CREATING PROJECT
/rroject008.mcl
func_call: bool Project::new_project(const char*)
Creating new project
Attempting to extend project file
extension succeeded, trying to close
Initializing project.. please wait
```

No apparent error here

jmamma, 2019-11-29:
I just swapped in a new MC and no problems here.

yatli, 2019-11-29:
On crash led2 turns on (and survives the crash, keep on to the next boot)

jmamma, 2019-11-29:
Is the MD connected?

yatli, 2019-11-29:
No.

jmamma, 2019-11-29:
Can you drop back to an earlier commit and see if it still crashes?

yatli, 2019-11-29:
Adding debug info to Grid::clear_row: not hit.
Trying an earlier build now;

yatli, 2019-11-29:
roll back to HEAD^^ c7e56a3ebb4464c7f9e5f7a213264b77675b1b28
project create success + blackscreen

yatli, 2019-11-29:
https://1drv.ms/u/s!Aql6jSQH9uyZhPpIG4OXO20HkXxYgQ?e=jzdRAH

Can you try the build artifact. I suspect it's a toolchain problem.

jmamma, 2019-11-29:
Yeah crash on new project when I upload that.

jmamma, 2019-11-29:
204388
MD5 (main.hex) = c58e2eeae0bde3e2a381c89403c3fb9e

yatli, 2019-11-29:
I'm on `arduino.exe Version: 0.5.0 Commit: 3be2287`
Upgrading to 0.6 now

yatli, 2019-11-29:
uh oh still crash after upgrading to `arduino-cli 0.6`, `avr-gcc 7.3`

jmamma, 2019-11-29:
yeah I dont think it's that. I think it's probably a linking issue.

jmamma, 2019-11-29:
your binary is larger than mine.

yatli, 2019-11-29:
yeah. I use the arduino toolchain, could be a different build env.

jmamma, 2019-11-29:
I can compile using the arduino ide,

ill do that now.

[19 more replies]

closed 2019-11-29 by yatli