jmamma/MCL #88: Patch the MD firmware to fix the SDS lockup bug?

yatli, 2019-11-27:
I admit this is ambitious, but I've been looking for relevant materials for a while.
Most significantly: https://github.com/mamedev/mame/blob/master/src/mame/drivers/elektronmono.cpp

... documents the specs and address space layout for the coldfire m68k processor.

https://edgeemu.net/details-39651.htm

... provides decoded binary image of the firmware.

Now, IF the bootloader is separated from the actual firmware, we can simply hack the firmware image, re-encode sysex, and burn it in without worrying about bricking the device (just be careful about I/O).

Toolings:

- gcc-m68k is good
- needs better decompiler than just objdump, maybe: https://github.com/REDasmOrg/REDasm
  - I'll see if I can write a m68k extension for this.

yatli, 2019-11-27:
The first 1MB is the firmware code section.
Chop it out:
```
dd if=elektron_sps1-1uw_os1.63.bin of=os_code.bin bs=1024 count=1024
```

Disassemble it -- when running as OS, the base address should be 0x10000000:

```
m68k-linux-gnu-objdump --adjust-vma 0x10000000 -b binary -mm68k:5206e -D os_code.bin 
```

Note: better first find the entry point for the OS! The head of the firmware may be the bootloader!

jmamma, 2019-11-28:
Nice.

I was looking at the binary over the weekend (trying to find Elvis)

The bootloader text (hold function on power on) can be seen at 0x24EO

jmamma, 2019-11-28:
There's also another possibility of manipulating MDRAM in realtime by sending a corrupt Kit.

I'm guessing that the LFOD is not being ranged checked. So perhaps this is an entry point for manipulating certain memory addresses.

yatli, 2019-11-28:
> The bootloader text (hold function on power on) can be seen at 0x24EO

That sounds a little scary XD

yatli, 2019-11-28:
There must be some kind of critical jump that switches from the bootloader to the OS.

jmamma, 2019-11-28:
@yatli you might find this interesting.

This time last year I had a brief correspondence with the author of the skeleton driver.

```
Yes, about 4 years ago we merged the MESS project (which was applying the MAME core emulation to non-arcade machines) into MAME itself and we'd already been working on various synths and drum machines, e.g. https://www.youtube.com/watch?v=ulQo5SjNTaQ&frags=pl%2Cwn .  Most recently we got ROMs for the Akai/Roger Linn MPC-3000, which for reasons we don't understand is also the sound section of the SNK Hyper NeoGeo-64 arcade system.

With regard to the Elektron and Access gear, most of the initial reverse engineering (including dumping the ROMs and extracting clean ROM dumps from the factory firmware downloads) was actually done by some folks at a Norwegian synth repair shop.  We do plan to continue the emulation, but we have limited resources in terms of people who can and want to work on emulating MIDI stuff.  This is part of why we've been so busy recently converting MAME from a macro jungle to mostly standard C++14; not only does that make tooling more able to work with the code, hopefully it reduces some of the difficulty of learning what's going on.```

yatli, 2019-11-28:
FYI the instruction set manual:

[CFPRM.pdf](https://github.com/jmamma/MIDICtrl20_MegaCommand/files/3901798/CFPRM.pdf)

yatli, 2019-11-28:
> which for reasons we don't understand is also the sound section of the SNK Hyper NeoGeo-64 arcade system.

pro gaming consoles.

yatli, 2019-11-28:
`0x1BEE` starts to write commands to UART1.

yatli, 2019-11-30:
I've got REDasm partially working with coldfire. Need a few more days to bring up the full functionality.

yatli, 2019-12-02:
@jmamma A lot of auxiliary routines have been decoded.
The critical jump is located at 0x2110:

[image]

In summary, it jumps to the main memory start address 0x200000, after unpacking the firmware from 0x4000.

That's why the actual program is not immediately obvious to a disassembler.

yatli, 2019-12-02:
The actual unpacking routine, located at 0x1c4:

```C

undefined4 UNPACK_FIRMWARE(byte *src,byte *dst)

{
  uint uvar_D0;
  uint uVar1;
  uint uvar_D1;
  uint uvar_D4;
  int ivar_A0;
  byte *ptr_2;
  byte *ptr_1;
  
  src = src + 8;
  uvar_D1 = 0;
  uvar_D4 = 1;
  do {
    uvar_D1 = uvar_D1 * 2;
    if ((char)uvar_D1 == '\0') goto LAB_000002c8;
LAB_000001e2:
    ptr_1 = src;
    if ((uvar_D1 & 0x100) == 0) {
      do {
        ivar_A0 = 1;
        do {
          while( true ) {
            uvar_D1 = uvar_D1 * 2;
            if ((char)uvar_D1 == '\0') {
              ptr_2 = ptr_1 + 1;
              uvar_D1 = (uint)*ptr_1 * 2 + 1;
              uvar_D0 = (uint)*ptr_1 * 2 >> 8;
            }
            else {
              uvar_D0 = uvar_D1 >> 8 & 1;
              ptr_2 = ptr_1;
            }
            ivar_A0 = uvar_D0 + ivar_A0 * 2;
            uvar_D1 = uvar_D1 * 2;
            if ((char)uvar_D1 == '\0') break;
            ptr_1 = ptr_2;
            if ((((byte)(uvar_D1 >> 8) ^ 1) & 1) == 0) goto LAB_00000220;
          }
          ptr_1 = ptr_2 + 1;
          uvar_D1 = (uint)*ptr_2 * 2 + 1;
        } while ((((byte)(uvar_D1 >> 8) ^ 1) & 1) != 0);
LAB_00000220:
        if (ivar_A0 != 2) {
          ivar_A0 = ((uint)*ptr_1 - 0x300) + ivar_A0 * 0x100;
          if (ivar_A0 == -1) {
            return 0;
          }
          ptr_1 = ptr_1 + 1;
          uvar_D4 = ivar_A0 + 1;
        }
        uvar_D1 = uvar_D1 * 2;
        if ((char)uvar_D1 == '\0') {
          ptr_2 = ptr_1 + 1;
          uvar_D1 = (uint)*ptr_1 * 2 + 1;
          uvar_D0 = (uint)*ptr_1 * 2 >> 8;
        }
        else {
          uvar_D0 = uvar_D1 >> 8 & 1;
          ptr_2 = ptr_1;
        }
        uvar_D1 = uvar_D1 * 2;
        if ((char)uvar_D1 == '\0') {
          src = ptr_2 + 1;
          uvar_D1 = (uint)*ptr_2 * 2 + 1;
          uVar1 = (uint)*ptr_2 * 2 >> 8;
        }
        else {
          uVar1 = uvar_D1 >> 8 & 1;
          src = ptr_2;
        }
        ivar_A0 = uVar1 + uvar_D0 * 2;
        if (ivar_A0 == 0) {
          ivar_A0 = 1;
          do {
            while ((char)(uvar_D1 * 2) == '\0') {
              ptr_1 = src + 1;
              ivar_A0 = ((uint)*src * 2 >> 8) + ivar_A0 * 2;
              uvar_D1 = ((uint)*src * 2 + 1) * 2;
              src = ptr_1;
              if ((char)uvar_D1 != '\0') goto LAB_00000294;
LAB_00000328:
              src = ptr_1 + 1;
              uvar_D1 = (uint)*ptr_1 * 2 + 1;
              if ((((byte)(uvar_D1 >> 8) ^ 1) & 1) == 0) goto LAB_000002a2;
            }
            ivar_A0 = (uvar_D1 * 2 >> 8 & 1) + ivar_A0 * 2;
            uvar_D1 = uvar_D1 * 4;
            ptr_1 = src;
            if ((char)uvar_D1 == '\0') goto LAB_00000328;
LAB_00000294:
          } while ((((byte)(uvar_D1 >> 8) ^ 1) & 1) != 0);
LAB_000002a2:
          ivar_A0 = ivar_A0 + 2;
        }
        ptr_2 = dst + -uvar_D4;
        ivar_A0 = ivar_A0 - -(0xd00 < uvar_D4);
        *dst = *ptr_2;
        ptr_1 = dst + 1;
        do {
          ptr_2 = ptr_2 + 1;
          dst = ptr_1 + 1;
          *ptr_1 = *ptr_2;
          ivar_A0 = ivar_A0 + -1;
          ptr_1 = dst;
        } while (ivar_A0 != 0);
        uvar_D1 = uvar_D1 * 2;
        if ((char)uvar_D1 != '\0') goto LAB_000001e2;
LAB_000002c8:
        ptr_1 = src + 1;
        uvar_D1 = (uint)*src * 2 + 1;
        if ((char)((uint)*src * 2 >> 8) != '\0') break;
      } while( true );
    }
    src = ptr_1 + 1;
    *dst = *ptr_1;
    dst = dst + 1;
  } while( true );
}
```

yatli, 2019-12-02:
@jmamma does that look like anything you know?

I'm at least sure about one thing, it takes data from `src`, and writes to `dst`.

jmamma, 2019-12-02:
I assume the DSP chips needed to be loaded up with code on boot.

yatli, 2019-12-02:
Yep, done before the jump.

jmamma, 2019-12-02:
So you're trying to ascertain what all the code in the unpack routine is ?

The bottom half looks like it's detecting strings. *shrugs*

yatli, 2019-12-02:
I'm trying to run this piece of code on the host machine and extract the actual OS.

yatli, 2019-12-02:
With MAME I can run the code, but it doesn't have the correct address mapping, so no data written

jmamma, 2019-12-02:
Is it sample loading?

yatli, 2019-12-02:
It's a general purpose unpack routine.
Example usage:

```c
      iVar2 = UNPACK_FIRMWARE((byte *)&PTR_DAT_00004000,(byte *)0x200000);
      puVar5 = (undefined4 *)&DAT_01000000;
      if (iVar2 == 1) {
        PRINTLN("E: OS CHECKSUM\n");
        do {
                    /* WARNING: Do nothing block with infinite loop */
        } while( true );
      }
      do {
        puVar5 = puVar5 + 1;
        *puVar5 = 0;
        puVar5 = puVar5;
                    /* bzero internal SRAM 0-0x2000 */
      } while (puVar5 != (undefined4 *)0x1002000);
      goto 0x200000;
```

jmamma, 2019-12-02:
What do you mean 

 doesn't have the correct address mapping, so no data written

?

yatli, 2019-12-02:
ah.

so the actual address mapping on MD goes like this:
[image]

Such assignment is completely missing from the emulator. The emulator thinks all the addresses map to the ROM (and we cannot write to the ROM at 0x200000)

jmamma, 2019-12-02:
Did you just answer your own question :D ?

How do you go about configuring the emulator to write to RAM.

yatli, 2019-12-02:
Real OS image decode success. Size = 404766, checksum OK (0)

yatli, 2019-12-02:
Adding the image to disassembly as an overlay, at 0x200000...

jmamma, 2019-12-02:
Excellent work.

When you have some time. Add to the documentation how you get the decompiler involved.

yatli, 2019-12-02:
Sure :)

Funny thing is, the code doesn't run well on x64.
Have to compile it as 32-bit x86 so that the bit chewing magic could work.

jmamma, 2019-12-02:
can you dump the extracted rom here ? Want to take a quick look.

yatli, 2019-12-02:
[machdrum_os.zip](https://github.com/jmamma/MIDICtrl20_MegaCommand/files/3909696/machdrum_os.zip)

jmamma, 2019-12-02:
YES!!!!

jmamma, 2019-12-02:
You've made history.

yatli, 2019-12-02:
what happened??

jmamma, 2019-12-02:
[image]

You uncovered Elvis.

jmamma, 2019-12-02:
nobody has a photo of him. until now.

yatli, 2019-12-02:
!

jmamma, 2019-12-02:
this could be the greatest achievement of the decade 😂

yatli, 2019-12-02:
When you first bring up the Elvis thing I thought 🤔 probably you're just joking...

Just dig out the backstory, whoops!

jmamma, 2019-12-02:
I guess we have a new sprite to add to MCL 2.50

jmamma, 2019-12-02:
[image]

jmamma, 2019-12-02:
2 frames. I think they're interlaced on display.

Text Should Read "THIS PIECE OF MACHINE IS OK"

jmamma, 2019-12-02:
[image]

yatli, 2019-12-02:
`#CASE_CLOSED`

jmamma, 2019-12-02:
Is the DSP code unpacked ?

I wonder how easy it would be to disable 12bit emulation

yatli, 2019-12-02:
They deliberately crip it down to 12bit??

yatli, 2019-12-02:
I haven't unpacked DSP code. It's located somewhere else, and requires some register tampering

yatli, 2019-12-02:
We can have a quick sync later today, I'll help to setup the environment for you.

jmamma, 2019-12-02:
thanks @yatli 

It's late evening here. If you have time to document I can async catch up tomorrow during the day.

jmamma, 2019-12-02:
> They deliberately crip it down to 12bit??

Yeah, it was a stylistic/marketing choice when the UW machines were released. Gritty 12bit samples were seen as desirable. 

AFAIK the input's and outputs are 24bit, you can see the spec of the audio interface board in the MAME notes.

The sample space consumed is 16bit worth. So they're either truncating on save, or on the way out. Personally if I was coding it, I would just be applying a 12bit effect on the output. I think they also perform some light filtering/eq as well.

yatli, 2019-12-02:
I know how to natively decode the firmware now.
Right before the emulator writes to ROM, redirect destination to 0x1000000 the internal RAM. It's always there.

The result: slight difference in decoded image...
[image]

Notice row 2, col 13-16, left: 0A2A right: 0D0A

..wat?

yatli, 2019-12-02:
Oh well. Lost some of my instincts -- `fwrite` should be used with mode "wb".
Problem fixed.

yatli, 2019-12-02:
Entering `MD_MAIN`

jmamma, 2019-12-02:
0722986d88e3ef7c609540cda9de141e0b49af88
😀

yatli, 2019-12-03:
`This MCL is O.K. Too` ❤

yatli, 2019-12-03:
I've sent you the invitation link to https://github.com/yatli/mdhack

yatli, 2019-12-03:
Getting close:

```c

/* WARNING: Globals starting with '_' overlap smaller symbols at the same address */

void SAMPLE_MGR(void)

{
  int iVar1;
  int iVar2;
  int iVar3;
  int iVar4;
  uint uVar5;
  char *local_34;
  byte abStack29 [25];
  
  if (_DAT_0028b810 == 0) {
    return;
  }
  FUN_00210f84((int *)&DAT_002819f2,2,7,0x60,0x19,0);
  switch(DAT_0028b804) {
  case 0:
    FUN_0022e7e6();
    return;
  case 1:
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)s_RECEIVING_SAMPLE_0024756c);
    local_34 = s_RECEIVING_SAMPLE_0024756c;
    goto LAB_0022e90a;
  case 2:
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)s_CLEANING_0024757d);
    local_34 = s_CLEANING_0024757d;
    uVar5 = 0x13;
    break;
  case 3:
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)s_LOADING_SAMPLES_0024715f);
    local_34 = s_LOADING_SAMPLES_0024715f;
LAB_0022e90a:
    uVar5 = 0x13;
    break;
  case 4:
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)s_MEMORY_FULL_00247586);
    iVar4 = -iVar3 + 100;
    if (iVar4 < 0) {
      iVar4 = -iVar3 + 0x65;
    }
    FUN_00211384((int *)&DAT_0025997c,(int *)&DAT_002819f2,iVar4 >> 1,0x13,-1,
                 (byte *)s_MEMORY_FULL_00247586);
    iVar4 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)s_MAX_SIZE:_00247592);
    iVar3 = DAT_0029e9f0;
    iVar1 = FUN_0020f190();
    iVar2 = FUN_0020e164(DAT_0028b80c);
    iVar3 = ((iVar3 - iVar1) + iVar2) * 2;
    if (iVar3 < 1) {
      FUN_00211d46((char *)abStack29,&DAT_0024755b);
    }
    else {
      FUN_0022deca((char *)abStack29,iVar3);
    }
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,abStack29);
    iVar1 = -(iVar4 + iVar3) + 100;
    if (iVar1 < 0) {
      iVar1 = -(iVar4 + iVar3) + 0x65;
    }
    FUN_00211384((int *)&DAT_0025997c,(int *)&DAT_002819f2,iVar1 >> 1,7,-1,
                 (byte *)s_MAX_SIZE:_00247592);
    iVar1 = (iVar4 - iVar3) + 100;
    if (iVar1 < 0) {
      iVar1 = (iVar4 - iVar3) + 0x65;
    }
    FUN_00211384((int *)&DAT_0025997c,(int *)&DAT_002819f2,iVar1 >> 1,7,-1,abStack29);
    return;
  case 5:
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)s_ERROR_00247022);
    iVar4 = -iVar3 + 100;
    if (iVar4 < 0) {
      iVar4 = -iVar3 + 0x65;
    }
    FUN_00211384((int *)&DAT_0025997c,(int *)&DAT_002819f2,iVar4 >> 1,0x13,-1,
                 (byte *)s_ERROR_00247022);
    if (DAT_0028b808 - 1U < 7) {
      switch(DAT_0028b808) {
      case 1:
        local_34 = s_WRONG_PACKET_002475da;
        break;
      default:
        local_34 = s_WRONG_CHECKSUM_002475a9;
        break;
      case 4:
        local_34 = s_INVALID_POSITION_002475b8;
        break;
      case 5:
        local_34 = s_BUFFER_FULL_0024759d;
        break;
      case 6:
        local_34 = s_FLASH_CONGESTION_002475c9;
        break;
      case 7:
        local_34 = s_INVALID_SMPL_RATE_002475e7;
      }
      FUN_00211d46((char *)abStack29,local_34);
    }
    else {
      FUN_00211d46((char *)abStack29,s_CODE:_%d_002475f9);
    }
    local_34 = (char *)abStack29;
    iVar3 = FUN_00211634((int *)&DAT_0025997c,-1,(byte *)local_34);
    uVar5 = 7;
    break;
  default:
    goto switchD_0022e88a_caseD_6;
  }
  iVar4 = -iVar3 + 100;
  if (iVar4 < 0) {
    iVar4 = -iVar3 + 0x65;
  }
  FUN_00211384((int *)&DAT_0025997c,(int *)&DAT_002819f2,iVar4 >> 1,uVar5,-1,(byte *)local_34);
switchD_0022e88a_caseD_6:
  return;
}

```

yatli, 2020-01-20:
Closing now.

closed 2020-01-20 by yatli

tomduncalf, 2020-01-20:
This is really interesting, I'd been really curious about trying to decode the firmware, mostly just out of curiosity! Anywhere to follow for further updates on your work @yatli?

TronicLabs, 2020-02-17:
@yatli I would also like to be able to follow your project,
because I would like to learn, to know what tools you are using, and how to.
Thanks for this wonderful work to both of us.

yatli, 2020-02-17:
@tomduncalf @TronicLabs please understand that we are still communicating with Elektron to see if we can open more information wrt the firmware.

tomduncalf, 2020-02-17:
@yatli no worries, I totally understand that it is a sensitive subject! Really cool that you were able to figure it out!

[1 more reply]