jmamma/MCL #92: SEQ Merge behaviour not clearly defined.

jmamma, 2019-12-05:
SeqMerge does just that, it merges the pattern data from the MD in to the internal sequencer.

I think in the general case you want to just copy the entire track from the MD, and overwrite any internal sequencer data. The current function does not do this, it sums the internal seq + md tracks together.

This is related to the track length bug, where merging would not take the MD.pattern length and instead user the internal sequencer's track length.

@yatli 

Thoughts?

yatli, 2019-12-05:
perhaps max(mdlen, intlen)?

jmamma, 2019-12-05:
it's not just the legnth.

Do we clear the internal sequencer data on merge.
Currently if MD pattern = 1111000
and            md_track.seq_data.pattern = 00001111
the resulting pattern will be 11111111

jmamma, 2019-12-05:
On Save page we should probably have 3 options

1) --
2) MD
3) MERGE

yatli, 2019-12-05:
Ok then:

```
SOURCE
1. SEQ
2. MD
3. MERGE
```

correct?

jmamma, 2019-12-05:
that's it

jmamma, 2019-12-05:
312 bool MDTrack::store_track_in_grid(int32_t column, int32_t row, int track, bool storepattern,
313                                   bool merge, bool online) {

bool merge needs to be converted to uint8_t.

and then replace if (merge)
with a switch (merge)

jmamma, 2019-12-05:
https://github.com/jmamma/MIDICtrl20_MegaCommand/pull/93

jmamma, 2019-12-06:
Hmm this seems a little broken.

jmamma, 2019-12-06:
I think no constructor for MDTrack is setting arraysize = random();
And merging collect fake locks.

jmamma, 2019-12-06:
@yatli 

I'm considering removing MERGE and limit save functionality to

MD
and
SEQ.

To avoid confusion.

jmamma, 2019-12-06:
Woops. continue discussion in PR.

jmamma, 2019-12-07:
I merged the PR. Parameter locks were broken. That is now fixed.

--

Functionality is implemented now.

Workflow feels unpredictable at times. Even i'm forgetting which save mode i'm in. 
Particularly when MD mode toggles to SEQ when sequencer starts.

jmamma, 2019-12-07:
Maybe we move the MD save function outside of the SAVE page.
And forego merge.

yatli, 2019-12-07:
Is MERGE fixed now?

jmamma, 2019-12-07:
yes it should be.

yatli, 2019-12-07:
Great, I think I should test it against the lockup bug again.

jmamma, 2019-12-07:
Which lockup bug? sample send?

yatli, 2019-12-07:
The chain-write lockup on some rows (and crazy tempo)

jmamma, 2019-12-07:
It's possible.

jmamma, 2019-12-07:
Found another bug.

Write page is not writing to the current kit when "writing all tracks [Shift 2]"

jmamma, 2019-12-07:
Actually, i think it's not re-loading the current kit after write.

jmamma, 2019-12-07:
305 void MCLActions::send_tracks_to_devices() {
...
386   MD.loadKit(MD.pattern.kit);

eek. This was left over from when we deprecated write mode.
Not sending the pattern to the MD, so MD.pattern.kit is out of date.

jmamma, 2019-12-08:
encoder[0]->cur = 1 was being set everytime save page opened.
i've removed this, as it has added to the confusion.

jmamma, 2019-12-08:
@yatli 
SavePage graphic
SEQ -> GRID

should be 

SND -
          ----> GRID 
SEQ -

?

jmamma, 2019-12-08:
Another save bug
    SEQ mode, would not copy sequencer data to track object for storage
899547be0e44624c505219e4f7510bae9dd4a85b

Hopefully the last.

yatli, 2019-12-08:
> @yatli
> SavePage graphic
> SEQ -> GRID
> 
> should be
> 
> SND -
> ----> GRID
> SEQ -
> 
> ?

Ah, yes.

yatli, 2019-12-08:
I remember I've implemented this?

jmamma, 2019-12-08:
what do you mean?

yatli, 2019-12-08:
On the Save page, do you see the (SND+SEQ) -> GRID data flow, or just SEQ->GRID?

jmamma, 2019-12-08:
just SEQ->GRID

jmamma, 2019-12-08:
Merge shows MD +SEQ -> GRID

yatli, 2019-12-08:
hmm. I'll update this. :)

jmamma, 2019-12-08:
I feel like 2.50 release is very close.

jmamma, 2019-12-10:
fyi: fixed
f1817aa4d69dbf934d94cd380082e8a8660f5c40

closed 2019-12-10 by jmamma

yatli, 2019-12-10:
Cool!