jmamma/MCL #101: Custom TI mode

yatli, 2020-01-20:
- [ ] Trig key handler
- [ ] Custom LEDs

jmamma, 2020-01-20:
How far along are you?

I am going to write up the Sysex handler now.

yatli, 2020-01-20:
MCL side? Still trying to bring up the dev env.
Can help testing the code tonight.

jmamma, 2020-01-20:
All good, just dont want to double up on work. i'm 70% of the way there.

yatli, 2020-01-23:
Tracking: #102

yatli, 2020-01-23:
Related: #73

jmamma, 2020-01-23:
Need to re-think global settings now. 

Should we allow the user to specify base channel, or just stick with the default channels used by MCL.

--

Also, because we no are no longer monitoring note messages, MIDI machines will be compatible with MCL.

yatli, 2020-01-23:
> Should we allow the user to specify base channel, or just stick with the default channels used by MCL.

What's the implication?

yatli, 2020-07-31:
Closing now. Implemented.

closed 2020-07-31 by yatli