jmamma/MCL #103: SDDrive: sample pack

yatli, 2020-02-24:
ROM slots snapshots.

yatli, 2020-02-24:
https://github.com/jmamma/MIDICtrl20_MegaCommand/commit/cf75574c561dbff0bc463c103a99b860c82e3571#diff-3ac5ec1e588b83481f58ae3442cf4bb3R8

Add target slot param?

jmamma, 2020-02-24:
> [cf75574#diff-3ac5ec1e588b83481f58ae3442cf4bb3R8](https://github.com/jmamma/MIDICtrl20_MegaCommand/commit/cf75574c561dbff0bc463c103a99b860c82e3571#diff-3ac5ec1e588b83481f58ae3442cf4bb3R8)
> 
> Add target slot param?

MIDISds classes have that functionality.
https://github.com/jmamma/MIDICtrl20_MegaCommand/blob/master/avr/cores/megacommand/Midi/MidiSDS.hh

The MidiSysexFile class was created to send a raw dump from a file off the sdcard. It's untested.

yatli, 2020-02-25:
> MIDISds classes have that functionality

Ah, superb.

Btw figured out that 48Khz SDS sample will be downsampled 2x (never tried 48Khz myself though!) on the host side.

jmamma, 2020-02-25:
Yeah. I've written support for sending/receiving SDS and also there is a fairly usable WAV library I created.

> Btw figured out that 48Khz SDS sample will be downsampled 2x (never tried 48Khz myself though!) on the host side.

What evidence do you have for this. What about for other sample rates?

Usually downsampling is accompanied with filtering at the nyquist frequency. So you'd lose everything above 12khz.

yatli, 2020-02-25:
Crude downsampling: 0x0020e806
Condition: 0x0020f574

yatli, 2020-02-25:
My brain isn't working now.
```
*piVar17 = ((iVar8 >> (uVar18 & 0x3f)) - iVar11 << (uVar13 & 0x3f)) >> (local_8 & 0x3f);
```
Does this look like it's also related to 12-bit resolution?

jmamma, 2020-02-25:
ha, i was looking at my MIDISds code before shuddering at all the bit shifting.

Now i'm reliving it @ 0x0020f60a

yatli, 2020-02-25:
This is what it really looks like:

```
      *piVar15 = ((iVar8 >> (uVar16 & 0x3f)) - bitcrush_mask << (bitcrush_up & 0x3f)) >>
                 (bitcrush_down & 0x3f);
```

Not really the 12-bit thing, because at format 0x10 (16-bit), both up and down are set to 0.

jmamma, 2020-02-25:
the first shift might be removing padding from the sample. I'm not sure what the subtraction does. Convert from unsigned to signed? *shrug*.

jmamma, 2020-02-25:
The SDS receive code from MIDICtrl.

```
      for (shift = 0; shift < midi_sds.midiBytes_per_word; shift++) {

        decode_val = decode_val << 7;
        decode_val |= (int32_t)(0x7F & sysex->getByte(n + 4 + shift));
      }
      // Remove 0 padding.
      decode_val = decode_val >> (8 - shift);
      // For bit depth greater than 8 we need to convert the sample
      // from unsigned to signed by subtracting offset

      if (midi_sds.bytes_per_word > 1) {
        signed_val = decode_val - midi_sds.sample_offset;
        // decode_val -= midi_sds.sample_offset;
      } else {
        signed_val = decode_val;
      }
```

yatli, 2020-02-25:
> sample_offset 

This, exactly. `bitcrush_mask = 1 << (sds_sample_format - 1U & 0x3f);`

jmamma, 2020-02-25:
maybe relabel that one :) .

also small typo here ``/* two samples in iVar8 and iVar7 now */`` should be iVar6 and iVar7

yatli, 2020-02-25:
ha, I shouldn't put the autovar names in comments -- they change when I label something

jmamma, 2020-02-25:
now you know the joys of SDS.

I think my implementation doesn't look too bad in comparison.

Interesting find regarding the > 48khz downsample + averaging filter.

I wonder how many samples i've uploaded at 48khz in the past.

jmamma, 2020-02-26:
> Or, a list of file names. Could be .syx or .wav, more flexible.

I prefer strongly for converting sds to wav, and saving as wav format.

The MIDISds recv implementation already does this by saving as WAV.

See: 

https://github.com/jmamma/MIDICtrl20_MegaCommand/blob/master/avr/cores/megacommand/Midi/MidiSDSSysex.cpp

yatli, 2020-03-01:
Can we move SDDrive to the top-level?

Sound Manager, Wav Designer, Loudness, SD-Drive?

jmamma, 2020-03-01:
Yeah we can move it up in Global Settings menu.

Leave quick select as it is for now.

jmamma, 2020-03-01:
https://www.youtube.com/watch?v=7kyaz4Zrd78 

ez pz.

closed 2023-08-01 by jmamma