jmamma/MCL #107: TRX-S2 related updates

yatli, 2020-03-14:
- Model name "S2"
- Parameter names
- More?

jmamma, 2020-03-14:
Tuning

jmamma, 2020-03-15:
PTC  HZ       Note

3    42.64    F2
7    93.27    F#2
11   98.60    G2
15   104      G#2
20   110      A2
24   116      A#2
30   120      B2
35   130      C3
41   138      C#3
47   147      D3
54   156      D#3
60   164      E3
68   175      F3
76   185      F#3
84   196      G3
92   207      G#3
101  219      A3
111  233      A#3
121  246      B3

jmamma, 2020-03-15:
https://github.com/jmamma/MIDICtrl20_MegaCommand/commit/274383d93f411b70859653401499c64d43a0b3e8

jmamma, 2020-03-15:
I dont know  what tuning_t->offset is supposed to be.
It's only made use of in MDClass::trackGetCCPitch

yatli, 2020-03-15:
Ruin & Wesen relics?

closed 2020-04-04 by yatli