jmamma/MCL #110: 3.0

jmamma, 2020-04-01:
- [ ] New project spec. Increase parameter locks to 8.
- [ ] Drop support for A4/EXT tracks.
- [ ] Project format conversion.
- [ ] Add proper MIDI machine support. 
- [ ] WAV editor
- [ ] Improve WAV designer.
- [ ] Advanced sampling functionalities. 

@yatli thoughts?

yatli, 2020-04-01:
- Double (or custom) step resolution for MD
- Automation tracks in the grid
- Piano roll? :x

yatli, 2020-04-01:
Wrt the project format:

- Self-descriptive metadata: DEVICE={MD|MM|A4|OT|...}, PRJVER=3.0, TRKSIZE=..., TRKCNT=...
- Sample assignment info for a project. On loading a project, prompt to sync the samples.

yatli, 2020-04-01:
Wrt A4/EXT/MM/whatsover support:

Initially we can simply drop the A4/EXT support.
If we could have a hardware abstraction layer, and later on if any of us feel like developing for a new machine, we can create multiple build profiles, mix-and-match the relevant pages.

I'm good with one-MC-per-machine. UART1=control the machine. UART2=interconnection.

jmamma, 2020-04-02:
Regarding poly groups. Perhaps each grid slot should be assignable to a group.

> * Self-descriptive metadata: DEVICE={MD|MM|A4|OT|...}, PRJVER=3.0, TRKSIZE=..., TRKCNT=...
> 
> * Sample assignment info for a project. On loading a project, prompt to sync the samples.

This can be built in to the project header class.

jmamma, 2020-04-02:
Automation tracks in the grid <- what did you have in mind @yatli

yatli, 2020-04-02:
> Automation tracks in the grid

RAM machine actions, LFO patterns + enable/disable etc.

jmamma, 2020-04-03:
Should we have an underlay grid for this ?

jmamma, 2020-07-29:
@yatli Looks like I want to keep the A4/Ext support going forward...

I am going to re-work the ExtSeq page.

Display the 4 notes on screen.

I want to abstract away the "on-off" sequencer by adding a length control knob for notes.
And also, allow for the ability to control each note value from the MC. (Currently you need to enter notes from and ext midi device)

@yatli any thoughts on a clever UI for adjusting notes and note length simultaneously. 4 notes per step etc...

yatli, 2020-07-29:
Any chance we also support MnM? :p

yatli, 2020-07-29:
I've actually been thinking about a new sequencer for a while.
The sequencer has to be friendly with the trig interface while maintaining both note trigs and lock trigs.
I suggest we decouple note editing and lock editing. In this way, we reuse the step sequencer UI for locks, but do notes with a new UI.

jmamma, 2020-07-29:
> I suggest we decouple note editing and lock editing

Can you elaborate a bit ?

--

Also, I almost forgot, I already allocated space for parameter locks in ExtSeqTrack.h so we could implement locks right away without changing project spec.

yatli, 2020-07-29:
> Can you elaborate a bit ?

On MnM there are 3 types of trigs: AMP, FILTER, LFO.
Basically, ALL=what we have on MD, and each component controls certain page of parameters.

When we decouple note editing from lock editing, we effectively separate the note page from the others.
The note page contain parameters like note1~4, length etc, and we can choose a different UI layout (e.g. no need to display knobs for notes)

And the other parameters can be locked in the Elektron style.

yatli, 2020-07-29:
[image]

Could fit 11 notes vertically + 2 overflow/underflow indicators

jmamma, 2020-07-29:
Are you thinking piano roll ?

yatli, 2020-07-29:
yep piano roll thingy, but designed for compact screen.

My favorite design is the Reason matrix step sequencer emulating LCD display:

[image]

jmamma, 2020-07-29:
MATRIX sequencer. brings back memories. 

Yeah we can do piano roll. does this mean we need a cursor for selecting notes?

yatli, 2020-07-29:
The TI leds can be used as a vertical projection of the display. so each note is displayed as a consecutive range on the TI.
Pressing a trig selects the notes that are covered in the range.
If ambiguity presents, then select the notes whose heads are nearest to the pressed trig.

jmamma, 2020-07-29:
how do we select individual notes in a chord?
for instance if we want to shift or lengthen a note

yatli, 2020-07-29:
MD arrow keys?

yatli, 2020-07-29:
Or we need to use a knob to indicate "selection"

jmamma, 2020-07-29:
i guess we could also use an encoder for that.

enc -> note select
enc -> horizontal shift
enc -> vertical shift
enc -> zoom ? 

we also need micro timing and conditional encoders.

and pattern length.

yatli, 2020-07-29:
> enc -> zoom ?

I think we probably don't need zoom.
Vertical scroll instead?

jmamma, 2020-07-29:
we dont need micro timing exactly.. if we have sufficient horizontal resolution we can represent utiming by shifts in start/end position of notes.

yatli, 2020-07-29:
> horizontal zoom might be useful here

Ah ok I see. A combination of horizontal zoom and "snap to the grid".
Can be put in the menu?

yatli, 2020-07-29:
Another option is to go modal. Let there be two modes, INSERT/SELECT, where the trigs behave as:

INSERT mode: 4 encoders bound to note pitch. Press trig toggles notes. Press and hold a trig edits pitch.
SELECT mode: 4 encoders bound to H/V shifting, length and expression (expression can be velocity, accent, porta etc.) Press trig toggles selection (in case of a chord, cycles through the notes). Press and hold go into temporary pitch editing.

jmamma, 2020-07-29:
yeah, pretty much what i was thinking. you have 2 sets of encoders. the second set relating to note position + selection is swapped in when you hold a trig.

yatli, 2020-07-29:
If done right, I feel this could form the basis of a great sequencer product...
The Elektron interface is nice, but not good enough for chords or strummings

jmamma, 2020-07-29:
We already beat the elektron sequencer, because the on+off sequencer design inherently support legato :D

jmamma, 2020-07-29:
just need to wrap it up in a better UI.

yatli, 2020-07-29:
One thing I love most about the FL Studio PR is its ability to click on one note to copy all its properties (length, velocity etc.) and then apply them to newly drawn notes.

This can apply to the INSERT mode, where you long press a chord and capture its properties. You don't need to repeatedly adjust length/chord for the newly toggled trigs.

jmamma, 2020-07-29:
Sounds good to me.

yatli, 2020-07-29:
(Talks about Reason and FL Studio buh buh)
So why do I end up staying with Bitwig Studio (

jmamma, 2020-07-29:
been meaning to download bitwig again. i've been slowly catching up on DAW + DSP + VST tech, since I left off 2-3 years ago? Lots to play with.

jmamma, 2020-07-29:
**Ext Sequencer Refresh Tutorial (Lesson 1):**

```
 34 class ExtSeqTrackData {
 35 public:
 36   uint8_t length;
 37   uint8_t speed; // Resolution = 2 / ExtPatternResolution
 38   uint8_t reserved[NUM_EXT_NOTES];
 39   int8_t notes[NUM_EXT_NOTES][NUM_EXT_STEPS]; // 128 steps, up to 4 notes per step
 40   uint8_t locks[NUM_EXT_LOCKS][NUM_EXT_STEPS]; 
 41   uint8_t locks_params[NUM_EXT_LOCKS];
 42   uint64_t lock_masks[NUM_EXT_LOCKS];
 43 
 44   uint8_t conditional[NUM_EXT_STEPS];
 45   uint8_t timing[NUM_EXT_STEPS];
 46 };
```

The Ext Seq uses an on+off approach. Each step can have up to 4 notes of different pitch, with a state in either NULL, ON or OFF. 

The on+off design has some drawbacks. Namely consecutive short notes.
Because notes on + note off events have to be placed on separate trigs, you need to double the resolution to get 16th notes
of the same pitch.

```
1 .            2.           3.           4.

[           ] [           ] [           ] [          ]   1x 
ON          OFF        ON         OFF

[    ][    ] [    ][    ] [    ][    ]  [    ][    ]  2x
ON OFF  ON OFF ON OFF  ON OFF
```

An alternative design would be to increase the polyphony to 8. and allow on + off  for the same pitch on the same step. This would consume the same amount of RAM, but be limited to 64 steps. Perhaps this would have been more lucrative as you gain additional polyphony. The drawback being, you would lose polyphony if playing consecutive notes quickly. 

The other design approach was to have uint8_t per note representing note length. This would also consume the same amount of RAM at 64 steps, but would require processing overhead, to keep track of when to turn notes off.

jmamma, 2020-07-29:
Originally I had attempted to conceal the doubling of resolution, by halving displayed track length and doubling the cur_step cursor speed.

This is evident in my selection of:
 42   uint64_t lock_masks[NUM_EXT_LOCKS];

Where locks patterns were only intended to only be 64 steps in length...

jmamma, 2020-07-29:
sleep time 🌔

yatli, 2020-07-29:
Good night.

yatli, 2020-08-10:
https://github.com/jmamma/MIDICtrl20_MegaCommand/pull/128

yatli, 2020-11-23:
@jmamma is the latest `dev` working? I'm trying to build it from Linux but it doesn't respond after upload.

jmamma, 2020-11-23:
dev branch is working.

If DEBUG_MODE is enabled, the sketch will be too large. Need to disable DEBUG_PRINT_FN()

yatli, 2020-11-23:
Check out https://github.com/jmamma/MIDICtrl20_MegaCommand/commit/7030e3a19908ad722ec819c8fb49984d617da578

Now you can do `make debug_upload -j` :D

jmamma, 2020-11-23:
I saw. good work!

mcl_setup.cpp <- #ifdef MCL  
is probably the issue.

can you fix?

yatli, 2020-11-23:
Disabled DEBUG_PRINT_FN, sketch size is: `avrdude: 238422 bytes of flash written`
Nothing coming out from the serial monitor.

yatli, 2020-11-23:
> mcl_setup.cpp <- #ifdef MCL

Oh I see. This one again. I'll fix it

yatli, 2020-11-23:
Got these:

```
/usr/bin/avr-ld: /tmp/main.elf.OXOu1f.ltrans0.ltrans.o: in function `MCLSeq::seq()':
/home/yatli/git/MIDICtrl20_MegaCommand/avr/cores/megacommand/MCL/MCLSeq.cpp:174: undefined reference to `SeqTrack::send_slides(unsigned char volatile*, unsigned char)'
/usr/bin/avr-ld: /home/yatli/git/MIDICtrl20_MegaCommand/avr/cores/megacommand/MCL/MCLSeq.cpp:174: undefined reference to `SeqTrack::send_slides(unsigned char volatile*, unsigned char)'
/usr/bin/avr-ld: /home/yatli/git/MIDICtrl20_MegaCommand/avr/cores/megacommand/MCL/MCLSeq.cpp:174: undefined reference to `SeqTrack::prepare_slide(unsigned char, int, int, signed char, signed char)'
/usr/bin/avr-ld: /home/yatli/git/MIDICtrl20_MegaCommand/avr/cores/megacommand/MCL/MCLSeq.cpp:174: undefined reference to `SeqTrack::prepare_slide(unsigned char, int, int, signed char, signed char)'
```

jmamma, 2020-11-23:
Thanks. forgot to check-in.

It's there now.

yatli, 2020-11-23:
yatli cold-booting back...

jmamma, 2020-11-23:
enjoy the piano-roll + automation editor 😁

yatli, 2020-11-23:
Press SAVE+WRITE then release: goes into Load Project.
Then press SAVE: goes back to global menu, but freezes.

jmamma, 2020-11-23:
nope.

shouldn't do that. no changes made there.

SAVE+WRITE opens up global settings for me.

jmamma, 2020-11-23:
avrdude: writing flash (241858 bytes):

make debug_upload -j -> binary is too big ?

should be:

avrdude: writing flash (226054 bytes):

jmamma, 2020-11-23:
Don't know why ```make debug_upload -j``` is resulting in a larger binary size?

make_and_upload.sh is working okay?

..

getting late here. i'm off

yatli, 2020-11-23:
catch you later. going home now :)

jmamma, 2020-11-24:
```
125 debug: CXXFLAGS += -DDEBUGMODE=1
126 debug: CFLAGS += -DDEBUGMODE=1
127 debug: main.hex
128 
129 debug_upload: debug upload
```

The reason the file size is 241858 is because DEBUG_MODE is enabled. 
The weird behaviour your were seeing is due to program corruption because the ROM size is too big.
Initially I thought this was the 8KB issue, but it looks like there is an additional problem when we hit 240KB ... 

If compiling with DEBUG_MODE need to disable PRINT_FN() to save a few KB.

yatli, 2020-11-24:
Guess what:

```
avr-gcc 10.2
   text    data     bss     dec     hex filename
 231286    6306   42409  280001   445c1 main.elf

avr-gcc 9.3
   text    data     bss     dec     hex filename
 230980    6088   42409  279477   443b5 main.elf

avr-gcc 8.3
   text    data     bss     dec     hex filename
 217344    6014   42409  265767   40e27 main.elf
```

Bigger is better (TM)

jmamma, 2020-11-24:
counterintuitive 

wonder if the optimisation options have changed.

jmamma, 2020-11-24:
https://www.avrfreaks.net/forum/code-size-gets-bigger-each-new-version-gcc

yatli, 2020-11-24:
lol

yatli, 2020-11-24:
The only compiler that works here is the one installed from arduino-cli (7.3-atmel-3.6.1-arduino)

jmamma, 2020-11-24:
what do you mean 'works'?

[64 more replies]

closed 2021-03-24 by jmamma