jmamma/MCL #114: Circular dependency MCLMemory.h

jmamma, 2020-05-15:


jmamma, 2020-05-15:
@yatli do you have an spare CPU cycles to help me troubleshoot this one?

I can't think of a clever way to get around the fact that MDTrack.h and MCLMemory.h both need to include each other

forward declaration of MDTrackLight wont work because sizeof() needs a complete type.

yatli, 2020-05-15:
Can we split a part of MCLMemory.h into a separate header?
MDTrack.h can include that one.

jmamma, 2020-05-15:
I tried moving BANK1_MD_TRACK_START
to MDTrack.h (where it is used) and include "MDTrack.h" in MCLMemory.h

but now A4Track.h is complaining it doesn't have the declaartion for BANK1_A4_TRACKS_START (which is in MCLMemory.h)

 51 #define BANK1_A4_TRACKS_START (BANK1_MD_TRACKS_START + MD_TRACK_LEN * NUM_MD_TRACKS)

🤯

yatli, 2020-05-15:
Does A4Track.h include MCLMemory?

jmamma, 2020-05-15:
yes

yatli, 2020-05-15:
```
MDTrack.h [origin] [defines MD_TRACK_LEN, BANK1_MD_TRACKS_START]
A4Track.h [includes MDTrack.h] [defines A4_TRACK_LEN, BANK1_A4_TRACKS_START]
MCLMemory.h includes [MDTrack.h] [A4Track.h]
```

Does this solve the problem?

yatli, 2020-05-15:
Hmm, do we need MCLMemory.h at all?

jmamma, 2020-05-15:
> Does this solve the problem?

Doesn't work. MD_TRACK_LEN needs to be defined in MDTrack.h

But A4Track.h doesnt see it.
/Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/A4Track.h:14:56: error: 'MD_TRACK_LEN' was not declared in this scope

I think because of

But because of ifndef.

```
  6 #ifndef MDTRACK_H__
  7 #define MDTRACK_H__
...
 58 class MDTrackLight
 59     : public GridTrack,
 60       public Bank1Object<MDTrackLight, 0, BANK1_MD_TRACKS_START> {
 61 public:
 62   MDSeqTrackData seq_data;
 63   MDMachine machine;
 64 };
 65 
 66 
 67 #define MD_TRACK_LEN (sizeof(MDTrackLight))
```

yatli, 2020-05-15:
Does A4Track.h include MDTrack.h?

jmamma, 2020-05-15:
yeah...

yatli, 2020-05-15:
If the ifndef guard hides the range, it's because it's already included. So it's impossible that the condition goes true and the requiring source doesn't see the content.

yatli, 2020-05-15:
Try replace the #define with `constexpr size_t MD_TRACK_LEN=sizeof(MDTrackLight);`?

jmamma, 2020-05-15:
Reset the branch to 1238c30  
                              
/Applications/Arduino.app/Contents/Java/hardware/valence/avr/cores/megacommand/MCL/MCLMemory.h:37:38: error: 'MDTrackLight' was not declared in this scope
 constexpr size_t MD_TRACK_LEN=sizeof(MDTrackLight);

jmamma, 2020-05-15:
oh right you were suggesting add it in A4Track.h ?

Getting late here... might pick this up in the morning.

jmamma, 2020-05-15:
Found a hacky way around this..

you had these two defines commented out. 

```
 26 #include "MDSeqTrack.h"
 27 #include "ExtSeqTrack.h"
 28 #include "MD.h"
 29 
 30 #define MD_TRACK_LEN (sizeof(GridTrack) + sizeof(MDSeqTrackData) + sizeof(MDMachine))
 31 #define A4_TRACK_LEN (sizeof(GridTrack) + sizeof(ExtSeqTrackData) + sizeof(A4Sound))
```

jmamma, 2020-05-15:
Okay... compiles. 

MCLActions refactored, and stack overflow avoided.
Code looks much cleaner now.

closed 2020-06-08 by jmamma