jmamma/MCL #115: non-C++ functions in memory.h use C++-only macro

jwakely, 2020-05-18:
Apologies if I'm misunderstanding something, but it looks like the functions outside the `#ifdef __cplusplus` block in `avr/cores/megacommand/memory.h` won't actually compile for C, because they use the `select_bank` macro which is only defined for C++.

e.g. 
```
#endif// __cplusplus

ALWAYS_INLINE() extern inline void memcpy_bank1(volatile void *dst, volatile const void *src, uint32_t len) {
  select_bank(1);
  memcpy(dst, src, len);
}
```
This uses `select_bank(1)` which is not defined unless `__cplusplus` is defined.

Do those functions actually need to get called by non-C++ code, or can they just be moved inside the `#ifdef __cplusplus` block ?

jmamma, 2020-05-19:
> Do those functions actually need to get called by non-C++ code, or can they just be moved inside the #ifdef __cplusplus block ?

Good catch. Those other functions  should be moved within the C++ guard.
We're currently getting away with it because no C code is calling them.

--

@jwakely thanks for checking out our project. I suspect you'll find other idiosyncrasies 😄

jwakely, 2020-05-19:
OK thanks for confirming. I'll send a pull request to move them into the guarded block.

Thanks for the project! I'm considering building a megacommand thought I'd have a look at the code.

yatli, 2021-03-01:
@jwakely should be fixed with ba2510b6

commit ba2510b6 (2021-03-01): fix warnings, fix #115, update build flags for arduino environment
closed 2021-03-05 by jmamma