jmamma/MCL #119: Wavetable synthesis

yatli, 2020-06-14:
We can prepare morphing waveforms (like the Ableton Live new synth) and pack them into one sample.

Load them up in a linear ROM machine, and control it to move in different directions...
Even if we only have a linear axis to control the offset, we can use different traverse algorithms to walk between different waves:

Linear morph loop:

[image]

Morph tree:

[image]

etc.

jmamma, 2020-06-15:
I've been thinking bout how to do this for a while.

I think you also need a 'static' region for each waveform 1 2 3, where the waveform is not morphing.

yatli, 2020-06-15:
Given that each wave itself needs an interval in the axis, yeah, the static region. But if the static regions are long we may not have enough resolution to do the morph smoothly.

What's your take on this? I remember you have knowledge about creating few cycle samples. What's the shortest possible waveform length?

jmamma, 2020-06-15:
Shortest possible length == waveform length of fundamental frequency.

closed 2021-04-24 by jmamma