jmamma/MCL #120: Slide functionality is causing memory corruption when chaining.

jmamma, 2020-06-15:
With slide functionality added i'm observing memory corruption when manual chaining of slots with sequencer running. 

```
In MDSeqTrack::seq() a call to 
send_slides();
was added.

in MCLSeq::seq() 

174   for (uint8_t i = 0; i < num_md_tracks; i++) {
175     md_tracks[i].recalc_slides();
176   }

was added.
```

I suspect that the stack is running out of memory and corrupting md_track->load_from_mem(n) data in GridTask::run();

However,  I have moved local variables in recalc_slides() to static variables, and the problem still persists.

jmamma, 2020-06-15:
Hmm, blocking ISR when performing memcpy seemingly fixes the problem....

```
@@ -12,19 +12,26 @@ class Bank1Object {
   public:
   inline bool store_in_mem(uint8_t column, uint32_t region = p_addr_base) 
   {
+    USE_LOCK();
+    SET_LOCK();
     uint32_t pos = region + sizeof(T) * (uint32_t)(column - n_column_base);
     volatile uint8_t *ptr = reinterpret_cast<uint8_t *>(pos);
     memcpy_bank1(ptr, this, sizeof(T));
+    CLEAR_LOCK();
     return true;
   }
 
+
   inline bool load_from_mem(uint8_t column, uint32_t region = p_addr_base)
   {
+    USE_LOCK();
+    SET_LOCK();
     uint32_t pos = region + sizeof(T) * (uint32_t)(column - n_column_base);
     volatile uint8_t *ptr = reinterpret_cast<uint8_t *>(pos);
     memcpy_bank1(this, ptr, sizeof(T));
+    CLEAR_LOCK();
     return true;
   }
 };
```

Thought our auto bank switching through switch_bank() command would negate the need for this.

jmamma, 2020-06-15:
This is concerning. Adding in unrelated lines of code fixes the problem.

diff --git a/avr/cores/megacommand/Midi/MidiClock.h b/avr/cores/megacommand/Midi/MidiClock.h
index b3446ad..b94be31 100644
--- a/avr/cores/megacommand/Midi/MidiClock.h
+++ b/avr/cores/megacommand/Midi/MidiClock.h
@@ -226,6 +226,7 @@ public:
 
     static bool inCallback = false;
     if (inCallback) {
+      oled_display.textbox("ISR ", "collide");
       DEBUG_PRINTLN("clock collision");
       return;
     } else {

jmamma, 2020-06-15:
@yatli thoughts on this. Problem with inlining and linking ?

jmamma, 2020-06-15:
Interestingly. Increase the size of SlideData (change uint8_t to uint16_t), pushes the corruption from pattern_mask to lock_mask.

SlideData is stored on the heap... I don't quite understand how this is influencing data in the stack

```
 33 class SlideData {
 34 public:
 35   int16_t err;
 36   int8_t inc;
 37   int16_t dx;
 38   int16_t dy;
 39   int16_t x0;
 40   int16_t y0;
 41   int16_t y1;
 42   bool steep;
 43   int16_t x1;
 44   uint16_t yflip;
 45   ALWAYS_INLINE() void init() {
 46     dy = 0;
 47     dx = 0;
 48   }
 49 };

yatli, 2020-06-15:
Could it be a overflow?
I've got to jump in and take a look at the code.

jmamma, 2020-06-15:
help 😭 

i'm leaning away from stack overflow being the problem.

SlideData  included by

MDSeqTrack { locks_slide_data[4];  }

is sitting in heap.

e.g :
MCLSeq {
MDSeqTrack[16];
}

yatli, 2020-06-15:
So, pattern_mask and lock_mask are both members of the same object holding the slide data?
Does it happen across tracks?

jmamma, 2020-06-15:
not exactly

MDSeqTrackData holds the pattern masks.

public data in MDSeqTrack is used by MCLSeq and is ephemeral
 
```
 51 class MDSeqTrack : public MDSeqTrackData {
 52 
 53 public:
 54 
 55   static int16_t x0, x1; 
 56   static int8_t y0, y1; 
 57   static uint8_t step, next_step;
 58   static uint8_t timing_mid;
 59 
 60   uint8_t track_number;
 61   uint8_t step_count;
 62   uint8_t mod12_counter;
 63 
 64   // Conditional counters
 65   uint8_t iterations_5;
 66   uint8_t iterations_6;
 67   uint8_t iterations_7;
 68   uint8_t iterations_8;
 69 
 70   uint32_t start_clock32th;
 71   uint64_t oneshot_mask;
 72   uint64_t slide_mask;
 73 
 74   uint8_t port = UART1_PORT;
 75   MidiUartParent *uart = &MidiUart;
 76 
 77   uint8_t locks_params_orig[4];
 78   SlideData locks_slide_data[4];
```

jmamma, 2020-06-15:
my current test is write a specific row on grid. 
all good.

sequencer start then slots tracks 13->16

slot 13 now contains extra steps in pattern mask

yatli, 2020-06-15:
ephemeral but then it persists between runs of MCLSeq::seq()?

yatli, 2020-06-15:
overwhelmed by the object hierarchy @_@

jmamma, 2020-06-15:
> ephemeral but then it persists between runs of MCLSeq::seq()?

yes. ephemeral in the sense that saving a track does not save that data.

---

 30 #define MD_TRACK_LEN (sizeof(GridTrack) + sizeof(MDSeqTrackData) + sizeof(MDMachine))

md_track.load_from_mem(); (1)
md_track.store_in_mem() ;   (2)

(1) and (2) only read/write MD_TRACK_LEN worth of data

(1) and (2) are used by GridTask to perform track/slot chaining, and this is where the corruption is likely occurring.

yatli, 2020-06-15:
So...

```
MDTrackLight = persistent data mapped from the grid file
MDTrackLight = MDSeqTrackData + MDMachine

MDSeqTrackData = persistent data of MD track sequence, pattern + lock

MDSeqTrack = MDSeqTrackData + runtime environment for MCLSeq
MDTrack = MDTrackLight + runtime environment for chain
```

?

jmamma, 2020-06-15:
yes, you got it.

with the correction:

MDTrackLight = GridTrack + MDSeqTrackData + MDMachine

MDTrack = MDTrack.data + MDTrackLight

yatli, 2020-06-15:
We've got to check the object layout. It may not be sequential

jmamma, 2020-06-15:
I've just written up a test scenario.

Replace GridTask::run();
and set row, to an appropriate row.
Then start sequencer.

Unfortunately, the test case is failing on master branch too. 
Maybe i've made an obvious error here.

```
void GridTask::run() {
  EmptyTrack empty_track;
  MDTrack *md_track = (MDTrack *)&empty_track;
  if (MidiClock.state != 2) {
    return;
  }
  GUI.removeTask(&grid_task);

  DEBUG_PRINT_FN();

  uint8_t row = 20; 

  uint16_t size = sizeof(GridTrack) + sizeof(MDSeqTrackData) + sizeof(MDMachine);

  for (uint8_t n = 0; n < 16; n++) {
    DEBUG_PRINTLN("loading");
    DEBUG_DUMP(n);
    md_track->load_track_from_grid( n, row, size);
    md_track->store_in_mem(n);
  }
  uint32_t start_clock = slowclock;
  while (clock_diff(start_clock, slowclock) < 2000) {
    GUI.loop();
  }
  EmptyTrack empty_track2;
  MDTrack *md_track2 = (MDTrack *)&empty_track2;

  for (uint8_t n = 0; n < 16; n++) {
    md_track->load_track_from_grid(n, row, size);
    md_track2->load_from_mem(n);
    if (uint16_t cbyte = memcmp(&(md_track), &(md_track2), size) != 0) {
      DEBUG_DUMP(n);
      DEBUG_PRINTLN("corruption");
      DEBUG_DUMP(cbyte);
    } else {
      DEBUG_DUMP(n);
      DEBUG_PRINTLN("load okay");
    }   
  }
  GUI.addTask(&grid_task);
}
```

jmamma, 2020-06-15:
Whoops. typo in memcmp.

I've patched the above.

Test passes on master

jmamma, 2020-06-15:
test passes in dev...

jmamma, 2020-06-15:
I'll try eating up the stack with an increasing array and see if corruption creeps in as memory usage increases.

yatli, 2020-06-15:
Here's the class layout dump for the relevant types:

```
Class MDSeqTrackData
   size=410 align=1
   base size=410 base align=1
MDSeqTrackData (0x0x5ddedc8) 0

Class SlideData
   size=15 align=1
   base size=15 base align=1
SlideData (0x0x5ddee38) 0

Class MDSeqTrack
   size=515 align=1
   base size=515 base align=1
MDSeqTrack (0x0x5e1aa00) 0
  MDSeqTrackData (0x0x5ddeee0) 0

Class MDTrackLight
   size=501 align=1
   base size=501 base align=1
MDTrackLight (0x0x5e6a6c0) 0
  GridTrack (0x0x5e6f310) 0
  Bank1Object<MDTrackLight, 0, 37680> (0x0x5e6f348) 0 empty

Class MDTrack
   size=1382 align=1
   base size=1382 base align=1
MDTrack (0x0x5e6a840) 0
  MDTrackLight (0x0x5e6a880) 0
    GridTrack (0x0x5e6f380) 0
    Bank1Object<MDTrackLight, 0, 37680> (0x0x5e6f3b8) 0 empty

```

jmamma, 2020-06-15:
> I'll try eating up the stack with an increasing array and see if corruption creeps in as memory usage increases.

I was able to use another 4480 bytes of stack RAM before megacommand crashed. No sign of corruption.

jmamma, 2020-06-15:
@yatli what do the hex numbers represent?

yatli, 2020-06-15:
MDTrackLight = GridTrack + MDSeqTrackData + MDMachine

Class MDTrackLight   size=501 align=1
Class MDSeqTrackData   size=410 align=1
Class GridTrack   size=20 align=1
Class MDMachine   size=71 align=1

Adds up correctly.

yatli, 2020-06-15:
ignore the hex numbers -- perhaps some kind of gcc internal representation

yatli, 2020-06-15:
MDSeqTrack = {fields} + MDSeqTrackData

Class MDSeqTrack   size=515 align=1
Class MDSeqTrackData size=410 align=1
fields = 45 + 4*15=105

adds up correctly.

jmamma, 2020-06-15:
thanks @yatli 

I have to call it a night.

Too many late ones.

yatli, 2020-06-15:
Regulation is required!
Good night.

jmamma, 2020-06-15:
@yatli The bug wasn't showing up in DEBUG_MODE.

Modifying GridTask::run() to oled_display.textbox("corruption");
i'm able to show data corruption when loading.

```
void GridTask::run() {
  EmptyTrack empty_track;
  MDTrack *md_track = (MDTrack *)&empty_track;
  if (MidiClock.state != 2) {
    return;
  }
  GUI.removeTask(&grid_task);

  DEBUG_PRINT_FN();

  // CHANGE ME
  uint8_t row = 20; 

  uint16_t size =
      sizeof(GridTrack) + sizeof(MDSeqTrackData) + sizeof(MDMachine);

  // Load slots from Grid and store in memory BANK2
  for (uint8_t n = 0; n < 16; n++) {
    md_track->load_track_from_grid(n, row, size);
    md_track->store_in_mem(n);
  }

  // Do nothing for 2 seconds.
  uint32_t start_clock = slowclock;
  while (clock_diff(start_clock, slowclock) < 2000) {
    GUI.loop();
  }
  EmptyTrack empty_track2;
  MDTrack *md_track2 = (MDTrack *)&empty_track2;

  // Compare slots from grid with load from memory BANK2
  for (uint8_t n = 0; n < 16; n++) {
    md_track->load_track_from_grid(n, row, size);
    md_track2->load_from_mem(n);
    if (uint16_t cbyte = memcmp((md_track), (md_track2), size) != 0) {
      char str[5];
      itoa(n, str, 10);
      oled_display.textbox("c ", str);

    } else {
    }   
  }
  GUI.addTask(&grid_task);
}
```

jmamma, 2020-06-15:
Corruption always happens on slot/track 12. regardless of row.

jmamma, 2020-06-15:
better check that my sram is okay...

jmamma, 2020-06-16:
@yatli 

Corruption is showing up in MASTER branch. but on slot 9 instead !!!

jmamma, 2020-06-16:
disabling interrupts when in  
md_track->store_in_mem(n);
md_track->load_from_mem(n);

corruption disappears ...

jmamma, 2020-06-16:
@yatli what do you think?

https://stackoverflow.com/questions/36729240/passing-argument-2-of-memcpy-discards-volatile-qualifier-from-pointer-target

```
103 ALWAYS_INLINE() extern inline void memcpy_bank1(volatile void *dst, volatile const void *src, uint32_t len) {
104   select_bank(1);
105   memcpy(dst, src, len);
106 }
107 
108 ALWAYS_INLINE() extern inline void put_byte_bank1(volatile uint8_t *dst, uint8_t byte) {
109   select_bank(1);
110   *dst = byte;
111 }
```

jmamma, 2020-06-16:
replaced memcpy above with

```
    while (len) {
    *((uint8_t *)dst++) = *((uint8_t *)src++);
    len--;
  }
```

still getting corruption.

looks like ISR block needed...

jmamma, 2020-06-16:
I have to put the ISR block in the store_in_mem(n) load_from_mem(n) functions.
adding it directiy in 
memcpy_bank1() does not resolve the corruption.

with ISR blocking we miss MIDI clock messages and the tempo goes out of sync 😭

yatli, 2020-06-16:
```
ALWAYS_INLINE() extern inline uint8_t switch_ram_bank(uint8_t x) {
  uint8_t old_bank = (uint8_t) (PORTL & PL6_MASK);

  if (x != old_bank) {
    PORTL ^= PL6_MASK;
    return old_bank;
  }
  return x;
}
```

put the ISR block here?

jmamma, 2020-06-16:
can't do that easily. switch_bank is called inside some ISR functions (MIDI ring buffer etc). if we re-enable ISR after block, it will exit ISR prematurely.

jmamma, 2020-06-16:
select_bank exclusively is called exclusively in iSR

with the exception of load_from_mem() / store_in_mem() which calls it indirectly via memcpy_bank1

There are two ISR routines that re-enable ISR during operation. I disabled re-enabling of ISR and the corruption is still there.

jmamma, 2020-06-16:
I patched switch_ram_bank to enable ISR blocking.
I then modifed memcpy_bank1 to use blocking.

```
ALWAYS_INLINE() extern inline uint8_t switch_ram_bank(uint8_t x, bool disable_isr = false) {
  
  USE_LOCK();
 if (disable_isr) {
  SET_LOCK();
  }
  uint8_t old_bank = (uint8_t) (PORTL & PL6_MASK);

  if (x != old_bank) {
    PORTL ^= PL6_MASK;
    return old_bank;
  }
  if (disable_isr) {
   CLEAR_LOCK();
  }
  return x;
}
```

corruption still occurs.

jmamma, 2020-06-16:
okay taking a break.

yatli, 2020-06-16:
how do you verify a corruption?

jmamma, 2020-06-16:
see my post above.

try yourself 

> @yatli The bug wasn't showing up in DEBUG_MODE.
> 
> Modifying GridTask::run() to call oled_display.textbox("corruption");
> i'm able to reliably show data corruption when loading.

yatli, 2020-06-16:
GridTask::run() allocates two empty tracks on stack.
Could it be too heavy?

jmamma, 2020-06-16:
let me know if instructions are no clear?

jmamma, 2020-06-16:
>> GridTask::run() allocates two empty tracks on stack.
Could it be too heavy?

This mimics what the real GridTask::run() function does. 

I did some stack testing last night. I was able to fill up another 4KB of stack in run() before crash.
This was with debug mode on. It seems to be timing related because it doesnt show up with DEBUG_MODE on

jmamma, 2020-06-16:
some more interesting results.

Adding interrupts in store_in_mem and load_from_mem, corruption disappears completely on grid page. but when I enter PageSelect corruption shows up.

```
template<class T, uint8_t n_column_base, uint32_t p_addr_base> 
class Bank1Object {
  public:
  inline bool store_in_mem(uint8_t column, uint32_t region = p_addr_base) 
  {
    cli();
    uint32_t pos = region + sizeof(T) * (uint32_t)(column - n_column_base);
    volatile uint8_t *ptr = reinterpret_cast<uint8_t *>(pos);
    memcpy_bank1(ptr, this, sizeof(T));
    sei();
    return true;
  }

  inline bool load_from_mem(uint8_t column, uint32_t region = p_addr_base)
  {
    cli();
    uint32_t pos = region + sizeof(T) * (uint32_t)(column - n_column_base);
    volatile uint8_t *ptr = reinterpret_cast<uint8_t *>(pos);
    memcpy_bank1(this, ptr, sizeof(T));
    sei();
    return true;
  }
};
```

yatli, 2020-06-16:
With the test fixture I'm able to get repro on master.

jmamma, 2020-06-16:
> With the test fixture I'm able to get repro on master.

at least we have a bit of level ground to stand on. last night it was shaky.

jmamma, 2020-06-16:
corruption appears on slot 9 in master, and slot 12 in dev.
By chance, when testing 2.70 for release, I noticed my patterns being messed up on slot 12.

yatli, 2020-06-16:
I don't see the pattern getting distorted but the text box does show up.
I'm using this instead of memcmp to probe the exact corruption location:

```c++

uint16_t compare_memory(void* _pa, void* _pb, uint16_t len)
{
  uint8_t *pa = (uint8_t*)_pa, *pb = (uint8_t*)_pb;
  for (uint16_t pos = 0; pos < len; ++pos) {
    if(*pa++ != *pb++) return pos;
  }
  return len;
}
```

yatli, 2020-06-16:
On master it's always track 9, offset 21

yatli, 2020-06-16:
the assertion chunk:

```c++
  // Compare slots from grid with load from memory BANK2
  for (uint8_t n = 0; n < 16; n++) {
    md_track->load_track_from_grid(n, row, size);
    md_track2->load_from_mem(n);
    uint16_t cbyte = compare_memory((md_track), (md_track2), size);
    if (cbyte != size) {
      char str[10];
      char pos[5];
      itoa(n, str, 10);
      itoa(cbyte, pos, 10);
      strcat(str, " ");
      strcat(str, pos);
      oled_display.textbox("HIT ", str);

    } else {
    }   
  }
```

jmamma, 2020-06-16:
> I don't see the pattern getting distorted but the text box does show up.
> I'm using this instead of memcmp to probe the exact corruption location:

The above GridTask::run() test doesn't actually load any sequencer data. So you wont see it in step page.

First loop:
Copy 16 tracks from SD card in to stack, then from stack in to BANK1

2nd loop
Copy SD card to stack, load BANK1 to stack. compare both slots..

yatli, 2020-06-16:
It's stable even if I crank the tempo up to 300

jmamma, 2020-06-16:
> It's stable

Yes. MCL has always been reliable and repeatable 😀

jmamma, 2020-06-16:
DEV Branch.

HIT: 12 62

jmamma, 2020-06-16:
FYI. memcmp

if return value > 0
``return == the first byte that does not match in both memory blocks has a greater value in ptr1``

yatli, 2020-06-16:
IIRC memcmp returns the diff value (*pa - *pb), not len

yatli, 2020-06-16:
yeah :D

yatli, 2020-06-16:
could that difference be from mdtrack layout change?

[75 more replies]

closed 2020-06-17 by jmamma