jmamma/MCL #121: Lock behavior/data changed in 2.7?

yatli, 2020-06-17:
When I load up the latest dev branch, some slots lose triggerless pitch locks

jmamma, 2020-06-17:
what do you mean lose? they dont play?

jmamma, 2020-06-17:
a4f47e2fbe69ffbcf296260c643918bd5eeeb9b9
and
87a0a926c053725af4f8b9e1e198d5986fa3f6c4

jmamma, 2020-06-17:
The change was intended to be:  if trig condition doesn't match. then lockless trig wont be sent.

jmamma, 2020-06-17:
found a somewhat related bug:
a1d7c620923fba6a9d48ff5ad8a9fd62b21ee2d5

yatli, 2020-06-17:
ooh I see what you mean. what I was expecting is that the trig is not conditionally fired but the lock is still sent. so it creates a kind of accent + porta if the conditional trig is not fired.

yatli, 2020-06-17:
Can we get that behavior also, like L2 vs. L2*, whereas L2* doesn't inhibit locks? I really love that porta sound :) :)

jmamma, 2020-06-17:
"what I was expecting is that the trig is not conditionally fired but the lock is still sent"
this was the old behaviour.

I can't recall my reason for wanting to change it.

I had a few patterns where I was expecting the new behaviour.

jmamma, 2020-06-17:
> I really love that porta sound :) :)

yeah the MD slides like a 303. the only 303 that counts.

"btw: please keep in mind that the only cool 303 is the 56303 ;-)"
sequencer.de

jmamma, 2020-06-17:
maybe I should revert the behaviour. and we can add 
L* modes in v3.0 when spec changes.

jmamma, 2020-06-20:
5d2d048fe282a88ae060f20a0e0dafaed277f952

not tested yet.

jmamma, 2020-06-20:
2762bcfa15c8e79e40e9400815cb82f81054798c

Offest L^ modes by 64.. this gives us room to pack more conditions in down the road.

closed 2020-06-21 by jmamma