jmamma/MCL #122: To be fixed.

jmamma, 2020-06-23:
[] - Display "POLY LINK" when changing parameters of linked poly tracked.
[] - Shift and reverse is not shifting/reversing slide mask.

yatli, 2020-06-23:
> Allow folder manipulation in load project page

folder manipulation?

jmamma, 2020-06-23:
cd
mkdir
etc.

albadort76, 2020-07-21:
Minor bug : in chromatic mode if the kit is changed in the MD, it's not following in the MC. Need to go out of the chromatic menu and then come back in for effective change.
MD 1.71, MC 2.70
Megacommand is fantastic :)

jmamma, 2020-07-28:
> Minor bug : in chromatic mode if the kit is changed in the MD, it's not following in the MC. Need to go out of the chromatic menu and then come back in for effective change.

Yeah, this is a known 'feature/issue'. We need to implement some changes on the MD side to alert the MC when the kit's machines are changed. 

> MD 1.71, MC 2.70
> Megacommand is fantastic :)

cheers. let us know if you run into anything else.

closed 2021-03-24 by jmamma