jmamma/MCL #137: New SDFat version

jmamma, 2020-11-21:
https://github.com/greiman/SdFat

@yatli ^

yatli, 2020-11-21:
So.. we can use it to reduce FW size?

jmamma, 2020-11-21:
yes, should be smaller. not sure how much.

jmamma, 2020-11-21:
i also discovered that there is a bug with arduino mega bootloader which caused the last 8KB of ROM space to be unusable.

yatli, 2020-11-21:
I followed the url you provided -- it looks like it's a FUSE setting that protects the region?

jmamma, 2020-11-21:
yes possibly.

yatli, 2020-11-21:
https://github.com/Optiboot/optiboot is an alternative bootloader with reduced size (512 bytes only)

Any mods done to the bootloader will require an ISP programmer, though.

yatli, 2020-11-21:
Oh wait...
This can be done with a loader program, self-programming the chip:

https://github.com/Optiboot/optiboot/wiki/InstallingOnChips#installing-using-a-specialized-bootloader-loader-sketch

jmamma, 2020-11-21:
I think you still need to go through ISP

yatli, 2020-11-21:
ow naw..

closed 2021-03-24 by jmamma