jmamma/MCL #140: ExtSeqTrack fixes required before 3.0

jmamma, 2020-12-04:
```
533 void ExtSeqTrack::seq() {
534   if (mute_until_start) {
535 
536     if ((clock_diff(MidiClock.div16th_counter, start_step) == 0)) {
537       if (start_step_offset > 0) {
538         start_step_offset--;
539       } else {
540         reset();
541       }
542     }
543   }
544   uint8_t timing_mid = get_timing_mid_inline();
545   if ((MidiUart2.uart_block == 0) && (mute_until_start == false) &&
546       (mute_state == SEQ_MUTE_OFF)) {
547 
548     // the range we're interested in:
549     // [current timing bucket, micro >= timing_mid ... next timing bucket, micro
550     // < timing_mid]
551 
552     uint16_t ev_idx, ev_end;
553 
554     // Locate CURRENT
555     locate(step_count, ev_idx, ev_end);  **<---- OPTIMIZE HIM**

We can keep track of current ev_idx. And then increase/decrease as items are added removed with ISR block.

```
 68   bool operator<(const ext_event_t &that) {
 69     // order by micro_timing
 70     if (this->micro_timing != that.micro_timing) {
 71       return this->micro_timing < that.micro_timing;
 72     }
 73     /*
 74     // order by lock
 75 
 76     if (this->is_lock != that.is_lock) {
 77       return this.is_lock;
 78     }
 79     */
 80     // off < on
 81     if (this->event_on != that.event_on) {
 82       return that.event_on;
 83     }
 84 
 85     return false;
 86   }
 87 };
```

Implement order by lock

jmamma, 2020-12-06:
Shift left and shift right are broken. Have been before since I was working on piano_roll.

jmamma, 2020-12-06:
velocity shifting was missing. 

still seems broken.

just to confirm @yatli we are shifting around visible length. not the entire data structure?

jmamma, 2020-12-06:
520ee3f

jmamma, 2020-12-06:
reverse has mysteriously disappeared from menu options... 
SEQ_MENU_REVERSE is not disabled anywhere...

jmamma, 2020-12-06:
If I add 

```
@@ -85,6 +85,8 @@ void SeqPage::init() {
   seq_menu_page.menu.enable_entry(SEQ_MENU_PIANOROLL, false);
   seq_menu_page.menu.enable_entry(SEQ_MENU_PARAMSELECT, false);
   seq_menu_page.menu.enable_entry(SEQ_MENU_SLIDE, false);
+  seq_menu_page.menu.enable_entry(SEQ_MENU_REVERSE, true);
+
```

I get a corrupt menu, and then lockup. 

@yatli can you investigate what's happening here.

yatli, 2020-12-06:
> just to confirm @yatli we are shifting around visible length. not the entire data structure?

IIRC yes

jmamma, 2020-12-06:
```
 33 class MenuBase {
 34 public:
 35   uint8_t entry_mask[2];

jmamma, 2020-12-06:
we've gone beyond 16bit.

lol.

yatli, 2020-12-06:
>   uint8_t entry_mask[2];

yatli, 2020-12-06:
ha you beat me to it

jmamma, 2020-12-06:
yes

jmamma, 2020-12-06:
it's a 32bit console era now

yatli, 2020-12-06:
hmm, how about splitting the seq page menu?

yatli, 2020-12-06:
let me prototype it (don't have MC at hand, but anyway I can try it)

jmamma, 2020-12-06:
I think we would be doubling up on menu items if we did that.

jmamma, 2020-12-06:
Running out of time for 3.0 release. I'm trying to wrap this one up ASAP.

jmamma, 2020-12-06:
```
 985   case DIR_REVERSE:
 986     uint16_t end = ev_end / 2;
 987     for (uint16_t i = 0; i < end; ++i) {
 988       auto tmp = events[i];
 989       auto j = ev_end - i - 1;
 990       events[i] = events[j];
 991       events[j] = tmp;
 992 
 993       // need to flip note on/off
 994       if (!events[i].is_lock) {
 995         events[i].event_on = !events[i].event_on;
 996       }
 997 
 998       if (!events[j].is_lock) {
 999         events[j].event_on = !events[j].event_on;
1000       }
1001     }
1002     for (uint8_t n = 0; n < length; n++) {
1003       uint8_t vel_tmp = velocities[n];
1004       uint8_t z = length - 1 - n;
1005       velocities[n] = velocities[z];
1006       velocities[z] = vel_tmp;
1007     }
1008     timing_buckets.reverse(length);
1009     break;
```
what's wrong here ^

yatli, 2020-12-06:
> Running out of time for 3.0 release. I'm trying to wrap this one up ASAP.

Ok, discarding.

yatli, 2020-12-06:
velocity is 0..127
event is 0..511

jmamma, 2020-12-06:
yes velocity has been separated. see above ^

yatli, 2020-12-06:
yep, but velocity reverse uses `length` which is for events.

jmamma, 2020-12-06:
I think the problem might be to do with microtiming ?

If I have the notes at full step length it's okay ?

yatli, 2020-12-06:
microtiming not reversed!

yatli, 2020-12-06:
try this:

```c++
  case DIR_REVERSE:
    uint16_t end = ev_end / 2;
    uint8_t timing_mid_2x = get_timing_mid() * 2;
    for (uint16_t i = 0; i < end; ++i) {
      auto tmp = events[i];
      auto j = ev_end - 1 - i;
      events[i] = events[j];
      events[j] = tmp;

      // need to flip note on/off
      if (!events[i].is_lock) { events[i].event_on = !events[i].event_on; }
      if (!events[j].is_lock) { events[j].event_on = !events[j].event_on; }

      // reverse microtiming
      events[i].micro_timing = timing_mid_2x - events[i].micro_timing;
      events[j].micro_timing = timing_mid_2x - events[j].micro_timing;
    }
    for (uint8_t n = 0; n < length; n++) {
      uint8_t vel_tmp = velocities[n];
      uint8_t z = length - 1 - n;
      velocities[n] = velocities[z];
      velocities[z] = vel_tmp;
    }
    timing_buckets.reverse(length);
    break;
```

yatli, 2020-12-06:
emm, 

```c++
    for (uint8_t n = 0; n < length; n++) {
      uint8_t vel_tmp = velocities[n];
      uint8_t z = length - 1 - n;
      velocities[n] = velocities[z];
      velocities[z] = vel_tmp;
    }
```

you end up swapping _twice_

jmamma, 2020-12-06:
It looks better on the editor, but no audible note playback on reverse.

Also now my shift menu has disappeared

jmamma, 2020-12-06:
oh should be length/2

yatli, 2020-12-06:
git pull it

jmamma, 2020-12-06:
> git pull it

?

yatli, 2020-12-06:
I pushed one commit for that.

jmamma, 2020-12-06:
which branch?

jmamma, 2020-12-06:
i'm working in dev

yatli, 2020-12-06:
emm, dev?

jmamma, 2020-12-06:
commit id ?

yatli, 2020-12-06:
0931bc78d249b1eee9f3bba07663f1ddd0f12dcf

jmamma, 2020-12-06:
I merged your commit with mine. 

Reverse playback is still broken. some notes not playing.

jmamma, 2020-12-06:
And shift menu has disappeared since increase menu_mask to 32bit.

yatli, 2020-12-06:
05e246da

yatli, 2020-12-06:
> Reverse playback is still broken. some notes not playing.

What if without microtiming reverse? Just no velocity?

jmamma, 2020-12-06:
yep. midi monitor showing 0 vel

jmamma, 2020-12-06:
vel_tmp is wrong.

jmamma, 2020-12-06:
i've well and truly exhausted my capabality this weekend. day job in 8 hours.

jmamma, 2020-12-06:
if you can get the A4 conversion problem sorted that would be a big help. that way I can tick off project conversion.

jmamma, 2020-12-06:
i'm off. 🛌

yatli, 2020-12-06:
> if you can get the A4 conversion problem sorted that would be a big help. that way I can tick off project conversion.

roger roger.

good night!

closed 2020-12-09 by jmamma