jmamma/MCL #141: A4Sound conversion 2.70 -> 3.0

jmamma, 2020-12-04:
Currently sound conversion is not working.

yatli, 2020-12-07:
@jmamma is `master` 2.7? I'd like to do some A/B testing.

yatli, 2020-12-07:
193961226bf35c791f1287448994c63a768df14f fixed sound conversion on my side.

jmamma, 2020-12-07:
Yes master is 2.7

I will test this soon.

jmamma, 2020-12-07:
Yes looking good!

Thankyou :rocket:

jmamma, 2020-12-07:
Didnt take long to find another bug... midi channel becomes corrupt during transition_load. okay during load_immediate.

time for dinner.

yatli, 2020-12-07:
Can confirm. Chaining while playback result in corrupted sound/sequence.

yatli, 2020-12-07:
The corruption is only triggered when M5 and M6 are involved?

A4: I told you my name is Analog *FOUR*

jmamma, 2020-12-07:
Hmm. This might be a different problem. In my converted project only tracks 1-4 are active.

jmamma, 2020-12-07:
Ah I think you're right. tracks 5 +6 get loaded anyway.

Thought I fixed this!!

yatli, 2020-12-07:
Yeah, could you try to chain all the tracks with TI?
I have problem with "chain group", but if I select the 4 tracks with TI it's fine

jmamma, 2020-12-07:
```
 34 void A4Class::init_grid_devices() {
 35   uint8_t grid_idx = 1;
 36 
 37   for (uint8_t i = 0; i < NUM_EXT_TRACKS; i++) {
 38     uint8_t track_type= EXT_TRACK_TYPE;
 39 
 40     if (i < NUM_A4_SOUND_TRACKS) {
 41        track_type = A4_TRACK_TYPE;
 42     }
 43 
 44     add_track_to_grid(grid_idx, i,  &(mcl_seq.ext_tracks[i]), track_type);
 45   }
 46 
 47 }
```
I did patch the device/slot allocation.

The problem must be with Chain mode loading the track as A4 instead of EXT

yatli, 2020-12-07:
Very likely...

A4:
[image]

jmamma, 2020-12-07:
no more bugs plz.

jmamma, 2020-12-07:
```
transition_load_ext
34130 <- seq ptr
17 <- slot number
1 <-track number
here
send a4 sound
transition_load_ext
36138
18
2
here
send a4 sound <--- A4 Tracks transmit sound.
transition_load_ext
38146
19
3
transition_load_ext 
40154
20
4
transition_load_ext
42162
21
5
```

Appears to be loading correctly during transition. or at least the correct track type

jmamma, 2020-12-07:
actually that is misleading... send_machine[n] determines if sound data is sent.
A4Track:transition_load calls ExtTrack::transition_load

jmamma, 2020-12-07:
i.e

```
134       auto *pmem_track = empty_track.load_from_mem(track_idx, track_type);
135 
136       if (pmem_track != nullptr) {
137         if (mcl_actions.send_machine[n] == 0) {
138         pmem_track->transition_send(track_idx, n);
139         }
140         pmem_track->transition_load(track_idx, seq_track, n);
141         grid_page.active_slots[n] = slots_changed[n];
142       }
143 
```

jmamma, 2020-12-07:
```
291   for (uint8_t n = 0; n < NUM_SLOTS; ++n) {
292 
293     SeqTrack *seq_track =
294         get_dev_slot_info(n, &grid_idx, &track_idx, &track_type, &dev_idx);
295     proj.select_grid(grid_idx);
296     if ((slot_select_array[n] == 0) || (track_type == 255)) {
297       // Ignore slots that are not device supported.
298       slot_select_array[n] = 0;
299       continue;
300     }
301     auto device_track = empty_track.load_from_grid(track_idx, row);
302     if (device_track == nullptr || device_track->active != track_type) {
303       empty_track.clear();
304       device_track = device_track->init_track_type(track_type);
305       send_machine[n] = 1;
306     } else {
307       send_machine[n] = 0;
308     }
309     DEBUG_PRINTLN("track type load");
310     DEBUG_DUMP(n);
311     DEBUG_DUMP(track_idx);
312     DEBUG_DUMP(track_type);
313     device_track->store_in_mem(track_idx);
314   }
```

```
track type load
18
2
5
2094
track type load
19
3
5 <- A4_TRACK_TYPE
2094
track type load
20
4
6
2094
track type load
21
5
6 <- EXT_TRACK_TYPE
```

jmamma, 2020-12-07:
device_track->active = 6 as well... hmm

jmamma, 2020-12-07:
gridtask loads the track from mem as EXT_TRACK_TYPE.

jmamma, 2020-12-07:
That's weird. It's only broken if 4 + 5 are transmitted together.
If I transmit 4 and then transmit 5 it's okay.

Transmit them together and SOUND4 appears as SOUND1 on the A4.

jmamma, 2020-12-07:
get_region() missing ??

jmamma, 2020-12-07:
./ExtTrack.h:  virtual uint32_t get_region() { return BANK1_A4_TRACKS_START; }

hmm nope.

jmamma, 2020-12-07:
76 constexpr size_t BANK1_A4_TRACKS_START = BANK1_AUX_TRACKS_START + FX_TRACK_LEN * NUM_FX_TRACKS;

this is wrong. but i dont think it's related to this issue.

 56 constexpr size_t FX_TRACK_LEN = 43;

is too small. MDLFOTrack is the largest type..

jmamma, 2020-12-07:
also 

./MCLMemory.h:constexpr size_t NUM_LFO_TRACKS = 2;

should be 1

jmamma, 2020-12-07:
We should probably create BANK1 buckets for each of the AUXTrack types.

jmamma, 2020-12-07:
🛌

jmamma, 2020-12-07:
constexpr size_t GRID2_TRACK_LEN = 2094;

I think this is too small ? @yatli can you check these

jmamma, 2020-12-07:
KWelcome to MegaCommand Live
3000
534
2094
1754
2094

yatli, 2020-12-07:
Sire, I'll check it tomorrow.

yatli, 2020-12-08:
These guards ain't for nothing: `    static_assert(sizeof(A4Track) <= GRID2_TRACK_LEN);`

jmamma, 2020-12-08:
```
76 constexpr size_t BANK1_A4_TRACKS_START = BANK1_AUX_TRACKS_START + FX_TRACK_LEN * NUM_FX_TRACKS;

this is wrong. but i dont think it's related to this issue.

56 constexpr size_t FX_TRACK_LEN = 43;

is too small. MDLFOTrack is the largest type..
```

Need to fix these up. 

The original bug feels like an overflow... but haven't detected the cause yet.

jmamma, 2020-12-08:
Think I found it...

314     device_track->store_in_mem(track_idx);

track_idx = 5 or 6

This is a new region. so they should be 0, 1

jmamma, 2020-12-08:
no wait. this is wrong. track_idx is okay for ExtTrack.

need to check track_idx for the AUXTracks.

jmamma, 2020-12-08:
```
 129   add_track_to_grid(grid_idx, MDFX_TRACK_NUM, &(mcl_seq.aux_tracks[0]), MDFX_TRACK_TYPE, GROUP_AUX);
 130   add_track_to_grid(grid_idx, MDROUTE_TRACK_NUM, &(mcl_seq.aux_tracks[1]), MDROUTE_TRACK_TYPE, GROUP_AUX);
 131   add_track_to_grid(grid_idx, MDLFO_TRACK_NUM,  &(mcl_seq.aux_tracks[2]), MDLFO_TRACK_TYPE, GROUP_AUX);
 132   add_track_to_grid(grid_idx, MDTEMPO_TRACK_NUM, &(mcl_seq.aux_tracks[3]), MDTEMPO_TRACK_TYPE, GROUP_TEMPO);
 133 }
```

164   void add_track(uint8_t track_idx, uint8_t slot_number, SeqTrack *seq_track, uint8_t track_type, uint8_t group_type = GROUP_DEV) {

```
 34 constexpr size_t NUM_FX_TRACKS = 2; <--- He's wrong.
 35 constexpr size_t MDFX_TRACK_NUM = 12; //position of MDFX track in grid
 36 constexpr size_t MDLFO_TRACK_NUM = 13; //position of MDLFO track in grid
 37 constexpr size_t MDROUTE_TRACK_NUM = 14; //position of MDROUTE track in grid
 38 constexpr size_t MDTEMPO_TRACK_NUM = 15; //position of MDTEMPO track in grid
```

jmamma, 2020-12-08:
yep track_idx is wrong too for

 device_track->store_in_mem(track_idx);

jmamma, 2020-12-08:
now how to fix this mess 🤔

yatli, 2020-12-08:
I was looking at these lines and wondering what is an AUX track and what FX.

jmamma, 2020-12-08:
Well I only had 1 FX track to begin with, the other 3 tracks came later. So I renamed the new group to "AUX (auxiliary) tracks"

There's a parent device class for them now. AUXTrack.h

jmamma, 2020-12-08:
7a493c3d0c3b6920db451fd434a4ba313146097d

jmamma, 2020-12-08:
More work needs to be done.

MCLActions::get_dev_slot_info(uint8_t slot_number, uint8_t *grid_idx,
 89                                         uint8_t *track_idx, uint8_t *track_type,
 90                                         uint8_t *dev_idx, uint8_t *group_type = nullptr)

Needs to be turned into:

MCLActions::get_dev_slot_info(uint8_t slot_number, uint8_t *grid_idx,
 89                                         uint8_t *track_idx, uint8_t *track_type,
 90                                         uint8_t *dev_idx, uint8_t *mem_slot_idx, uint8_t *group_type = nullptr)

But now it's really ugly.

uint8_t grid_idx, track_idx, track_type, dev_idx, mem_slot_idx, group_type;

663       SeqTrack *seq_track =
664           get_dev_slot_info(n, &grid_idx, &track_idx, &track_type, &dev_idx, &mem_slot_idx, &group_type);

jmamma, 2020-12-08:
Refactor?

GridDeviceTrack *p = get_dev_slot_info(n, &grid_idx, &dev_Idx);

SeqTrack *seq_track = p->seq_track;
track_idx = p->track_idx
etc...

yatli, 2020-12-08:
```c++
class DeviceSlotInfo {
public:
  uint8_t grid_idx;
  uint8_t track_idx;
  uint8_t track_type;
  uint8_t dev_idx;
  uint8_t group_type;
  SeqTrack* seq_track;
};

bool get_dev_slot_info(uint8_t slot_number, DeviceSlotInfo* pinfo);

// example:

  for (i = 0; i < NUM_SLOTS; i++) {
    if (slot_select_array[i] > 0 && get_dev_slot_info(i, &dev_info)) {
      save_dev_tracks[dev_idx] = true;
    }
  }

```

jmamma, 2020-12-08:
i'm working through the code now.

```
    GridDeviceTrack *gdt = get_grid_dev_track(i, &track_idx, &dev_idx);
    uint8_t grid_idx = get_grid_idx(i);
    proj.select_grid(grid_idx);

    if ((slot_select_array[n] == 0) || (gdt != nullptr)) {
```

yatli, 2020-12-08:
I'll push mine to another branch for your review.

yatli, 2020-12-08:
https://github.com/jmamma/MIDICtrl20_MegaCommand/pull/145

jmamma, 2020-12-08:
shit.

jmamma, 2020-12-08:
```
224       midi_active_peering.get_device(UART2_PORT)->asElektronDevice(),
225   å;
226   if ((mcl_cfg.chain_mode > 0) && (MidiClock.state == 2)) ä
227     for (uint8_t i = 0; i < NUM_DEVS; ++i) ä
228       if (elektron_devsÄiÅ != nullptr &&
229           elektron_devsÄiÅ->canReadWorkspaceKit()) ä
230         auto kit = elektron_devsÄiÅ->getKit();
231         if (kit != nullptr &&
232             elektron_devsÄiÅ->currentKit != kit->getPosition()) ä
233           elektron_devsÄiÅ->getBlockingKit(0x7F);
234         å
235       å
236     å
237     prepare_next_chain(row, slot_select_array);
238     return;
239   å
240   for (uint8_t i = 0; i < NUM_DEVS; ++i) ä
241     if (elektron_devsÄiÅ != nullptr &&
242         elektron_devsÄiÅ->canReadWorkspaceKit()) ä
243       elektron_devsÄiÅ->getBlockingKit(0x7F);
244     å
245   å
```

my vim just destroyed all my changes.

jmamma, 2020-12-08:
F@#$@$@#$@#$

jmamma, 2020-12-08:
hmm. i think it's just the editor being broken. file is still okay 🤞

jmamma, 2020-12-08:
@yatli  I looked at your implementation.

I think mine will use less progmem in the end ?

1539ad6

yatli, 2020-12-08:
> my vim just destroyed all my changes.

my horror story from 2012:

I was trying to switch to Emacs with the `evil` layer -- after a month or so I got very comfortable with it, and it's got elisp, embedded graphics and all the goodies...
Until my `init.el` was destroyed by emacs itself. Back to square one 0.0

jmamma, 2020-12-08:
just an automated reset. nothing to worry about.

emacs is a beast I'm too afraid to get in to.

jmamma, 2020-12-08:
d7bd6c58005101d79176c46bff030969af10bbf4 <- I added the track sizes.

We just need to create the 4 regions for the 4 AUX tracks, then update the classes for get_region. 

I need to grab some food.

yatli, 2020-12-08:
You're doing great work.. Thanks!!

I'll get to the new static assertions for AUX

yatli, 2020-12-08:
MCLActionsEvents.cpp, L37:
`  for (uint8_t n = 0; n < NUM_TRACKS; n++) {`
Just instrument tracks, no AUX?

jmamma, 2020-12-08:
good find. that's a bug. should be NUM_SLOTS

closed 2020-12-09 by jmamma