jmamma/MCL #142: Understand the bootloader limitations

yatli, 2020-12-05:
The bootloader for arduino 2560 can be located here: `avr/bootloaders/stk500v2/`
To compile, it should be set up like this (extracted from the makefile):
```
-funsigned-char -funsigned-bitfields -fpack-struct -fshort-enums -fno-jump-tables
-Wall -Wstrict-prototypes
-DBOOTLOADER_ADDRESS=3E000
-D_MEGA_BOARD_
-DMCU=atmega2560
-DF_CPU=16000000
-D__AVR_ARCH__=__AVR_ATmega2560__
-D__AVR_ATmega2560__
```

The boot section size is pre-programmed in the fuse bits. For Arduino Mega 2560, it is programmed to have 8192 words, which is 16KB.
```c
//#define BOOTSIZE 1024
#if FLASHEND > 0x0F000
	#define BOOTSIZE 8192
#else
	#define BOOTSIZE 2048
#endif

#define APP_END  (FLASHEND -(2*BOOTSIZE) + 1)

```

Therefore, the application section is `[0, 3C000) / [0, 245760)` -- that's the limit in the bootloader code.

This contradicts `boards.txt` in the core: `mega.menu.cpu.atmega2560.upload.maximum_size=253952`, which assumes that the bootloader takes 4K words, 8KB.

Interestingly, the BOOTSZ fuse bits for arduino specifies:

[image]
with 8KB as the maximum, so 16KB is an impossible value.

However, there are some early ATMega2560 batches with the defunct/workaround:
[image]

So that may be why the stk500v2 bootloader decided to do that.

If we switch to optiboot, we can reclaim 15.5KB of PROGMEM.

jmamma, 2020-12-05:
Great work. 

So we switch to optiboot for development. that will give us extra headroom for debug. 

245760 being our max binary size for release.

yatli, 2020-12-05:
Let me convert this to a doc.

closed 2020-12-05 by yatli