jmamma/MCL #144: RAM Page slice locks are not being set correctly

jmamma, 2020-12-07:


jmamma, 2020-12-07:
@yatli 
```
444 bool MDSeqTrack::set_track_locks_i(uint8_t step, uint8_t lockidx,
445                                    uint8_t value) {
446   auto lock_slot = get_lockidx(step, lockidx);
447   if (lock_slot == NUM_MD_LOCK_SLOTS) {
448     auto idx = get_lockidx(step);
449     auto nlock = popcount(steps[step].locks & ((1 << lockidx) - 1));
450     lock_slot = idx + nlock;
451 
452     if (lock_slot >= NUM_MD_LOCK_SLOTS) {
453       return false; // memory full!
454     }
455 
456     memmove(locks + lock_slot + 1, locks + lock_slot,
457             NUM_MD_LOCK_SLOTS - lock_slot - 1);
458     if (step < step_count) { cur_event_idx++; }
459     steps[step].locks |= (1 << lockidx);
460   }
461   locks[lock_slot] = min(value, 127);
462   steps[step].locks_enabled = true;
463   return true;
```

line 449. is this a bug?

jmamma, 2020-12-07:
```
185 bool RAMPage::slice(uint8_t track, uint8_t linked_track) {
186   uint8_t model = (MD.kit.models[track]);
187 
188   if (grid_page.active_slots[track] != SLOT_RAM_PLAY) {
189     return false;
190   }
191   uint8_t slices = 1 << encoders[2]->cur;
192 
193   uint8_t sample_inc = 128 / slices;
194   uint8_t track_length = encoders[3]->cur * 4;
195   uint8_t step_inc = track_length / slices;
196   bool clear_locks = true;
197   bool send_params = false;
198 
199   auto &trk = mcl_seq.md_tracks[track];
200   auto &ln_trk = mcl_seq.md_tracks[linked_track];
201   trk.clear_track(clear_locks, send_params);
202 
203   trk.locks_params[0] = ROM_STRT + 1;
204   trk.locks_params[1] = ROM_END + 1;
205   uint8_t mode = encoders[1]->cur;
206   DEBUG_PRINTLN("slices");
207   DEBUG_PRINTLN(slices);
208   DEBUG_PRINTLN(step_inc);
209   for (uint8_t s = 0; s < slices; s++) {
210     uint8_t n = s * step_inc;
211 
212     if ((linked_track < track) || (linked_track == 255)) {
213       trk.steps[n].trig = true;
214     }
215 
216     trk.timing[n] = trk.get_timing_mid();
217     if (linked_track < track) {
218       trk.set_track_locks_i(n, 0, ln_trk.get_track_lock(n, 0));
219       trk.set_track_locks_i(n, 1, ln_trk.get_track_lock(n, 1));
220     } else if (RAMPage::slice_modes[page_id] == 0) {
221       DEBUG_PRINTLN("slice here");
222       trk.set_track_locks_i(n, 0, sample_inc * s + 0);
223       auto val = (sample_inc) * (s + 1) + 0;
224       trk.set_track_locks_i(n, 1, val);
225       DEBUG_DUMP(val);
```

224 trk.set_track_locks_i(n, 1, val); memory is not persisting. Why?

yatli, 2020-12-07:
So, first lock set ok, but second one (id=1) doesn't?

jmamma, 2020-12-07:
sorry. was meaning to close this one. It's fixed in latest commits.

closed 2020-12-07 by jmamma