jmamma/MCL #154: Assortment of recent bugs

yatli, 2021-02-26:
- Chain mode isn't effective
- LEV of all tracks stuck at 127 on chain write
- Chain write do not reliably activate all selected tracks
- Chain write on a tight time budget result in off-beat tracks. The problem goes away if I stop/restart playback.
- Save results in wrong kit name

yatli, 2021-02-26:
```
C:\Users\Yatao\Documents\Arduino\hardware\MIDICtrl20_MegaCommand\avr\cores\megacommand\MD\MDMessages.cpp:214:57: warning: left shift count >= width of type [-Wshift-count-overflow]
 uint8_t MDMachine::get_model() { return model &= ~(1 << 17); }
```

jmamma, 2021-02-27:
```C:\Users\Yatao\Documents\Arduino\hardware\MIDICtrl20_MegaCommand\avr\cores\megacommand\MD\MDMessages.cpp:214:57: warning: left shift count >= width of type [-Wshift-count-overflow]
 uint8_t MDMachine::get_model() { return model &= ~(1 << 17); }```

Bit 17 is set as tonality. So we must clear it when returning model type ??

jmamma, 2021-02-27:
```Chain write on a tight time budget result in off-beat tracks. The problem goes away if I stop/restart playback.```

Yeah this one is on the list.

jmamma, 2021-02-27:
```LEV of all tracks stuck at 127 on chain write```

This is expected behaviour. CONFIG -> MACHINEDRUM -> NORMALIZE.
(LEV is boosted to 127, and VOL is lowered to compensate.)

jmamma, 2021-02-27:
void m_putc(uint8_t *msg, uint16_cnt) { ... }
ALWAYS_INLINE() void sendRaw(const uint8_t *msg, uint16_t cnt) { m_putc(msg, cnt); }

Is this conversion bad practice? I've got an issue with this function not working.

yatli, 2021-02-27:
>  uint8_t MDMachine::get_model() { return model &= ~(1 << 17); }
> 
> Bit 17 is set as tonality. So we must clear it when returning model type ??

uint8_t return type will automatically clear that

yatli, 2021-02-27:
> This is expected behaviour. CONFIG -> MACHINEDRUM -> NORMALIZE.

I see. I think I hear the difference because I have an LFO on VOL -- should turn normalization off.

jmamma, 2021-02-27:
uint8_t return type will automatically clear that

good point.

jmamma, 2021-02-27:
I see. I think I hear the difference because I have an LFO on VOL -- should turn normalization off.

The is normalisation code should scale LFOs accordingly, but i think the relationship is not linear so there may be some differences. 

```
11 void MDMachine::scale_vol(float scale) {
 12   DEBUG_PRINT_FN();
 13   params[MODEL_VOL] = (uint8_t)((float)params[MODEL_VOL] * scale);
 14   if (params[MODEL_VOL] > 127) {
 15     params[MODEL_VOL] = 127;
 16   }
 17   if ((lfo.destinationParam == MODEL_VOL) && (lfo.destinationTrack == track)) {
 18     params[MODEL_LFOD] = (uint8_t)((float)params[MODEL_LFOD] * scale);
 19 
 20     if (params[MODEL_LFOD] > 127) {
 21       params[MODEL_LFOD] = 127;
 22     }
 23     lfo.depth = params[MODEL_LFOD];
 24   }
 25 }
 26 
 27 float MDMachine::normalize_level() {
 28   DEBUG_PRINT_FN();
 29   if (level == 127) {
 30     return 1.0;
 31   }
 32 
 33   float scale = (float)level / (float)127;
 34   level = 127;
 35 
 36   scale_vol(scale);
 37   return scale;
 38 }
```

jmamma, 2021-02-27:
Seems like the timing issue only occurs when MIDI tracks on grid 2 are loaded.

jmamma, 2021-02-27:
3ad60ec

Timing feels good now. Can you confirm @yatli

jmamma, 2021-02-27:
```
Save results in wrong kit name
In X.04 with trig chain of >=3 tracks, sometimes one of the tracks is double-triggered (b-boom)
```

I can't replicate these.

I also cant replicate your merge bug with GND-X machines.

yatli, 2021-03-01:
I can't repro the trig chain problem now.

yatli, 2021-03-01:
Another bug -- chaining while playing result in wrong automation -- I have a track without slides, but chaining to it result in a knob being modulated from 0 to 127 and back again rapidly.

jmamma, 2021-03-01:
LFO ?

yatli, 2021-03-01:
oh didn't thought of that.
But when I stop/start it's gone

jmamma, 2021-03-01:
Can you replicate the chain movement that got you in to the state?

jmamma, 2021-03-03:
@yatli 
```
/Applications/Arduino_nightly.app/Contents/Java/hardware/MIDICtrl20_MegaCommand/avr/cores/megacommand/CommonTools/RingBuffer.h:111:16: error: invalid conversion from 'volatile void*' to 'void*' [-fpermissive]
     memcpy(buf + wr, src, s * sizeof(C));
            ~~~~^~~~
```
?

yatli, 2021-03-03:
Add `-fpermissive` to g++ flags

yatli, 2021-03-03:
FYI current arduino platform.txt options:

```
compiler.cpp.flags=-std=gnu++1z -c -Os {compiler.warning_flags} -MMD -Wno-attributes -Wno-write-strings -fpermissive -fno-exceptions -fno-threadsafe-statics -fno-rtti -ffunction-sections -fdata-sections -flto=jobserver -Wl,--gc-section,--defsym=__stack=0x8021ff,--section-start,.data=0x802100,--defsym=__heap_end=0x80ffff,--relax -I{build.core.path}/Midi -I{build.core.path}/GUI -I{build.core.path}/CommonTools -I{build.core.path}/Elektron -I{build.core.path}/MD -I{build.core.path}/MidiTools -I{build.core.path}/Sequencer -I{build.core.path}/MNM -I{build.core.path}/PageTools -I{build.core.path}/MDSketches -I{build.core.path}/MDPages -I{build.core.path}/A4 -I{build.core.path}/SPI -I{build.core.path}/Wire -I{build.core.path}/Adafruit-GFX-Library -I{build.core.path}/Adafruit_SSD1305_Library -I{build.core.path}/SdFat -I{build.core.path}/SDCard -I{build.core.path}/MCL
```

jmamma, 2021-03-03:
nice

yatli, 2021-03-05:
Converted the X tunnel song to MCL -- timing is good. Also tight chain writes are robust! 🎖

yatli, 2021-03-05:
Trying to extract that project for you via MegaCom:

[image]

yatli, 2021-03-05:
MegaCom behavior is interesting -- matches the design goals -- more bandwidth on simpler pages (17KB/s on MIXER), and "nicer" on fancy pages (7KB/s on GRID)

yatli, 2021-03-05:
[X_mcl.zip](https://github.com/jmamma/MIDICtrl20_MegaCommand/files/6088062/X_mcl.zip)

yatli, 2021-03-05:
hmm I still cannot activate chain mode

jmamma, 2021-03-05:
Haven't even had a chance to look at megacom yet. you'll need to recreate the binaries because last time I tried they didnt work.

I'll be heading off for a few hours.  back later tonight.

jmamma, 2021-03-05:
> hmm I still cannot activate chain mode

what do you mean?

yatli, 2021-03-05:
Set row 0 to chain to row 1, manual
Chain write row 0, expecting it to jump to row 1 after playback
Not jumping, not displaying the estimated jump time below current time

jmamma, 2021-03-05:
there used to be a chain bug on row 0. maybe it came back.

yatli, 2021-03-05:
Tried row 1 no banana

jmamma, 2021-03-05:
Did you set chain mode to AUTO

jmamma, 2021-03-05:
Loops need to be greater than 0 per slot

jmamma, 2021-03-05:
@yatli can you double check and report back.

jmamma, 2021-03-05:
⏳

yatli, 2021-03-05:
Woke up

jmamma, 2021-03-05:
Wake up Samurai, we've got a firmware to burn.

yatli, 2021-03-05:
Just tested -- chain update on any row, loop=1, row=next, cannot activate.
Side effect is, kit name is lost after a load.

jmamma, 2021-03-05:
Chain mode ?

yatli, 2021-03-05:
Yes, the button 3 menu, use encoder3 to select all tracks, set row to next, set loop to 1, release button 3 -- kit name is now blank.

jmamma, 2021-03-05:
No i mean... from grid page.. hold shift 2. Set CHAIN mode from MANUAL to AUTO

yatli, 2021-03-05:
It works!
AUTO never worked for me before -- I was sticking with MANUAL all along

jmamma, 2021-03-05:
haha 😆

jmamma, 2021-03-05:
I think MCL 3.10 is done. I've updated the changelog.

jmamma, 2021-03-05:
That means it's time.

yatli, 2021-03-05:
[image]

yatli, 2021-03-05:
Yahoooo

jmamma, 2021-03-05:
preparing binaries...

jmamma, 2021-03-05:
Ready to go here. I sent you a message elsewhere on github for your verification .

jmamma, 2021-03-05:
okay tested here.

🚀

jmamma, 2021-03-05:
Going live...

jmamma, 2021-03-05:
https://github.com/jmamma/MIDICtrl20_MegaCommand/releases/tag/3.10

closed 2021-03-24 by jmamma