jmamma/MCL #160: Deprecate Param/Locks page?

jmamma, 2021-04-07:
Do we need this anymore?

Does it afford us any advantages?

yatli, 2021-04-07:
- UI for cond trig
- UI for lock assignment (unique)

yatli, 2021-04-07:
But I guess that's all there's left.

yatli, 2021-04-07:
Also, is there possibility that we will support non-MD device on GridA?

jmamma, 2021-04-07:
> * UI for cond trig
> * UI for lock assignment (unique)

can already edit the various step masks from the StepEdit page. Or did you mean something else?

jmamma, 2021-04-07:
> Also, is there possibility that we will support non-MD device on GridA?

It's technically possible. The implementation is flexible. What did you have in mind ?

yatli, 2021-04-07:
The more I play with MC+MnM, the more I want P-Lock instead of event drawing.

jmamma, 2021-04-07:
make up your mind 😀

yatli, 2021-04-07:
My current setup:

[image]

Sweet spot for me.

jmamma, 2021-04-07:
wheres the MNM ?

jmamma, 2021-04-07:
[image]

This has been my setup for the past 3 weeks.  I miss making music 😭

yatli, 2021-04-07:
oww.

yatli, 2021-04-07:
> wheres the MNM ?

MnM/A4 cannot stand that they cannot be P-Locked so they paired up and ran away to the side desk.

yatli, 2021-04-07:
OT: why me in box??

jmamma, 2021-04-07:
🗑️

yatli, 2021-04-08:
> 🗑️

OT: me box, me box

jmamma, 2021-08-22:
deprecated

closed 2021-08-22 by jmamma