jmamma/MCL #164: Tonal setting not saved as part as sound in Sound Manager

larskinn, 2021-04-21:
Steps to reproduce:
----
1. Create a sound in e.g. POS 01-BD in a kit on the Machinedrum. Set it to be Tonal.
2. Save the sound using Sound Manager on MCL
3. Change to e.g POS 02-SD on the Machinedrum
4. Load the sound you saved in step 2
5. Listen to sound, it will be pitched differently from the sound in POS 01-BD
6. Edit sound, see that it is set to Default instead of Tonal

Actual result
----
The loaded sound is always set to Default, no matter what the sound was saved as, or what sound is already in the position you're loading the sound into.

Expected result
----
The loaded sound has the same Tonal/Default setting as it had when it was saved

Version info
----
Machinedrum hardware: SPS-1 MKII (non-UW, non-plus)
Machinedrum firmware: X.04
MCL: 3.10d

jmamma, 2021-04-23:
will investigate.

jmamma, 2021-04-24:
fixed, will be in next release.

closed 2021-04-24 by jmamma