jmamma/MCL #170: Firmware 4.0 Grid load page instead of Save page when several slots selected using MD Fn + Yes

albadort76, 2021-08-24:
Hi there! Thx very much for this incredible update.
I noticed something this afternoon :
When in Grig page if you select several slots with Md arrows and then you want to save with Fn + Yes it’s opening the Load page instead of the Save page expected.
i find an other way to save several slots in the grid by using No, when you push again No, Slots are saved if you change something in the page (jump, loop, Row etc…)
Can you replicate?
Just a question : if i load a 24 bits, 48khz .wav file in the SD card wav folder, will the MD be able to read it keeping this settings in a rom slot or will it downgrade at 12 bit?

jmamma, 2021-08-24:
Only loading (not saving) is currently supported when you "select several slots with Md arrows" followed by the YES button.

If you select multiple slots and then change jump, loop, row, length. The change will be applied to the selected slots for the altered settings only.

--

MD only supports 12bit playback, the MD reduces the bitrate upon upload.

albadort76, 2021-08-25:
That's fine, thanks!

closed 2021-08-25 by albadort76