jmamma/MCL #177: HexUploader no go on MacOS Catalina, Using Ardiuno IDE works... But MCL4.00 is displayed at startup.

BartBral, 2022-01-22:
1). 
The version of HexUploader doesn't work on (my) MacOS Catalina.

2).
I was perfectly able to get MCL to work on my MegaCommand using Ardiuno IDE, BUT... 
... When I start my MegaCommand, the text in the display says MCL 4.00.
and I double check and yes I have the version 4.04 folder.

So my question is:
Is this just an error in the version-number in the code of 4.04?
Or is the .HEX file different from the one Arduino IDE compiles out of the code?

jmamma, 2022-01-23:
The Catalina problem is a known issue.

If you know how to use terminal, you can try uploading via command line.

```
FILE=~/Downloads/mcl_4_0_4.hex;
DEV=/dev/$(ls /dev | grep usb | grep cu | tail -n 1); AVR_DIR="/Applications/Arduino.app/Contents/Java/hardware/tools/avr"; ${AVR_DIR}/bin/avrdude -C${AVR_DIR}/etc/avrdude.conf -v -V -patmega2560 -cwiring -P${DEV} -b115200 -D -Uflash:w:$FILE
```

  When I start my MegaCommand, the text in the display says MCL 4.00.

Did you compile the firmware yourself? If so, it seems as though you've compiled an older version.

BartBral, 2022-01-23:
> _"If you know how to use terminal, you can try uploading via command line."_

Cheers! I'll try that... think I'll be okay

...

> _"Did you compile the firmware yourself? If so, it seems as though you've compiled an older version."_

Not sure to be honest... I just ran the .INO file from the 4.04 directory... I thought that would do the job. I'll give it another go tomorrow when I've slept a bit. 

Cheers!

BartBral, 2022-01-25:
All worked fine with the Terminal !

I found out (afterwards) that you can simply make HexUploader work, by using:
sudo xattr -cr /path_to_the_file/HexUploader.app

Might have already been stated elseware... But I thought I let you know anyway.

frnktrpz, 2022-11-07:
I've tried using terminal to update my megacommand but get an error saying there's no java/hardware/etc. directory in my arduino app contents directory. Changing the attributes did nothing for me either. Clearly I have no idea what I'm doing. Anyone that could point me in the right direction?

closed 2023-08-01 by jmamma