jmamma/MCL #187: Can't compile

ctag-fh-kiel, 2024-06-29:
C:\arduino-1.8.18\hardware\MCL-4.51\avr\cores\megacommand\wiring.c:26:10: fatal error: helpers.h: No such file or directory
 #include "helpers.h"

helpers.h seems to be missing

ctag-fh-kiel, 2024-06-30:
When I change in wiring.c l.26 to "CommonTools/helpers" compilation works. 
However, the produced .hex file seems dysfunctional. 
When flashing, the verification step fails and on the hardware there is a boot screen produced, which doesn't include the version number.
I have tried various different Arduino versions >=1.8.5 and <2.0 same results. In fact 1.8.5 doesn't compile as the toolchain doesn't yet support the required c++ version.

Any advice is greatly appreciated :)

jmamma, 2024-06-30:
What method are you using to compile ?

The included makefile, or via the IDE?

jmamma, 2024-06-30:
I suspect the Arduino compiler options in platform.txt are out of date with the Makefile.

Which means compiling via the IDE does not currently work.

jmamma, 2024-06-30:
On linux/mac terminal. 
If you cd to ```hardware\MCL-4.51\avr\cores\megacommand``` and then run make.

jmamma, 2024-06-30:
I've just remembered, there's also another step that is now required **before compilation**. Which might explain your corrupt .hex

You'll need to cd ./resource and run either ```gen-resource-linux.ps1``` or ```gen-resource.ps1```

These are powershell scripts that compress the graphics and menu data before they are compiled in to the binary.

---

I will get around to fixing/documenting the above next week.

ctag-fh-kiel, 2024-06-30:
Hi there,

Thanks for getting back! 

Compiling resources seemed to work.

Building directly with make stops here
Arduino/MCL/avr/cores/megacommand/MCL/Grid.cpp:105:3: error: jump to label 'end'
  105 |   end:

Using avr-gcc 9.4.0 which is default from https://github.com/osx-cross/homebrew-avr

Fixing docs would be great, thx for that. I will gladly help to get a working build recipe for Mac.

:)

jmamma, 2024-06-30:
Can you compile with avr-gcc 7.30 . That's the version included with the legacy IDE.

Later versions of avr-gcc don't play well (increased binary size and stability issues)

ctag-fh-kiel, 2024-06-30:
Ok, will do, I realized the binary size increase as well.

Somehow my Mega2560 doesn't respond any more after the last flashing, I
guess I have to wait until a new one arrives.

ctag-fh-kiel, 2024-06-30:
Compilation runs thru with avr-gcc 7.30 (from legacy IDE).
Binary size of my version is slightly smaller than yours (689174 vs. 689334).

I can't give it a try just now 'cause of my dead Mega2560 ;)

ctag-fh-kiel, 2024-07-08:
Compilation works and I was able to flash and test on fresh Arduino Mega board.
Thx!

Updating the readme for a build how-to may be nice. Let me know if you'd like me to help.

jmamma, 2024-07-08:
Yes, if you'd like to submit a pull request to add any useful + concise information that is missing from the README.

Thank you.

ctag-fh-kiel, 2024-07-08:
Will do...

closed 2024-07-15 by ctag-fh-kiel