jmamma/MCL #189: Midi machines, automation issues

ctag-fh-kiel, 2024-07-13:
I have a question with regard to the following scenario: 

1. MD Midi Machine assigned to a track
2. Punched in a couple of trigs in enhanced mode
3. Automatized some Midi Machine parameters per trig in grid mode within enhanced mode
4. Set up MCL internally routing Midi 1 IN to Midi Out 2 (pg. 24f of MCL 4.5.1 documentation)
5. Midi Out 2 data only reflects note on/offs of Midi Machine but none of the recorded automations

It would be great if the automations are sent as well, making this useful for sequencing external gear with a MD Midi Machine and MCL in enhanced mode.

Is this currently the desired behaviour or am I missing something?

jmamma, 2024-07-14:
MD MIDI machines are currently not supported with MCL. The PianoRoll can be used to transmit MIDI notes on port 2.

ctag-fh-kiel, 2024-07-14:
Thanks, good to know!
Is it possible to record CC automation data in the PianoRoll?

Which are the points in the source code to possibly implement MD Midi
machine support, I could have a look at? Maybe I can figure something out.
Or is it more a technical limitation as for each step on the grid Midi is
sent from MCL to MD and would then be re-sent from MD to MCL and forwarded
to port 2 (which does sound to be timing wise a challenge)?

Cheers :)

jmamma, 2024-07-14:
> Is it possible to record CC automation data in the PianoRoll?

Yes, see the PianoRoll section of the manual. Each track can automate up to 8 CCs.

> Or is it more a technical limitation

You can find the main sequencer code MCLSeq::seq() in MCL/MCLSeq.cpp, 
We have the MDSeqTrack::seq() and ExtSeqTrack::seq() classes for the MD and PianoRoll sequencers. 

The idea would be to emulate the MD's MIDI sequencing via the MCL sequencer, and transmit directly from the MegaCommand's port 2.

There's at least three main issues. The first being, we would need to implement a distinct type of sequencer routine for the MIDI tracks. The MDSeqTrack::seq() routine, only needs to trigger a track once for each step trig. With the MIDI machines you need to send MIDI_NOTE_ON and then wait some duration before sending the MIDI_NOTE _OFF. 

The second issue is the GUI aspect. Currently the MIDI tracks on the MD transmit on their specific channel. I.e MID-05 transmits all control changes on channel 5. In normal circumstances (all other machines), the MD transmits within the range of (globalBaseChannel, globalBaseChannel + 4). MCL tracks parameter changes via MIDI CC. So any parameter value changes on the MD-MIDI tracks need to be intercepted.

The GUI implementation on the MD is accessible to me, so I would likely be able to change the GUI MIDI channel behavior for Enhanced Mode.

The third issue would be related to the number of parameters with locks. The MDSeqTrack::seq() routines were designed for 8 parameters, due to memory and cpu constraints. The MIDI machines really lend themselves towards 8+ parameters with locks.

--

> sent from MCL to MD and would then be re-sent from MD to MCL and forwarded
to port 2 (which does sound to be timing wise a challenge)?

And can't really do it this way because MCL transmit quite a bit of sysex data back and forth with the MD. Large sysex transfers result in latency issues for MIDI note data.

ctag-fh-kiel, 2024-07-15:
Thx for the detailed reply.

Now I am wondering if in the automation editor rack menu one could learn the Midi CC also from Midi in 1, coming from the MD. If one has set up a midi machine on the MD, one could use that to send Midi CC to MCL via moving the MD pots. This could be recorded in piano roll.

Somehow CC can only be learned from Midi in 2.

Could this be technically feasible?

jmamma, 2024-07-16:
From memory it's mostly coded to use port2, See the SeqPtcMidiEvents::onControlChangeCallback functions in SeqPtcPage.cpp.

--

On a side note I fixed the MD GUI issues for the MIDI tracks when in enhanced mode.

Just coding up first pass of the MIDI Machine sequencer code.

ctag-fh-kiel, 2024-07-16:
Very cool, thx. I will have a look at it. 
My use case is to connect a Midi version of [CTAG TBD](https://github.com/ctag-fh-kiel/ctag-tbd). TBD is a flexible audio processing module, currently available as Eurorack. The Midi version (in the works) can be a great extension to Elektron boxes by using their sequencer or the MCL. 
Let me know if you are interested, I could spare a prototype of the Midi TBD.

jmamma, 2024-07-16:
@ctag-fh-kiel jump on our discord when you have a chance and I can share the updated MD firmware. 

https://discord.gg/reFptpH6gp

jmamma, 2024-07-16:
https://github.com/jmamma/MCL/pull/191 <--- untested.

closed 2024-10-18 by jmamma