jmamma/MCL #192: machinedrum samples 12-bit playback but 16-bit stored

Tonguc-Endem, 2025-02-07:
It seems there is a bug or rather a very poor and irresponsible programming in the original firmware of Machinedrum.
The UW device utilizes samples at 12bit (sampling and playback) but stores them as 16bits thus wasting %25 of the storage.
This is easily testable by uploading a sample then receving it back and comparing the sent and received sample.  You're not getting a file with 12-bit dithering or a 16bit file with padded 12bit audio in it.  You're getting the exact 16bit file you initially sent.  This means there is a %25 waste in how the samples are stored in the machinedrum.  12bit drums sounds great and there is no reason to store them as 16bit files.
If its technically possible to playback actually properly 12bit stored samples, we should be able to get %25 more storage with a firmware that properly can store 12bit samples.
But then how do we send 12bit dithered files to the MD? maybe as SDS files? I'm not sure if c6 can do that. 
I'm thinking if a firmware can automatically dither 16bit received files to 12bits, store them as 12bit and play 12bit files, this could work and we'll get %25 more storage without loosing audio fidelity.
I'd like to think of this as a storage wasting bug due to bad programming.

emuyia, 2025-02-07:
I've experimented with this a bit.

Awave Studio can convert to 12bit SDS, so I used that to convert a 16bit mono 32 kHz WAV file, which was originally 433 kB. After conversion, it came to 459 kB, however to my understanding it's only bigger because of the SDS header, so I figured if MD discards the header during the transfer it shouldn't be a problem.

Unfortunately, no matter the bit depth, the file still appeared as 433 kB on the MD itself.

I echo your sentiment, and hope it's possible to work around this.

Tonguc-Endem, 2025-02-07:
@emuyia, thanks a lot for this comment, I was just getting ready to test that :-)
I would appreciate it if you would share your experience in the
[electronauts thread](https://www.elektronauts.com/t/machinedrum-sample-transfer-12-bit-playback-but-16-bit-stored/228707)